zephyrproject-rtos/zephyr #53911: Info request: SMP Hash

pmagnuson, 2023-01-18:
This is a follow-on to https://github.com/zephyrproject-rtos/zephyr/issues/53799, which is now closed.  
I really appreciate all the responses to the info request.  It's helping a lot.  

I have a file, `app_update_z_ot1.bin`, that I loaded onto the nRF dev board.  
`imgtool` reports:

```
imgtool verify app_update_z_ot1.bin
Image was correctly validated
Image version: 0.0.0+0
Image digest: ebe393b41171036e223fc2686bd03ab7211fc2e6d0f86415750f9dadf06c7283
```

When the code does an SMP request for State of Images, the response includes:
```
"hash":"6-OTtBFxA24iP8Joa9A6tyEfwubQ-GQVdQ-drfBscoM"
```

I'm not sure why the reported hash is different than the `imgtool` report.  Is there another transformation performed on the digest hash?

de-nordic, 2023-01-19:
When you compile app you get something like app.bin, imgtool adds header to it, padds image, calculates SHA overt it and and then adds the TLV with the SHA:
https://github.com/mcu-tools/mcuboot/blob/4427e4c9775938a9617536ca6cb4d368368f36db/docs/design.md?plain=1#L108
The binary you upload is extended with additional fields (TLV) that also carry the calculated SHA.

More context: https://github.com/mcu-tools/mcuboot/blob/main/docs/design.md?plain=1#L1134-L1148

The SHA reported by image list is read from the TLV, not calculated over uploaded image, from `img_mgmt_read_info`
https://github.com/zephyrproject-rtos/zephyr/blob/ad06dfc784cab4923863cea35909f582951126e1/subsys/mgmt/mcumgr/grp/img_mgmt/src/img_mgmt.c#L178-L186

The SHA returned as hash and the one you use in upload is sha of entire binary blob you upload. This differs because you are expected to pass SHA of the thing you are uploading, to SHA you have calculated on the entire bin blob. Then the device will at the end re-read what has been written and return the has it has calculated over uploaded size. If these differ, then something went wrong or there is flash read error.

nordicjm, 2023-01-19:
> "hash":"6-OTtBFxA24iP8Joa9A6tyEfwubQ-GQVdQ-drfBscoM"

That is not a valid hash in sha256 format, it would be a hex string, this looks like it is in another format or has some encoding applied to it. Did you use the MCUmgr go application?

pmagnuson, 2023-01-19:
Thank you all very much for helping me with this issue.  I am new to this so it's really gracious of you to educate me on these topics.  
My goal is to write my own mobile app that can update FW on our Bluetooth devices.  

here are the steps I am going through.

1. Code file is compressed in this .zip [app_update_z_ot1.bin.zip](https://github.com/zephyrproject-rtos/zephyr/files/10458062/app_update_z_ot1.bin.zip)
2. imgtool reports the signature as: 
```
imgtool verify app_update_z_ot1.bin
Image was correctly validated
Image version: 0.0.0+0
Image digest: ebe393b41171036e223fc2686bd03ab7211fc2e6d0f86415750f9dadf06c7283
```
3. I used [MCUmgr-IOS ](https://github.com/JuulLabs-OSS/mcumgr-ios) to download this code file to a nRF dev board.    
4. Using my own application (Flutter app running on IOS) a `Image status request` is issued to SMP:
```
send: 
{ op: 0, groupID: 1, command: 0, sequence: 0, length: 1, data: {} }
```
5. The response is: 
```
receive: 
{
  op: 1,
  groupID: 1,
  command: 0,
  sequence: 0,
  length: 131,
  data:
    {
      "images":
        [
          {
            "slot": 0,
            "version": "0.0.0",
            "hash": "6-OTtBFxA24iP8Joa9A6tyEfwubQ-GQVdQ-drfBscoM",
            "bootable": true,
            "pending": false,
            "confirmed": true,
            "active": true,
            "permanent": false,
          },
        ],
      "splitStatus": 0,
    },
}

```

6. Based on the conversation in [53799](https://github.com/zephyrproject-rtos/zephyr/issues/53799) I am expecting the signature of the imgtool image digest to be returned in `"hash"`.  
7. In addition, if I download a new code file, I believe this issue is also preventing the new code file from being accepted.  

Again, thank you so much for looking into this.  Your help is greatly appreciated.

de-nordic, 2023-01-19:
How do you catch the response?
The hash is binary string, not a text string. How do you decode it?

de-nordic, 2023-01-19:
If you have followed the doc and decoded it as a string, then sorry but there has been a mistake in documentation,
https://github.com/zephyrproject-rtos/zephyr/pull/53941

pmagnuson, 2023-01-20:
Thank you for all the comments and input.  Based on your comments, I believe I had an error in the cbor encoding/decoding.  Thank you all very much.

closed 2023-01-20 by pmagnuson