zephyrproject-rtos/zephyr #53917: clang-format key incompatible with IntelliJ IDEs

yperess, 2023-01-19:
**Describe the bug**
A relatively new addition to `.clang-format` is the `InsertBraces: true` line which was added in clang format 15. CLion (stable) hasn't quite caught up to that and this breaks using clang-format in the IDE

**To Reproduce**
Steps to reproduce the behavior:
1. Open zephyr using CLion
2. Enable clang-format
3. Edit any `.c` or `.h` file

**Expected behavior**
Clang format should work

**Impact**
Can no longer use my IDE

**Logs and console output**
```
Error reading /home/peress/workspace/zephyrproject/zephyr/.clang-format:80:0: Invalid argument - unknown key 'InsertBraces'
```

keith-zephyr, 2023-01-19:
Does CLion provide any hooks to run a custom program on file save? This might allow you to use an up to date clang-format until the built-in version gets updated.

kartben, 2023-05-17:
closing this with the proper reason (i.e it is *not* completed). @yperess please feel free to re-open, of course.

closed 2023-05-17 by kartben