zephyrproject-rtos/zephyr #53937: usb: stm32g0: sometimes get write error during CDC ACM enumeration when using USB hub

jeromephilbert, 2023-01-19:
**Describe the bug**
On the Nucleo G0B1RE board, I run the USB CDC ACM sample program.
The board is connected to a Linux PC on 2 USB ports (see attached pictures):
- USB ST-Link => USB port 1
- PA11/PA12/GND => USB cable => USB hub => USB port 2

Sometimes when I plug the USB cable to the USB hub, I get USB write errors in the console:
`
[00:26:37.864,000] <err> usb_dc_stm32: Unable to get write lock (-16)                               
[00:26:37.864,000] <err> usb_dc_stm32: Unable to get write lock (-16)                               
[00:26:37.864,000] <err> usb_dc_stm32: Unable to get write lock (-16)                               
[00:26:42.992,000] <err> usb_dc_stm32: Unable to get write lock (-16)                               
[00:26:42.992,000] <err> usb_dc_stm32: Unable to get write lock (-16)                               
[00:26:42.992,000] <err> usb_dc_stm32: Unable to get write lock (-16)                               
[00:26:42.992,000] <err> usb_dc_stm32: Unable to get write lock (-16)                               
[00:26:48.117,000] <err> usb_dc_stm32: Unable to get write lock (-16)                               
[00:26:48.117,000] <err> usb_dc_stm32: Unable to get write lock (-16)                               
[00:26:48.117,000] <err> usb_dc_stm32: Unable to get write lock (-16)                               
[00:26:48.117,000] <err> usb_dc_stm32: Unable to get write lock (-16)                               
[00:26:58.868,000] <inf> usb_cdc_acm: Device configured
`
This adds about 20 seconds to the enumeration phase.
If I remove the USB hub and connect the USB cable from the board directly to the 
PC, I do not encounter such errors.
I tested with 2 different USB hubs and I got the same results.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect ST-Link USB port to the PC
2. Build the USB CDC ACM sample for nucleo_g0b1re: west build -b nucleo_g0b1re samples/subsys/usb/cdc_acm
3. Flash it
4. Connect PA11/PA12/GND to a USB cable, plug it to the PC through a USB hub
5. Plug/unplug USB cable to/from the USB hub, in loop until getting errors in the console

**Expected behavior**
No error should occur.

**Impact**
Sometimes USB enumeration takes up to 20 seconds, which is not acceptable in production.

**Environment (please complete the following information):**
 - PC OS: Linux Ubuntu 18.04.6 LTS
 - PC with USB 3 ports
 - USB 3 hub
 - Toolchain: Zephyr SDK
 - Zephyr sha1 3d6f3a52cf075d8d9f3a94fe2c30156423fc1066 (Jan 13 2023)
 - Board: nucleo_g0b1re 

[image]
[image]

nordicjm, 2023-01-19:
Adding both for possible USB and/or STM32 issue since Brix has seen similar issues with hubs/docks on different boards

jfischer-no, 2023-01-19:
> If I remove the USB hub and connect the USB cable from the board directly to the PC, I do not encounter such errors.
> I tested with 2 different USB hubs and I got the same results.

There is no dedicated USB port. The only thing left is to find out if it is the shim driver or your cable connection between controller and device.

jeromephilbert, 2023-01-19:
The issue also occurs with a shorter USB cable.
[image]

jfischer-no, 2023-01-19:
@erwango VBUS sensing is required, right?

jeromephilbert, 2023-01-20:
> @erwango VBUS sensing is required, right?

You mean the VDD pin of the USB cable?
I also tested with VDD connected to 5V on the board, it does not change anything, I get the same errors.

erwango, 2023-01-20:
@jeromephilbert I think VBUS sensing refers as what was implemented here recently https://github.com/zephyrproject-rtos/zephyr/pull/53239. This is worth having a try.

jeromephilbert, 2023-01-20:
> @jeromephilbert I think VBUS sensing refers as what was implemented here recently #53239. This is worth having a try.

VBUS detection is not supported by the USB controller of the STM32G0.
This could be done with a dedicated GPIO but this seems not supported by the actual STM32 USB driver.

jeromephilbert, 2023-02-09:
Yes, we can close this issue, since it is expected according to the USB specification.

jfischer-no, 2023-02-09:
> Yes, we can close this issue, since it is expected according to the USB specification.

I have to ask, what is to be expected?

jeromephilbert, 2023-02-09:
VBUS sensing is mandatory for self-powered devices according to the USB specification, that means that if the MCU is using an external power input the USB driver will operate correctly only if the voltage sensing is enabled.
Problem is that VBUS sensing is not supported by the USB controller of the STM32G0, but it is another topic.

jfischer-no, 2023-02-09:
> VBUS sensing is mandatory for self-powered devices according to the USB specification, that means that if the MCU is using an external power input the USB driver will operate correctly only if the [...]

Can you please test if it works without issues when your board is self-powered, means remove cable from USB_STLINK and connect VBUS line to board's external power pin, not sure there is one, maybe +5 on arduino connector?

aurel32, 2023-04-06:
I already encountered similar behaviour (but on other STM32 families) and this smells like a clock accuracy issue. By default Zephyr uses the HSI48 oscillator for USB, however it does not enable the CRS to increase the accuracy. I usually use the following configuration code in my projects to workaround the issue:

```
LL_APB1_GRP1_EnableClock(LL_APB1_GRP1_PERIPH_CRS);
LL_CRS_EnableAutoTrimming();
LL_CRS_EnableFreqErrorCounter();
```

If it affects more people, it might be the time to fix it properly. I have opened PR #56639 as a first step.

erwango, 2023-07-18:
@jeromephilbert did you had a try using https://github.com/zephyrproject-rtos/zephyr/pull/56639 ?

Desvauxm-st, 2023-12-19:
@jeromephilbert 
can you try again we push this PR
[drivers: clock_control: stm32: add an option to enable CRS for HSI48](https://github.com/zephyrproject-rtos/zephyr/pull/63111#top)
#63111

closed 2024-05-12 by github-actions[bot]