zephyrproject-rtos/zephyr #53946: `k_pipe_put`: data parameter should be const?

Crzyrndm, 2023-01-19:
**Is your enhancement proposal related to a problem? Please describe.**
Docs: https://docs.zephyrproject.org/3.1.0/kernel/services/data_passing/pipes.html#c.k_pipe_put
The data parameter not being marked as const limits the data sources that can be passed to pipes. I use const where I can and the lack of const here makes pipes very difficult to use correctly

The [message queue equivalent is const](https://docs.zephyrproject.org/3.1.0/kernel/services/data_passing/message_queues.html#c.k_msgq_put). It would be highly irregular to actually write to the source buffer during the put operation, and adding const has no backward compatibility issues (it only increases the valid data sources)

peter-mitsis, 2023-01-31:
I'm happy to make this change.

peter-mitsis, 2023-02-28:
Update: This issue is still on my radar. :) I started digging into this a little more, and remembered why I did not originally use `const`; it has to do with an implementation detail. I'll explain.Threads waiting on either `k_pipe_put()` and `k_pipe_get()` use a custom structure that tracks the amount of data left to copy--either by reading or by writing. We currently use a single pointer for this since a thread can not be simultaneously reading and writing from a pipe. This had the drawback of `k_pipe_put()` not utilizing the `const` keyword with its data. Broadly speaking, I can imagine two solutions to this.

1. Surface. Only change the `k_pipe_put()` API to use `const`, but internally cast it to ignore the `const` keyword. I do not particularly like this approach as it violates the `const` integrity--in essence merely pushing the problem that it described in this issue down one level.
2. Update the structure `_pipe_desc` to use distinct  source and destination pointers and then update the buffer copying code to utilize this modified `_pipe_desc structure`. This would be a preferable solution. At the present moment, I have a few higher priority items on my plate, so it will take me a little bit of time to get around to fixing this.

PR #66331 (2023-12-15): kernel: Tweak k_pipe_put() API to use 'const' with data parameter
In the function _k_pipe_put()_, the _data_ parameter, which points to a buffer that is written to the pipe, now uses the `const` keyword. Internally, that pointer is cast to drop the 'const' to work with the common code that both _k_pipe_put()_ and _k_pipe_get()_ share as it is known that k_pipe_put() will take a path where the contents of that buffer will not be modified.

Fixes #53946
closed 2023-12-15 by nashif