zephyrproject-rtos/zephyr #53952: USB C PD sink sample stops working when connected to a non-PD source

deyro, 2023-01-20:
**Describe the bug**

I am testing the `sink` sample of USB C stack on a ST `b_g474e_dpow1` board. The sample works fine with a USB C PD source, but when I connect a non-PD source (wall adapter or PC) via a USB type A to type C cable, the application stops responding (hangs). If I attach the PD source again there are no log messages.

I also have a custom board with PD capabilities using STM32G473 where I face the same problem. I found out that the program loops in the driver `ucpd_stm32.c`, where `ucpd_manage_tx()` function keeps getting called and the `ucpd_tx_state is STATE_HARD_RESET`. The control never breaks from the below do-while loop. 

```
	/*
	 * Manage PD tx messages. The state machine may need to be
	 * called more than once. For instance, if
	 * the task is woken at the completion of sending a GoodCRC,
	 * there may be a TCPM message request pending and just changing
	 * the state back to idle would not trigger start of transmit.
	 */
	do {
		ucpd_manage_tx(info);
	} while (data->ucpd_tx_state != STATE_IDLE);
```

Is this a bug or I am doing something incorrectly?

**Impact**
Showstopper

**Logs when I connect a non-PD source**
I have added prints in the `port1_notify function of the main.c file`
```
*** Booting Zephyr OS build zephyr-v3.2.0-3518-g7b5780e00764 ***
[00:00:00.000,000] <inf> usbc_stack: Disabled
[00:00:00.000,000] <inf> usbc_stack: PE_SUSPEND
[00:00:00.000,000] <inf> usbc_stack: PRL_HR_SUSPEND
[00:00:00.000,000] <inf> usbc_stack: PRL_TX_SUSPEND
[00:00:00.000,000] <inf> usbc_stack: ErrorRecovery
[00:00:00.245,000] <inf> usbc_stack: Unattached.SNK
[00:00:00.250,000] <inf> usbc_stack: AttachWait.SNK
[00:00:00.459,000] <inf> usbc_stack: Attached.SNK
[00:00:00.464,000] <inf> usbc_stack: PE_SNK_Startup
[00:00:00.464,000] <inf> main: NOT_PD_CONNECTED
[00:00:00.464,000] <inf> usbc_stack: PRL_INIT
[00:00:00.464,000] <inf> usbc_stack: PRL_HR_Wait_for_Request
[00:00:00.464,000] <inf> usbc_stack: PRL_Tx_PHY_Layer_Reset
[00:00:00.464,000] <inf> usbc_stack: PRL_Tx_Wait_for_Message_Request
[00:00:00.469,000] <inf> usbc_stack: PE_SNK_Discovery
[00:00:00.474,000] <inf> usbc_stack: PE_SNK_Wait_For_Capabilities
[00:00:00.484,000] <inf> main: PWR DEF
```

**Environment**
 - OS: Linux
 - Toolchain: zephyr-sdk-0.15.2
 - Zephyr SDK zephyr-v3.2.0-3518-g7b5780e007
 - Hardware: ST b_g474e_dpow1 and custom hardware using STM32G473

sambhurst, 2023-01-22:
@deyro Yes this is a bug and I've found the problem. I'll upload a PR shortly. Thanks

deyro, 2023-01-22:
@sambhurst thank you!

closed 2023-01-24 by sambhurst