zephyrproject-rtos/zephyr #53955: [RFC] mgmt: mcumgr: Add command result code (rc) optional response

nordicjm, 2023-01-20:
**Is your enhancement proposal related to a problem? Please describe.**
At present, in MCUmgr, there is a "rc" result code variable that is returned to the client, if an error has occurred. This variable is designed around returning an error at the SMP-level, however it has been wrongly used in the past for returning the status of an actual command i.e. command level error, despite the fact that the MCUmgr command was received and processed correctly. For command handlers, error handling is a mess, it is in fact impossible to know what an error is.

For fs_mgmt, the following error codes are returned:

- MGMT_ERR_ENOENT: if a file does not exist
- MGMT_ERR_EUNKNOWN: any sort of non-0 return code from a function (could be any sort of error, what does it actually mean?)
- MGMT_ERR_EINVAL: if an invalid value was send, if the underlying state is wrong, if the write size is larger than the original size provided (these are very different errors)

for img_mgmt, the follow error codes are returned:

- MGMT_ERR_EINVAL: if CBOR decoding is unsuccessful, if offset/length field is missing, if first packet size is too small, if image header magic is incorrect, if sha256 length is too long, if offset and header address are different
- MGMT_ERR_ENOENT: if no free slot
- MGMT_ERR_EUNKNOWN: any sort of non-0 return code from flash functions, from version check function, if upload slot already contains an image and progressive erase is not enabled, if image ID is not valid, if slot is erased, if reading slot header fails, if slot header magic is not valid/erased, if image TLV lookup fails, if more than one hash is present, if a hash is not found

As can be seen from the groupings above, from a MCUmgr client, it is impossible to know what the error is, conversely it is also impossible to know how to handle that error because error handling depends upon what the problem is - if the problem is that (e.g. img_mgmt during an upload) a corrupt image was uploaded and it needs to be erased, how would a client know that is the actual error?

**Describe the solution you'd like**
To add a new result code response, with a different name and meaning to the existing one, it could be named something like "cmdrc" and either have an error code list that is unique to each transport, or matches up with zephyr's errno - the latter would probably be preferred because then functions can just return this directly, e.g. calls to opening files no longer need a translation layer to convert from errno to cmdrc then cmdrc to errno, but this may be problematic because from a clients perspective, those errno errors could match up to multiple different causes still whereas a dedicated error list for each group would not, e.g. file_open_failed, file_seek_failed, file_write_failed, file_in_use, etc. could all be defined

**Describe alternatives you've considered**
Replacing the MCUmgr rc directly with this - that then becomes an issue because then it is not known if there is an issue with the command itself or execution of the command
Not doing anything - the same problem that MCUmgr has now, inability of knowing what an error means or how to do anything with it

nordicjm, 2023-01-20:
CC @de-nordic @carlescufi

nordicjm, 2023-01-20:
Things to also consider:

- If this change is agreed, how will it be implemented - will it replace the existing groups making it incompatible, for devices in the field which upgrade, suddenly the commands will respond different? Should a new group be added for returning something like the MCUmgr system version - then if a device has that, it can be queried to know what the command structure should be (this could also be used going forward if breaking changes are introduced in the future) and if a device does not have that, it would use the old method. Or would it be introduced with new group IDs, keeping the old group IDs available and deprecating them so that they can still be enabled?

de-nordic, 2023-01-26:
This addresses issue I have been also rising here https://github.com/zephyrproject-rtos/zephyr/issues/42590. I am in favor of giving the "rc" codes the meaning they should have, as SMP protocol processing return codes.

Regarding the backward compatibility issue, I was considering using the `res` field of header (https://docs.zephyrproject.org/latest/services/device_mgmt/smp_protocol.html) to pass SMP protocol version; for some time we could support two version (00 and 01), where "rc" use in  responses would differ depending on header version in request.

PR #54041 (2023-05-15): mgmt: mcumgr: Add command result codes
A work in progress at the possible way of adding a command result code which can indicate status of the command in a generic way back to the client without relying upon the SMP result code.

Fixes #53955

- [x] new function
- [x] result codes
- [x] fs group
- [x] img group
- [x] ~~os group~~
- [x] zephyr group
- [x] stat group
- [x] shell group
- [x] MCUmgr hook updates
- [ ] ~~error code description for documentation~~ *deferring to another PR*
- [x] release notes
- [x] documentation update
closed 2023-05-15 by carlescufi