zephyrproject-rtos/zephyr #53967: net: http client: HTTP timeout can lead to deadlock of global system queue

wcappelle, 2023-01-20:
**Describe the bug**
To be able to have http timeouts working, a driver should implement the socket shutdown call.
Therefore I'm implementing the shutdown call in my own offload driver (bg95), but during testing I've come up with a problem which leads to a deadlock in the http client design.

When doing a http_client_req in a different thread, that thread is blocking on a receive call (socket receive) after it's done the post. This socket is being locked (https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/lib/sockets/sockets.c#L54) during this blocking receive call. When a timeout occurs, this is not in the thread context, but on the global system queue. This call is trying to obtain the lock again in the socket to do the shutdown call (https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/lib/http/http_client.c#L499) on the socket, but it's blocked due to a different thread requesting the lock. This is blocking the global system queue forever.

This seems to me like a bad design of the http client since it's calling in different threads on the same socket.
Any suggestion on how to fix would be welcome (eg: replacing the timeout by a poll timeout on the receive)

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - Toolchain Zephyr SDK 15.1
 - Zephyr Release 3.2

rlubos, 2023-01-23:
Generaly, when mutex protection was introduced for sockets, the idea was to allow access to a socket from different threads, by unlocking the mutex when socket function is blocked (for instance in `recv()`). Native socket implementation achieves that by using a conditional variable, and offloaded sockets should follow this pattern (in worst case by simply releasing the mutex for the offloaded call). 

I'm ok though with updating the http_client implementation by using `poll()`, IMO it's better to use built-in mechanisms to handle timeout.

wcappelle, 2023-01-23:
@rlubos , thanks for the clarification. For the unlocking of the socket mutex, it feels not as a good design to control a mutex from socket layer in the driver? I also couldn't find any documentation on this (as far as I could see). Since you're okay with changing this to poll, i will update it as propsed.

rlubos, 2023-01-24:
> it feels not as a good design to control a mutex from socket layer in the driver?

Well, if it was a regular driver, then yes it'd be awkward, but here we'll talking about offloaded devices. In this case the offloaded implementation takes full control of the socket call, so IMO it's ok to handle the mutex (just like the native implementation does).

> I also couldn't find any documentation on this (as far as I could see)

Very likely it's not documented well.

commit aaefd3db (2023-01-24): net: http: Use poll for http timeout instead of shutdown

Change the http timeout mechanism to use poll instead of shutdown.
This should fix a problem where the shutdown will be called in a
different thread context which can lead to deadlocks on certain
driver implementations like offloaded modem drivers.
Fixes #53967
closed 2023-01-24 by carlescufi