zephyrproject-rtos/zephyr #53991: LE Audio: samples/bluetooth/broadcast_audio_sink configure error

rylox, 2023-01-22:
**Describe the bug**
Sample can't be build due to configure error raised by Kconfig warnings.

**To Reproduce**
Steps to reproduce the behavior:
open bash
cd /c/ncs/v2.2.0/zephyr/samples/bluetooth/broadcast_audio_sink
west build -b nrf52840dk_nrf52840 -p auto -d 52840

**Logs and console output**
```
Leon@DESKTOP-8R9HEG7 MINGW64 /c/ncs/v2.2.0/zephyr/samples/bluetooth/broadcast_audio_sink ((v3.2.99-ncs1))
$ west build -b nrf52840dk_nrf52840 -p auto -d 52840
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: C:/ncs/v2.2.0/zephyr/samples/bluetooth/broadcast_audio_sink
-- Using NCS Toolchain 2.2.0 for building. (C:/ncs/toolchains/v2.2.0/cmake)
-- Found Python3: C:/ncs/toolchains/v2.2.0/opt/bin/python.exe (found suitable exact version "3.8.2") found components: Interpreter
-- Cache files will be written to: C:/ncs/v2.2.0/zephyr/.cache
-- Zephyr version: 3.2.99 (C:/ncs/v2.2.0/zephyr)
-- Found west (found suitable version "0.14.0", minimum required is "0.7.1")
-- Board: nrf52840dk_nrf52840
-- Found host-tools: zephyr 0.15.1 (C:/ncs/toolchains/v2.2.0/opt/zephyr-sdk)
-- Found toolchain: zephyr 0.15.1 (C:/ncs/toolchains/v2.2.0/opt/zephyr-sdk)
-- Found Dtc: C:/ncs/toolchains/v2.2.0/opt/bin/dtc.exe (found suitable version "1.4.7", minimum required is "1.4.6")
-- Found BOARD.dts: C:/ncs/v2.2.0/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
-- Generated zephyr.dts: C:/ncs/v2.2.0/zephyr/samples/bluetooth/broadcast_audio_sink/52840/zephyr/zephyr.dts
-- Generated devicetree_generated.h: C:/ncs/v2.2.0/zephyr/samples/bluetooth/broadcast_audio_sink/52840/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: C:/ncs/v2.2.0/zephyr/samples/bluetooth/broadcast_audio_sink/52840/zephyr/dts.cmake

warning: BT_CTLR_SCAN_DATA_LEN_MAX (defined at subsys/bluetooth\controller/Kconfig:584) was assigned
the value '251' but got the value ''. Check these unsatisfied dependencies: BT_OBSERVER (=n),
BT_CTLR_ADV_EXT (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_BT_CTLR_SCAN_DATA_LEN_MAX and/or look up
BT_CTLR_SCAN_DATA_LEN_MAX in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

warning: BT_CTLR_SYNC_ISO (defined at subsys/bluetooth\controller/Kconfig:699,
subsys/bluetooth\controller/Kconfig:708) was assigned the value 'y' but got the value 'n'. Check
these unsatisfied dependencies: ((BT_OBSERVER && BT_CTLR_SYNC_ISO_SUPPORT && BT_CTLR_ADV_EXT &&
BT_CTLR && BT_HCI && BT) || (BT_CTLR_ADV_EXT && BT_CTLR && BT_HCI && BT)) (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_BT_CTLR_SYNC_ISO and/or look up
BT_CTLR_SYNC_ISO in the menuParsing C:/ncs/v2.2.0/zephyr/Kconfig
Loaded configuration 'C:/ncs/v2.2.0/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840_defconfig'
Merged configuration 'C:/ncs/v2.2.0/zephyr/samples/bluetooth/broadcast_audio_sink/prj.conf'
Merged configuration 'C:/ncs/v2.2.0/zephyr/samples/bluetooth/broadcast_audio_sink/boards/nrf52840dk_nrf52840.conf'
config/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.

warning: BT_CTLR_SYNC_ISO_PDU_LEN_MAX (defined at subsys/bluetooth\controller/Kconfig:762) was
assigned the value '155' but got the value ''. Check these unsatisfied dependencies:
BT_CTLR_SYNC_ISO (=n), BT_CTLR_ADV_EXT (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_BT_CTLR_SYNC_ISO_PDU_LEN_MAX and/or look up
BT_CTLR_SYNC_ISO_PDU_LEN_MAX in the menuconfig/guiconfig interface. The Application Development
Primer, Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual
might be helpful too.

warning: Experimental symbol BT_AUDIO is enabled.

warning: Experimental symbol BT_AUDIO_BROADCAST_SINK is enabled.

warning: BT_ISO_SYNC_RECEIVER (defined at subsys/bluetooth\Kconfig.iso:54) has direct dependencies (!BT_CTLR || BT_CTLR_SYNC_ISO_SUPPORT) && BT_HCI && BT with value n, but is currently being y-selected by the following symbols:
 - BT_AUDIO_BROADCAST_SINK (defined at C:/ncs/v2.2.0/zephyr/subsys/bluetooth/host\../audio\Kconfig.baps:152, subsys/bluetooth\host\../audio\Kconfig.baps:152), with value y, direct dependencies (BT_PERIPHERAL && BT_PAC_SNK && BT_AUDIO && BT_HCI_HOST && BT_RPC_STACK) || (BT_PERIPHERAL && BT_PAC_SNK && BT_AUDIO && BT_HCI_HOST && BT_HCI && BT) (value: y), and select condition BT_PERIPHERAL && BT_PAC_SNK && BT_AUDIO && BT_HCI_HOST && BT_HCI && BT (value: y)

warning: default value 1 on BT_AUDIO_BROADCAST_SNK_COUNT (defined at C:/ncs/v2.2.0/zephyr/subsys/bluetooth/host\../audio\Kconfig.baps:168, subsys/bluetooth\host\../audio\Kconfig.baps:168) clamped to 0 due to being outside the active range ([0, 0])

error: Aborting due to Kconfig warnings

CMake Error at C:/ncs/v2.2.0/zephyr/cmake/modules/kconfig.cmake:328 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  C:/ncs/v2.2.0/nrf/cmake/modules/kconfig.cmake:29 (include)
  C:/ncs/v2.2.0/zephyr/cmake/modules/zephyr_default.cmake:108 (include)
  C:/ncs/v2.2.0/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  C:/ncs/v2.2.0/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: 'C:\ncs\toolchains\v2.2.0\opt\bin\cmake.EXE' '-DWEST_PYTHON=C:\ncs\toolchains\v2.2.0\opt\bin\python.exe' '-BC:\ncs\v2.2.0\zephyr\samples\bluetooth\broadcast_audio_sink\52840' -GNinja -DBOARD=nrf52840dk_nrf52840 '-SC:\ncs\v2.2.0\zephyr\samples\bluetooth\broadcast_audio_sink'

Leon@DESKTOP-8R9HEG7 MINGW64 /c/ncs/v2.2.0/zephyr/samples/bluetooth/broadcast_audio_sink ((v3.2.99-ncs1))
$
```

**Environment (please complete the following information):**
 - OS: Windows 10
 - Toolchain nRF Connect SDK
 - Version 2.2.0

rylox, 2023-01-23:
Resolved by changing the Bluetooth controller from the ncs iso to the zephyr controller.

closed 2023-01-23 by rylox

JoeBakalor, 2023-05-11:
> Resolved by changing the Bluetooth controller from the ncs iso to the zephyr controller.

@rylox what changes did you have to make to use the zephyr controller?