zephyrproject-rtos/zephyr #53993: platform: Raspberry Pi Pico area: USB Default config should be bus powered device for the Raspberry Pi Pico

MrGreensWorkshop, 2023-01-23:
**Describe the bug**
Raspberry Pi Pico is designed to be bus powered device, but Zephyr RTOS default USB config is self powered device.

check out for USB Configuration Descriptor attribute (bmAttributes) bits.
https://www.keil.com/pack/doc/mw/USB/html/_u_s_b__configuration__descriptor.html

**To Reproduce**
Steps to reproduce the behavior:
1. Compile Raspberry Pi Pico usb cdc example 
2. Connect the device to the host.
4. Check the bmAttributes like below. (the lsusb command in Linux is preferred). 

```shell
# 0xc0 means, self powered device.
# device is connected to 3:6 on my environment, this could be differ.
~$ lsusb -s 3:6 -v | grep bmAttributes
    bmAttributes         0xc0
        bmAttributes            3
        bmAttributes            2
        bmAttributes            2
 ```

**Expected behavior**
This is what would you expect when you connect bus powered device.
```shell
# 0x80 means, bus powered device.
~$ lsusb -s 3:3 -v | grep bmAttributes
    bmAttributes         0x80
        bmAttributes            3
        bmAttributes            2
        bmAttributes            2
```
**Impact**
It doesn't comply with USB specs.

**Environment (please complete the following information):**
all

**Additional context**

This could be fixed by adding `CONFIG_USB_SELF_POWERED=n` to the [rpi_pico_defconfig](https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/rpi_pico/rpi_pico_defconfig) As you know, unless the user adds `CONFIG_USB_DEVICE_STACK=y` these settings do not change anything.

In addition, because this is a bus-powered device, I believe you should also consider configuring USB max power. These are the settings that match the pico_sdk.

```yml
#set to bus powered mode, because default is y
CONFIG_USB_SELF_POWERED=n
#Maximum Power Consumption in 2mA units (125 -> 250mA)
CONFIG_USB_MAX_POWER=125

CONFIG_USB_DEVICE_VID=0x2e8a
CONFIG_USB_DEVICE_PID=0x000a
CONFIG_USB_DEVICE_PRODUCT="Pico"
CONFIG_USB_DEVICE_MANUFACTURER="Raspberry Pi"
```

nordicjm, 2023-01-23:
>  Raspberry Pi Pico is designed to be bus powered device

No it's not, it's designed so that it can be in any configuration the user wants it to be in

>  It doesn't comply with USB specs.

It does.

> This could be fixed by adding CONFIG_USB_SELF_POWERED=n to the [rpi_pico_defconfig](https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/rpi_pico/rpi_pico_defconfig). As you know, unless the user adds CONFIG_USB_DEVICE_STACK=y these settings do not change anything.

That would be the required setup for a project

> In addition, because this is a bus-powered device, I believe you should also consider configuring USB max power. These are the settings that match the pico_sdk.

Again, invalid, it depends entirely upon the end device - what other components are on the circuit, what peripherals the MCU is using, what clock speed it is running at.

MrGreensWorkshop, 2023-01-23:
To be clear, I'm talking about Raspberry Pi Pico board. I didn't say any single thing about RP2040. 
I'm talking about boards/arm/rpi_pico/rpi_pico_defconfig which is board config.

nordicjm, 2023-01-23:
> To be clear, I'm talking about Raspberry Pi Pico board. I didn't say any single thing about RP2040.

The RP2040 is wrongly labelled as pico in the whole zephyr tree:

```
./dts/bindings/watchdog/raspberrypi,pico-watchdog.yaml
./dts/bindings/usb/raspberrypi,pico-usbd.yaml
./dts/bindings/spi/raspberrypi,pico-spi.yaml
./dts/bindings/serial/raspberrypi,pico-uart.yaml
./dts/bindings/sensor/raspberrrypi,pico-temp.yaml
./dts/bindings/reset/raspberrypi,pico-reset.yaml
./dts/bindings/pwm/raspberrypi,pico-pwm.yaml
./dts/bindings/pinctrl/raspberrypi,pico-pinctrl.yaml
./dts/bindings/gpio/raspberrypi,pico-gpio.yaml
./dts/bindings/flash_controller/raspberrypi,pico-flash-controller.yaml
./dts/bindings/adc/raspberrypi,pico-adc.yaml
```

Hence the confusion

yonsch, 2023-01-23:
This naming has been discussed before and even went through dev review. RP2040 is a SoC name, not a series name. RP2 would also be wrong, because the 2 refers to two cores. RP/rpi/raspberrypi is confusing because the single board computers can be used both for development and as Zephyr targets.
The only clue we have as to what the series is called is the SDKs name - pico-sdk. I agree that this is confusing, but it's the best we can do until RP will release another SoC.

MrGreensWorkshop, 2023-01-23:
@nordicjm Maybe I didn't explain it well. I edited the description. 
I'm taking about boards/arm/rpi_pico/rpi_pico_defconfig,   (boards folder)
I didn't say anything about bindings or devicetree.. I don't see any reason to confuse.
Why don't you create an issue about that? Can we please get to the issue?

jfischer-no, 2023-01-23:
> This could be fixed by adding `CONFIG_USB_SELF_POWERED=n` to the [rpi_pico_defconfig](https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/rpi_pico/rpi_pico_defconfig). As you know, unless the user adds `CONFIG_USB_DEVICE_STACK=y` these settings do not change anything.

There are a few other boards where we should change it accordingly, for example bl654_usb, nrf52840dongle_nrf52840...

commit bb437e30 (2023-06-07): rp2040: Fix USB self powered for all boards

The USB configuration for all rp2040 boards wrongfully left
USB_SELF_POWERED as true by default. This commit fixes it for
all rp2040 boards.

Fixes #53993
PR #58815 (2023-06-07): rp2040: Fix USB self powered for all boards
The USB configuration for all rp2040 boards wrongfully left USB_SELF_POWERED as true by default. This commit fixes it for all rp2040 boards.

Fixes #53993
closed 2023-06-07 by carlescufi