zephyrproject-rtos/zephyr #53994: net: ethernet: Multicast receive packets statistics are not getting updated

cperera-aud, 2023-01-23:
**Describe the bug**
Multicast receive packets statistics is not getting updated in `ethernet_update_rx_stats` because the `pkt` argument passed into the function doesn't have the ethernet header anymore. In the `ethernet_recv` function just before `ethernet_update_rx_stats` gets called for the 2nd time, there is a `net_buf_pull` call which alters the `pkt->frags->data` pointer. After the alteration, the `pkt` pointer points to the data beyond the ethernet header. My suggestion is to change the `ethernet_update_rx_stats` function to accept the ethernet header instead of the pointer to the full `net_pkt` structure. That is, change

```
static void ethernet_update_rx_stats(struct net_if *iface,
				     struct net_pkt *pkt, size_t length)
```

to

```
static void ethernet_update_rx_stats(struct net_if *iface,
				     struct net_eth_hdr *hdr, size_t length)
```

The `ethernet_update_rx_stats` function only cares about the ethernet header so leaving out the `net_pkt` structure doesn't change its functionality. 

If this makes sense, feel free to assign the case to me and I'll create a PR for the changes. 

On a side note, when the zephyr protocol stack encounters unknown ethernet packet headers should it result in increasing the ethernet rx error count? On my test setup I have a Cisco SG300 PoE switch which generates Cisco Discovery packets which are not handled by the Zephyr protocol stack and the ethernet rx error count increases. The device that we're building periodically sends monitoring data to a PC based monitoring tool which logs the network utilisation and error counts from the device. The error counts on the device is going up because its dropping the unknown ethernet packet type. Would it be alright to just silently drop the packet without increasing the rx error count?

rlubos, 2023-01-23:
> My suggestion is to change the ethernet_update_rx_stats function to accept the ethernet header instead of the pointer to the full net_pkt structure. 

This sounds reasonable, it'd be appreciated if you could send a PR for that.

> On a side note, when the zephyr protocol stack encounters unknown ethernet packet headers should it result in increasing the ethernet rx error count? 

I agree that counting such packets as errors is a bit unfortunate. Perhaps we should introduce a separate counter for dropped packets into `struct net_stats_eth`? Something like `struct net_stats_pkts drop;`. Note though there is already `net_stats_t tx_dropped;` field, but it's barely used.

jukkar, 2023-01-23:
> On a side note, when the zephyr protocol stack encounters unknown ethernet packet headers should it result in increasing the ethernet rx error count?

I am just wondering how the network stack could differentiate between unknown Ethernet packet from malformed packet, to me both cases look the same from software point of view. If you want to fix this, just send PR that implements the unknown protocol support.

cperera-aud, 2023-01-25:
Thank you @rlubos and @jukkar for your feedback. I've made the fixes in this PR: https://github.com/zephyrproject-rtos/zephyr/pull/54083

PR #54083 (2023-01-25): net: ethernet: Fixes for ethernet stats reporting.
The following fixes were applied:
 - Multicast Rx packets stats were not recorded due not parsing the ethernet header. The function that tried to parse the ethernet header was parsing the ethernet packet beyond the ethernet header.
 - Added a new stats for unknown protocol which gets updated when the ethernet layer encounters an unknown ethernet packet type.

Fixes #53994
commit ce49beb2 (2023-01-25): net: ethernet: Fixes for ethernet stats reporting.

The following fixes were applied:
 - Multicast Rx packets stats were not recorded due not parsing the
ethernet header. The function that tried to parse the ethernet header
was parsing the ethernet packet beyond the ethernet header.
 - Added a new stats for unknown protocol which gets updated when the
ethernet layer encounters an unknown ethernet packet type.

Fixes #53994
closed 2023-01-25 by fabiobaltieri