zephyrproject-rtos/zephyr #53996: bt_conn_foerach() includes invalid connection while advertising

xudongzheng, 2023-01-23:
**Describe the bug**

When advertising is active (testing on Nordic nRF52840 Dongle), `bt_conn_foreach()` appears to include a connection that's not yet established with address matching `BT_ADDR_LE_NONE`.

**To Reproduce**

```
static void example_callback(struct bt_conn *conn, void *data) {
	const bt_addr_le_t *addr = bt_conn_get_dst(conn);
	if (!bt_addr_le_cmp(addr, BT_ADDR_LE_NONE)) {
		LOG_ERR("addr none");
	} else {
		LOG_INF("addr valid");
	}
}

...
	bt_conn_foreach(BT_CONN_TYPE_LE, example_callback, (void *) shell);
...
```

**Expected behavior**

It's not clear from the documentation whether this is the expected behavior or if there's is a missing check within `bt_conn_foerach()`.

**Impact**

Workaround involves checking the address in the callback.

hermabe, 2023-01-23:
Possible duplicate: https://github.com/zephyrproject-rtos/zephyr/issues/53254

jhedberg, 2023-01-25:
I think what's happening is that `bt_conn_foreach()` is iterating all connections with a non-zero reference count. When you advertise, a connection object gets pre-allocated so that it can be assigned to a potential new connection. At this point it has a non-zero reference count, but before a connection happens the destination address is not set (`BT_ADDR_LE_NONE`).

This is simply the current behaviour of `bt_conn_foreach()`. The documentation doesn't say more than that it iterates all "existing" connections, so the question becomes what does "existing" mean. "Any allocated connection object" sounds equally plausible as does "any connection object in CONNECTED state". I agree the latter might make more sense, and would indeed be an easy fix, but this is really more of an enhancement or feature request in that case, not a bug.

Thalley, 2023-01-25:
> but this is really more of an enhancement or feature request in that case, not a bug.

Agreed. The behavior is too undefined to specify this as a bug, rather than an enhancement.

hermabe, 2023-01-25:
As a workaround the application can use `bt_conn_get_info()` and check `bt_conn_info::state` to check if the connection is connected or not.

carlescufi, 2023-03-02:
> As a workaround the application can use `bt_conn_get_info()` and check `bt_conn_info::state` to check if the connection is connected or not.

Alternative approach: add a new `bt_conn_foreach_connected()` that does exactly that.

closed 2023-03-15 by fabiobaltieri