zephyrproject-rtos/zephyr #54014: usb: using Bluetooth HCI class in composite device leads to conflicts

desowin, 2023-01-23:
**Describe the bug**
Bluetooth Core Specification Version 5.3 Vol 4, Part B: USB Transport Layer, 2.1.1 Controller descriptors:
> The Controller configuration consists of two interfaces. The first interface has
no alternate settings and contains the bulk and interrupt endpoints. The second
interface provides scalable isochronous bandwidth. The recommended
configuration for the second interface has four alternate settings that provide
different bandwidth. The default interface is empty so that the device is capable
of scaling down to zero isochronous bandwidth.

Zephyr Bluetooth USB class https://github.com/zephyrproject-rtos/zephyr/blob/02ee85c89321081e172e4966bcbf6e531ecac8bb/subsys/usb/device/class/bluetooth.c#L53-L58 only implements the first interface whereas host drivers expect two interfaces.

[Linux kernel btusb](https://github.com/torvalds/linux/blob/2475bf0250dee99b477e0c56d7dc9d7ac3f04117/drivers/bluetooth/btusb.c#L4081-L4088) driver will claim interface next to the Bluetooth Programming Interface (Class: 0xE0 Subclass: 0x01 Protocol: 0x01) regardless of that interface Class/Subclass/Protocol.

Zephyr implementation does not cause much issue on Linux if Bluetooth class is the only class and/or is the last class appearing in CONFIGURATION descriptor. In both cases, the `usb_ifnum_to_if(data->udev, ifnum_base + 1);` will return `NULL` and therefore btusb won't claim the isochronous interface.

If Zephyr Bluetooth class implementation is used in composite device and Bluetooth is not the last interface, then Linux btusb driver will claim both the Bluetooth and one more interface, effectively preveting other composite function from working. Example include Bluetooth + DFU combination, where df-util will fail to claim interface because it is already claimed by btusb.

**To Reproduce**
Steps to reproduce the behavior:
1. Combine `samples/subsys/usb/dfu` with `samples/bluetooth/hci_usb` to form USB composite device
2. Try to use dfu-util to switch from DFU Runtime to DFU Mode Interface
3. Observe `dfu-util: Cannot claim interface` error

**Expected behavior**
Other class interfaces are not claimed by btusb driver and therefore DFU works as expected in Bluetooth+DFU composite device. 

**Impact**
Annoyance, but can be showstopper for people who need both DFU and Bluetooth.

**Logs and console output**
```
Opening DFU capable USB device...
Device ID 2fe3:000b
Run-Time device DFU version 0110
Claiming USB DFU (Run-Time) Interface...
dfu-util: Cannot claim interface 1: LIBUSB_ERROR_BUSY
```

**Environment (please complete the following information):**
 - OS: Linux 6.1.7
 - Toolchain: Zephyr SDK 0.15.2
 - Zephyr 10130d8e6de8c3889f42d498d3de6115db8f2d35

**Additional context**
The workaround is to add dummy interface without any endpoints at the end of `struct usb_bluetooth_config`. Such a dummy interface will then get claimed by Linux kernel, however the lack of isochronous endpoints in the interface will result in `hci0: invalid SCO descriptors` error.

Another workaround is to ensure that Bluetooth interface appears last (with highest `bInterfaceNumber`) in CONFIGURATION descriptor. In such case, Linux will fail to claim the non-existent interface.

A better solution would be to at least add interface with 2 empty (`wMaxPacketSize` = 0) isochronous endpoints. Ideally, full isochronous support would be added with alternate configurations with varying `wMaxPacketSize` values.

jfischer-no, 2023-01-23:
@Vudentz would it be possible to change btusb driver so that it does not claim subsequent interface if bInterfaceClass, bInterfaceSubClass, and bInterfaceProtocol do not match codes defined in Bluetooth spec?

desowin, 2023-01-23:
I cannot seem to find the hardcoded interface numbers (as referred to in Linux kernel source code) anywhere in Bluetooth Specification. Can it be that it used to that way and it got later removed from the specification? In such case, a patch to linux kernel would have a chance to get accepted.

honestech74, 2023-01-24:
@desowin @jfischer-no I have a workaround for the claimed interface by `btusb` on Linux(not on Zephyr).
```
sudo modprobe -r btusb
sudo dfu-util --alt 1 --download build/zephyr/zephyr.signed.bin
sudo modprobe -a btusb
```
So it will remove `btusb` driver temporarily on Linux while `dfu-util` is claiming its interface. It might be helpful temporarily to someone and hopefully fix this issue soon.

desowin, 2023-01-25:
Bluetooth specification since version 4.0 only recommends and no longer requires specific interface numbers.

While earlier Bluetooth versions, i.e. 2.1 + EDR and 3.0 + HS, contain required configuration table in Volume 4 - Host Controller Interface Part B - USB Transport Layer, Bluetooth Core Specification Addendum 2 changes the table from required to recommended configuration.

I tried to remove the hardcoded interface number requirement from Linux btusb driver, but it then failed with
```
[  216.670590] Bluetooth: hci0: Opcode 0x c03 failed: -32
```

It turned out that SETUP packet "20 00 00 00 00 00 03 00 03 0c 00" was sent and Zephyr device replied to it with STALL. This is because `bluetooth_class_handler` is only called when the request wIndex is equal to Interface Number, even if the request is directed to Device (and not Interface).

closed 2023-05-17 by kartben