zephyrproject-rtos/zephyr #54017: Manifest: Add qcbor for TF-M 1.7.0

microbuilder, 2023-01-23:
## Origin

https://github.com/laurencelundblade/QCBOR.git

## Purpose

TF-M relies on QCBOR for CBOR and COSE functionality, and currently downloads
QCBOR during builds, which is incompatible with Zephyr project policies. We require
a local fork of this repo under `zephyrproject-rtos` to comply with these build
requirements.

> **NOTE**: This repo will not be used directly by Zephyr, and is only required by 
TF-M during the managed secure firmware build process, to prevent downloading
the build requirement at compile time.

## Maintainership

@microbuilder, @joerchan

## Pull Request

https://github.com/zephyrproject-rtos/zephyr/pull/53899/commits/8df64b76d3a3db2e097ddfba15aeb93cb06d2a0f

## Revision

Version or SHA you would like to integrate initially

`b0e7033268e88c9f27146fa9a1415ef4c19ebaff`

## License

`BSD-3-Clause`

stephanosio, 2023-02-07:
Marking it as a release blocker since we are currently pulling the QCBOR component via an undocumented and unauthorised method, and we need some form of resolution on this (ideally, approve it for inclusion as a module).

For instance, building `tests/drivers/entropy/api/drivers.entropy.psa_crypto` for `bl5340_dvk_cpuapp_ns`:

```
-- The C compiler identification is GNU 12.1.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/toolchains/zephyr-sdk-0.15.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Populating qcbor
-- Configuring done
-- Generating done
-- Build files have been written to: /__w/zephyr/zephyr/twister-out/bl5340_dvk_cpuapp_ns/tests/drivers/entropy/api/drivers.entropy.psa_crypto/tfm/lib/ext/qcbor-subbuild
[1/9] Creating directories for 'qcbor-populate'
[1/9] Performing download step (git clone) for 'qcbor-populate'
Cloning into 'qcbor-src'...
remote: Enumerating objects: 3776, done.        
remote: Counting objects:   0% (1/642)        
remote: Counting objects:   1% (7/642)    
...
``` 

cc @nashif @laurenmurphyx64

galak, 2023-02-08:
So I hate to through a wrench into this, but qcbor isn't pure BSD-3-CLAUSE and thus wouldn't conform to Zephyr's guidelines.

galak, 2023-02-08:
https://github.com/laurencelundblade/QCBOR/blob/master/README.md?plain=1#L527

microbuilder, 2023-02-08:
@galak Terrible timing, but good catch, thanks for bringing this up.  

I've sent an email off to the author to see if something can be done quickly about a compliant license, but I'll look at how this can be isolated from TF-M builds as a fallback in case a last minute license change for something OSI-compliant isn't possible.

I believe only the attestation token code requires QCBOR, but I'll have to dig deeper to verify, and see how complicated it is to disable that in Zephyr.

microbuilder, 2023-02-09:
License diff, for reference sake:

[image]

@nashif What kind of time frame would be involved to ask the board for an exception here?

Presumably too long, since any exception vote request from the TSC will go to an email vote already (I suspect), then escalation to the board itself. The terms here aren't fundamentally different either.

I agree with @carlescufi (discussion on Discord) that ultimately, enabling support in t_cose -- which has it's [source included in TF-M](https://github.com/laurencelundblade/t_cose), and introduces the QCBOR dependency -- to optionally use zcbor as another backend in place of QCBOR likely solves this long term, and also gives us COSE support which we don't have today in zcbor, but that's not going to happen before the release. Discussion: https://discord.com/channels/720317445772017664/884102464985518110/1073182094525726750

Eugene (@quic-egmc) was also looking with Qualcomm legal at license changes, but it looks like this involves about 20 people, and co-ordinating signatures with all of them for a license change, which is neither quick nor guaranteed short term.

A license exception feels like the path of least resistance in the short term, given that the license terms don't seem that far off the beaten path either? That's unlikely in an acceptable time frame, though? (Also, obligatory IANAL!!!)

If an exception isn't feasible, I'll look at updating t_cose to try to pull QCBOR out in a reasonable manner, but still trying to scope how far the build dependency goes.

microbuilder, 2023-02-10:
@stephanosio This should solve any issues causing QCBOR to be downloaded in the test case you mentioned, as well as others: https://github.com/zephyrproject-rtos/zephyr/pull/54690

That set of PRs removes the ability to enable the initial attestation secure service in TF-M as part of a standard Zephyr build -- which is what was pulling in t_cose and then the problematic QCBOR depenency -- until we can properly sort out a license exception, or another solution for QCBOR.

closed 2023-02-10 by stephanosio

joerchan, 2023-02-11:
@microbuilder Should we have another issue to re-enable initial attestation service?
Now we've closed this as a temporary workaround for the release, and not as a proper solution. I suggest we make another issue with most of the information from this issue.

microbuilder, 2023-02-11:
> @microbuilder Should we have another issue to re-enable initial attestation service? Now we've closed this as a temporary workaround for the release, and not as a proper solution. I suggest we make another issue with most of the information from this issue.

Added: #54742