zephyrproject-rtos/zephyr #54023: compliance: constant comparison: false positives for c++ templates

cfriedt, 2023-01-23:
**Is your enhancement proposal related to a problem? Please describe.**
Currently, our compliance checker kind of flakes-out for C++ templates thinking that they are comparisons, which results in a lot of false positives (both the `<` and `>` are considered comparisons).

```cpp
#ifdef __cplusplus
template <typename T> static inline constexpr int foo(T x) { return -42; }
#endif
```

See https://github.com/zephyrproject-rtos/zephyr/actions/runs/3933421562/jobs/6727104410 

**Describe the solution you'd like**
Improved robustness for running compliance checks against C++ code, which makes fairly extensive use of `<` and `>` for purposes other than comparison.

**Describe alternatives you've considered**
Continuing on with this false positive madness 😱

Really though, it discourages people from reviewing code when they see the big red X, so should be fixed.

**Additional context**

voinov-dev, 2026-09-07:
Hi,

I'd like to work on this issue if nobody is already on it.

I re-checked the current scripts/checkpatch.pl against the example from this issue. The problem is still reproducible: C++ template angle brackets are treated as comparison operators and trigger SPACING false positives:

```
SPACING: spaces required around that '<' (ctx:WxV)
SPACING: spaces required around that '>' (ctx:VxW)
```
CONSTANT_COMPARISON no longer fires on this snippet after a37254140b0. That change was aimed at comparisons with constants on both sides, not at templates, so it only removed the warning as a side effect. The remaining SPACING errors still break compliance.

My plan:

- reproduce and document the remaining false positives for template <typename T> (and similar forms);
- add a focused regression case so this does not regress;
- fix the SPACING check so template angle brackets are not treated as comparisons, without weakening real comparison checks;
- open a draft PR for review.
Would it be okay if I take this?

voinov-dev, 2026-09-08:
Looked at this a bit more on current main.

The SPACING errors aren't limited to template <typename T>, any template argument list in a .h file triggers them, including nested ones and casts. Comparisons without spaces are still caught, which is good.

One correction to my earlier comment: CONSTANT_COMPARISON isn't actually gone, it just doesn't fire on the example from the report. Other template declarations still warn.

Also worth noting: checkpatch only runs the code checks on .c and .h, .cpp/.hpp are skipped entirely. Same for the clang-format check. So in practice this only hits C++ living in shared headers under #ifdef __cplusplus.

What I want to do: detect template argument lists and skip the angle brackets (and the */& inside them) in both the spacing and the constant comparison checks. Both template<T> and template <T> would pass; I don't want to add a new style rule here.

Two things I'd like an opinion on:

* Running checkpatch on .cpp/.hpp too — that's a bigger change, separate issue?

* There are no tests for checkpatch. Would a small pytest under scripts/tests/checkpatch/ be welcome, or is before/after output in the commit message enough?

keith-zephyr, 2026-09-09:
> Two things I'd like an opinion on:
> 
> * Running checkpatch on .cpp/.hpp too — that's a bigger change, separate issue?

Yes, adapting checkpatch to run against .cpp/.hpp files should be investigated as a separate issue. 

> * There are no tests for checkpatch. Would a small pytest under scripts/tests/checkpatch/ be welcome, or is before/after output in the commit message enough?

Unit tests for checkpatch would be very welcome.

closed 2026-09-18 by fabiobaltieri