zephyrproject-rtos/zephyr #54034: QSPI: Unable to build the project when introduced DMA into external flash interfacing   #DMA #QSPI #STM32H743

Shrini-007, 2023-01-24:
[image]

erwango, 2023-01-24:
@Shrini-007 Please rework your issue description. Having an image makes this quite hard to read (and potentially impossible for people with strong visual impairment)

Shrini-007, 2023-01-24:
Sorry, for inconvenience....So here it goes 
I am working on the external flash on QSPI interface.
I am able to read/write the external flash successfully by using interrupt method.
Now I am trying the same with DMA, but as soon as I build my project after enabling the DMA in .dts file I am getting this error.

Error Message start:
[221/287] Building C object zephyr/drivers/flash/CMakeFiles/drivers__flash.dir/flash_stm32_qspi.c.obj
**FAILED:** zephyr/drivers/flash/CMakeFiles/drivers__flash.dir/flash_stm32_qspi.c.obj
ccache /home/shri/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBUILD_VERSION=zephyr-v2.7.2 -DCORE_CM7 -DHSE_VALUE=25000000 -DKERNEL -DPB_MAX_REQUIRED_FIELDS=64 -DSTM32H743xx -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR_SUPERVISOR__ -D__ZEPHYR__=1 -I/home/shri/rcg2/zephyr/include -I/home/shri/rcg2/build/zephyr/include/generated -I/home/shri/rcg2/zephyr/soc/arm/st_stm32/stm32h7 -I/home/shri/rcg2/zephyr/lib/util/fnmatch/. -I/home/shri/rcg2/zephyr/drivers -I/home/shri/rcg2/zephyr/soc/arm/st_stm32/common -I/home/shri/rcg2/zephyr/subsys/settings/include -I/home/shri/rcg2/modules/hal/cmsis/CMSIS/Core/Include -I/home/shri/rcg2/modules/hal/stm32/stm32cube/stm32h7xx/soc -I/home/shri/rcg2/modules/hal/stm32/stm32cube/stm32h7xx/drivers/include -I/home/shri/rcg2/modules/hal/stm32/stm32cube/stm32h7xx/drivers/include/Legacy -I/home/shri/rcg2/modules/hal/stm32/stm32cube/common_ll/include -I/home/shri/rcg2/modules/lib/nanopb -I/home/shri/rcg2/modules/crypto/tinycrypt/lib/include -I/home/shri/rcg2/macaroon/coconut/zboss/include -I/home/shri/rcg2/macaroon/coconut/zboss/include/zcl -I/home/shri/rcg2/macaroon/coconut/zboss/include/ha -I/home/shri/rcg2/macaroon/coconut/zboss/include/se -I/home/shri/rcg2/macaroon/coconut/zboss_platform/osif/include -I/home/shri/rcg2/macaroon/coconut/. -I/home/shri/rcg2/macaroon/canary/. -I/home/shri/rcg2/macaroon/subsys/scene_module/include -I/home/shri/rcg2/macaroon/subsys/demand_response_module_client/include -I/home/shri/rcg2/macaroon/subsys/demand_response_module_server/include -I/home/shri/rcg2/macaroon/light_engine/include -I/home/shri/rcg2/macaroon/include -I/home/shri/rcg2/macaroon/. -I/home/shri/rcg2/rcg2/include -I/home/shri/rcg2/rcg2/drivers/can/can_mcp25xxfd/canfdspi -I/home/shri/rcg2/rcg2/drivers/can/can_mcp25xxfd/. -I/home/shri/rcg2/zephyr/drivers/can -isystem /home/shri/rcg2/zephyr/lib/libc/minimal/include -isystem /home/shri/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/include -isystem /home/shri/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/include-fixed -mno-unaligned-access -Os -imacros /home/shri/rcg2/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m7 -mthumb -mabi=aapcs -mfp16-format=ieee -imacros /home/shri/rcg2/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-address-of-packed-member -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/shri/rcg2/rcg2/app/area_hub=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/shri/rcg2/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/shri/rcg2=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -MD -MT zephyr/drivers/flash/CMakeFiles/drivers__flash.dir/flash_stm32_qspi.c.obj -MF zephyr/drivers/flash/CMakeFiles/drivers__flash.dir/flash_stm32_qspi.c.obj.d -o zephyr/drivers/flash/CMakeFiles/drivers__flash.dir/flash_stm32_qspi.c.obj -c /home/shri/rcg2/zephyr/drivers/flash/flash_stm32_qspi.c
/home/shri/rcg2/zephyr/drivers/flash/flash_stm32_qspi.c: In function 'flash_stm32_qspi_init':
/home/shri/rcg2/zephyr/drivers/flash/flash_stm32_qspi.c:863:18: warning: implicit declaration of function '__LL_DMA_GET_CHANNEL_INSTANCE'; did you mean 'IS_BDMA_CHANNEL_INSTANCE'? [-Wimplicit-function-declaration]
  863 |  hdma.Instance = __LL_DMA_GET_CHANNEL_INSTANCE(dev_data->dma.reg,
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                  IS_BDMA_CHANNEL_INSTANCE
/home/shri/rcg2/zephyr/drivers/flash/flash_stm32_qspi.c:863:16: warning: assignment to 'void *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
  863 |  hdma.Instance = __LL_DMA_GET_CHANNEL_INSTANCE(dev_data->dma.reg,
      |                ^
In file included from /home/shri/rcg2/modules/hal/stm32/stm32cube/stm32h7xx/drivers/include/stm32h7xx_hal_rcc.h:29,
                 from /home/shri/rcg2/modules/hal/stm32/stm32cube/stm32h7xx/drivers/include/stm32h7xx_hal_conf.h:246,
                 from /home/shri/rcg2/modules/hal/stm32/stm32cube/stm32h7xx/drivers/include/stm32h7xx_hal.h:30,
                 from /home/shri/rcg2/modules/hal/stm32/stm32cube/stm32h7xx/soc/stm32h7xx.h:223,
                 from /home/shri/rcg2/zephyr/soc/arm/st_stm32/stm32h7/soc.h:12,
                 from /home/shri/rcg2/zephyr/include/arch/arm/aarch32/cortex_m/cmsis.h:17,
                 from /home/shri/rcg2/zephyr/include/arch/arm/aarch32/mpu/arm_mpu_v7m.h:11,
                 from /home/shri/rcg2/zephyr/include/arch/arm/aarch32/mpu/arm_mpu.h:14,
                 from /home/shri/rcg2/zephyr/include/arch/arm/aarch32/arch.h:187,
                 from /home/shri/rcg2/zephyr/include/arch/cpu.h:19,
[... cut ...]
   66 |                               (__HANDLE__)->__PPP_DMA_FIELD__ = &(__DMA_HANDLE__); \
      |                                             ^~~~~~~~~~~~~~~~~
Error Message end.

Here are my .dts configurations for external flash. I have done this in proj.conf file CONFIG_DMA=y.
I have attached the Kconfig.stm32_qspi file also for zephyr version 2.7.2

[image]
[image]

I am not able understand what does this error mean….& what I am missing.

NOTE: I am using STM32H743 board for interfacing it with external flash & zephyr version is 2.7.2

erwango, 2023-01-24:
@Shrini-007 I'm converting this to discussion, since for now that's rather a question than a real bug.