zephyrproject-rtos/zephyr #54065: How To Change C++ Version Compilation Option For Freestanding Application?

nuertey, 2023-01-24:
Hello All,
I am developing a Freestanding C++ application in Zephyr RTOS and need to set the C++ version to C++14 or higher. How do I accomplish that?

I am trying to port the state machine library https://github.com/boost-ext/sml which is just a header file include that requires just C++14 support. It is a key component in my application.

Currently I am encountering the errors below which seems to indicate that I may not be compiling with "-std=c++14". Here are the particular parameters of my setup:

```
~/zephyrproject/zephyr$ git describe --tag 
zephyr-v3.2.0-2681-g9cf89603df

~/Zephyr_Freestanding_Applications/led_state_machine$ pwd
/home/.../Zephyr_Freestanding_Applications/led_state_machine
```

I am building so:

```
~/Zephyr_Freestanding_Applications/led_state_machine$ west -v build -p always -b nucleo_f207zg
```

prj.conf is like so:

```
CONFIG_GPIO=y

CONFIG_PRINTK=y
CONFIG_LOG=y

#CONFIG_CPP=y
CONFIG_CPLUSPLUS=y
#CONFIG_LIB_CPLUSPLUS=y
CONFIG_STD_CPP14=y
CONFIG_CPP_MAIN=y
CONFIG_MINIMAL_LIBC_MALLOC_ARENA_SIZE=128
#CONFIG_GLIBCXX_LIBCPP=y
```

Based upon some of the forum comments, I have tried several options in the CMakeLists.txt and it now looks like this, below. Perhaps it is because mine is a Freestanding app that is why none of the compiler switches take? 

I have also looked at the change request below, but it seems confusing and perhaps is not even applicable to my version of Zephyr RTOS hence my version tag above: 

https://github.com/zephyrproject-rtos/zephyr/pull/52946/files/ef932260ea813d40aa0c7b485a3d3de8c803f99a#top

```
cmake_minimum_required(VERSION 3.20.0)
find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})

project(led_state_machine LANGUAGES C CXX)
 
set(CMAKE_CXX_STANDARD 14)
set(CMAKE_CXX_STANDARD_REQUIRED True)
set(CMAKE_CXX_EXTENSIONS OFF)
set(CMAKE_BUILD_TYPE DEBUG)

set(CMAKE_C_FLAGS_DEBUG "${CMAKE_C_FLAGS_DEBUG} -std=c++14")
set(CMAKE_CXX_FLAGS_DEBUG "${CMAKE_CXX_FLAGS_DEBUG} -std=c++14")

#zephyr_library_compile_options(-std=c++14)

file(GLOB app_sources "src/*.c" "src/*.cpp")

target_sources(app PRIVATE ${app_sources})

target_compile_options(app PRIVATE -std=c++14)

#zephyr_library_compile_options(-std=c++14)

#set_target_properties(app PROPERTIES COMPILE_FLAGS "-std=c++14")
```

Anyway that I can set the compiler switch "-std=c++14" and not have it overridden by the west build system, including even just passing it to west or the correct way it ought to be done, kindly let me know.

So ultimately, attached is the error message that I observe. 

Regards. And thanks.
Nuertey

https://github.com/nuertey/RandomArtifacts/blob/master/zephyr_Cpluplus_freestanding_application_compile_error_message.txt

nuertey, 2023-01-24:
@krzysztof-jusiak

krzysztof-jusiak, 2023-01-24:
It seems like the errror points to the `cxa_guard_acquire` which requires some additional flags for the arm compiler (since C++11 static variables in the functions are thread safe by default)

I'd try if app is single threaded
```
 -fno-threadsafe-statics
```

or
```
-lstdc++
```
to avoid the undefined issue

nuertey, 2023-01-24:
Thank you so much @krzysztof-jusiak. It built with your suggestion. So you will be glad to know that your excellent state machine builds in Zephyr RTOS for STM32s and actually works. I am even actually using its DebugLogger feature per below, per your excellent instructions, so I can actually trace the state machine's progression. Thanks. Much appreciated:

```
// Just debug logging utilities for the State Machine defined according
// to Boost.SML's Domain-Specific Language (DSL) format:
struct DebugLogger_t
{
    template <class SM, class TEvent>
    void log_process_event(const TEvent&)
    {
        printf("[%s][process_event] %s\n", sml::aux::get_type_name<SM>(), sml::aux::get_type_name<TEvent>());
    }

    template <class SM, class TGuard, class TEvent>
    void log_guard(const TGuard&, const TEvent&, bool result)
    {
        printf("[%s][guard] %s %s %s\n", sml::aux::get_type_name<SM>(), sml::aux::get_type_name<TGuard>(),
               sml::aux::get_type_name<TEvent>(), (result ? "[OK]" : "[Reject]"));
    }

    template <class SM, class TAction, class TEvent>
    void log_action(const TAction&, const TEvent&)
    {
        printf("[%s][action] %s %s\n", sml::aux::get_type_name<SM>(), sml::aux::get_type_name<TAction>(),
               sml::aux::get_type_name<TEvent>());
    }

    template <class SM, class TSrcState, class TDstState>
    void log_state_change(const TSrcState& src, const TDstState& dst)
    {
        printf("[%s][transition] %s -> %s\n", sml::aux::get_type_name<SM>(), src.c_str(), dst.c_str());
    }
};
//...
```

And to help other folks and engineers struggling with same, here are the lines that I added to my CMakeLists.txt file. I added too many lines than probably needed as I was anxious to get it working quickly, but probably only one line is actually needed, probably the last one. Not sure. An expert in Zephyr RTOS backend tooling should chime in here to correct me for benefit of Posterity and other engineers:

CMakeLists.txt excerpt:
```
...
set(CMAKE_C_FLAGS_DEBUG "${CMAKE_C_FLAGS_DEBUG} -std=c++14 -fno-threadsafe-statics")
set(CMAKE_CXX_FLAGS_DEBUG "${CMAKE_CXX_FLAGS_DEBUG} -std=c++14 -fno-threadsafe-statics")

zephyr_library_compile_options(-std=c++14 -fno-threadsafe-statics)
...

# And at the end:
target_compile_options(app PRIVATE -std=c++14 -fno-threadsafe-statics)
```

Thanks
Nuertey

stephanosio, 2023-02-07:
Not sure why this was marked as bug, converting this to a discussion.