zephyrproject-rtos/zephyr #54085: USB MSC Sample does not work for native_posix over USBIP

katyo, 2023-01-25:
**Describe the bug**

I build and run USB Mass Storage sample for native_posix.
Sample starts well but after attaching via `usbip attach` it detaches right away.

**To Reproduce**

Steps to reproduce the behavior:
1. `west build -p auto -t run -b native_posix zephyr/samples/subsys/usb/mass`
2. `sudo usbip attach -r localhost -b 1-1`
4. USBIP client detaches

**Expected behavior**

A working Mass Storage device which does not detaches.

**Impact**

It does not work as expected.

**Logs and console output**

The start log is the following:
```
*** Booting Zephyr OS build zephyr-v3.2.0  ***
Mount /RAM:: 0
/RAM:: bsize = 512 ; frsize = 512 ; blocks = 158 ; bfree = 158
/RAM: opendir: 0
End of files
[00:00:00.060,000] <inf> main: The device is put in USB mass storage mode.
```

After attaching it prints the following:
```
[00:00:18.100,000] <err> native_posix_adapt: recv wrong length: 0
```

Linux dmesg log:
```
[1137724.163443] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
[1137724.163453] vhci_hcd vhci_hcd.0: devid(65538) speed(2) speed_str(full-speed)
[1137724.163471] vhci_hcd vhci_hcd.0: Device attached
[1137724.327678] vhci_hcd: vhci_device speed not set
[1137724.379933] usb 5-1: new full-speed USB device number 16 using vhci_hcd
[1137724.443938] vhci_hcd: vhci_device speed not set
[1137724.495671] usb 5-1: SetAddress Request (16) to port 0
[1137725.632852] usb 5-1: New USB device found, idVendor=2fe3, idProduct=0008, bcdDevice= 3.02
[1137725.632862] usb 5-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[1137725.632866] usb 5-1: Product: Zephyr MSC sample
[1137725.632869] usb 5-1: Manufacturer: ZEPHYR
[1137725.632872] usb 5-1: SerialNumber: 0123456789ABCDEF
[1137725.702791] usb-storage 5-1:1.0: USB Mass Storage device detected
[1137725.703194] scsi host5: usb-storage 5-1:1.0
[1137726.953759] vhci_hcd: cannot find a urb of seqnum 245 max seqnum 245
[1137726.953903] vhci_hcd: stop threads
[1137726.953910] vhci_hcd: release socket
[1137726.953916] vhci_hcd: disconnect device
[1137726.974667] vhci_hcd: vhci_device speed not set
[1137727.658020] usb 5-1: USB disconnect, device number 16
```

Captured USBIP traffic: [msc_usbip_capture.zip](https://github.com/zephyrproject-rtos/zephyr/files/10496832/msc_usbip_capture.zip)

**Environment (please complete the following information):**

 - OS: Linux 6.1
 - Toolchain: 0.15.2
 - Zephyr: 3.2.0

**Additional context**

Target: `native_posix`

katyo, 2023-01-25:
Full log with debug enabled:
```
[00:00:00.000,000] <dbg> usb_descriptor: usb_get_device_descriptor: __usb_descriptor_start 0x411540
[00:00:00.000,000] <dbg> usb_descriptor: usb_get_device_descriptor: __usb_descriptor_end 0x4115cc
[00:00:00.000,000] <dbg> usb_descriptor: usb_fix_descriptor: Configuration descriptor 0x411552
[00:00:00.000,000] <dbg> usb_descriptor: usb_fix_descriptor: Interface descriptor 0x41155b
[00:00:00.000,000] <dbg> usb_descriptor: usb_fix_descriptor: Endpoint descriptor 0x411564
[00:00:00.000,000] <dbg> native_posix: usb_dc_ep_check_cap: ep 81, mps 64, type 2
[00:00:00.000,000] <dbg> usb_descriptor: usb_validate_ep_cfg_data: Fixing EP address 82 -> 81
[00:00:00.000,000] <dbg> usb_descriptor: usb_validate_ep_cfg_data: endpoint 0x81
[00:00:00.000,000] <dbg> usb_descriptor: usb_fix_descriptor: Endpoint descriptor 0x41156b
[00:00:00.000,000] <dbg> native_posix: usb_dc_ep_check_cap: ep 1, mps 64, type 2
[00:00:00.000,000] <dbg> usb_descriptor: usb_validate_ep_cfg_data: Fixing EP address 1 -> 1
[00:00:00.000,000] <dbg> usb_descriptor: usb_validate_ep_cfg_data: endpoint 0x1
[00:00:00.000,000] <dbg> usb_descriptor: usb_fix_descriptor: Now the wTotalLength is 32
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: idx_max 11, ascii_idx_max 5, buf 0x411578
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char R : 52, idx 5 -> 11
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char Y : 59, idx 4 -> 9
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char H : 48, idx 3 -> 7
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char P : 50, idx 2 -> 5
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char E : 45, idx 1 -> 3
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char Z : 5a, idx 0 -> 1
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: idx_max 33, ascii_idx_max 16, buf 0x411586
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char e : 65, idx 16 -> 33
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char l : 6c, idx 15 -> 31
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char p : 70, idx 14 -> 29
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char m : 6d, idx 13 -> 27
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char a : 61, idx 12 -> 25
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char s : 73, idx 11 -> 23
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char   : 20, idx 10 -> 21
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char C : 43, idx 9 -> 19
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char S : 53, idx 8 -> 17
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char M : 4d, idx 7 -> 15
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char   : 20, idx 6 -> 13
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char r : 72, idx 5 -> 11
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char y : 79, idx 4 -> 9
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char h : 68, idx 3 -> 7
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char p : 70, idx 2 -> 5
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char e : 65, idx 1 -> 3
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char Z : 5a, idx 0 -> 1
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: idx_max 31, ascii_idx_max 15, buf 0x4115aa
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char F : 46, idx 15 -> 31
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char E : 45, idx 14 -> 29
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char D : 44, idx 13 -> 27
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char C : 43, idx 12 -> 25
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char B : 42, idx 11 -> 23
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char A : 41, idx 10 -> 21
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char 9 : 39, idx 9 -> 19
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char 8 : 38, idx 8 -> 17
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char 7 : 37, idx 7 -> 15
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char 6 : 36, idx 6 -> 13
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char 5 : 35, idx 5 -> 11
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char 4 : 34, idx 4 -> 9
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char 3 : 33, idx 3 -> 7
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char 2 : 32, idx 2 -> 5
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char 1 : 31, idx 1 -> 3
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char 0 : 30, idx 0 -> 1
*** Booting Zephyr OS build zephyr-v3.2.0  ***
[00:00:00.000,000] <inf> main: No file system selected
[00:00:00.000,000] <dbg> usb_device: usb_enable: lock usb_enable_lock mutex
[00:00:00.000,000] <dbg> native_posix: usb_dc_attach:
[00:00:00.000,000] <dbg> native_posix: thread_main:
[00:00:00.000,000] <dbg> native_posix_adapt: usbip_start: Starting
[00:00:00.000,000] <dbg> native_posix: usb_dc_ep_check_cap: ep 0, mps 64, type 0
[00:00:00.000,000] <dbg> native_posix: usb_dc_ep_check_cap: ep 80, mps 64, type 0
[00:00:00.000,000] <dbg> native_posix: usb_dc_ep_set_callback: ep 0 callback 0x405798
[... cut ...]
[00:00:21.450,000] <dbg> native_posix: usb_dc_ep_set_stall: ep 1
[00:00:21.450,000] <dbg> native_posix: usb_dc_ep_write: ep 80 len 0
[00:00:21.450,000] <dbg> native_posix: usbip_ep_is_enabled: ep 80
[00:00:21.450,000] <dbg> native_posix: usb_dc_ep_write: ep 81 len 13
[00:00:21.450,000] <dbg> native_posix: usbip_ep_is_enabled: ep 81
[00:00:21.560,000] <dbg> native_posix_adapt: usbip_start: Got cmd
[00:00:21.560,000] <err> native_posix_adapt: recv wrong length: 0
[00:00:21.560,000] <dbg> native_posix_adapt: usbip_start: Closing connection
```

Hunki, 2023-02-10:
It's not working also on v2.7.4

closed 2023-04-27 by github-actions[bot]