zephyrproject-rtos/zephyr #54086: test:rio:frdm_k64f: rio user_space test fails with zephyr-v3.2.0-3842-g7ffc20082023

hakehuang, 2023-01-25:
**Describe the bug**
rio user_space test fails

Please also mention any information which could help others to understand
the problem you're facing:
in fromd K64f rio test fails when enable usespace, tests/subsys/rtio/rtio_api,

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. west build -b frdm_k64f -- -DCONFIG_USERSPACE=y 
3. west flash

**Expected behavior**
TEST PASS

**Impact**
RIO in user speace

**Logs and console output**

```
*** Booting Zephyr OS build zephyr-v3.2.0-3842-g7ffc20082023 ***

    Running TESTSUITE rtio_api

    ===================================================================

    START - test_rtio_chain

    rtio chain simple

    starting one shot

    sqe ok callback

    starting one shot

    sqe ok callback

    [******** Truncated '237' Lines Log ********]

    producer acquiring

    producer acquiring

    producer acquiring

    producer acquiring

    producer acquiring

    producer yield

    consumer yield

    producer acquiring

    producer acquiring

    producer acquiring

    producer acquiring

    producer acquiring

    producer yield

    consumer yield

    producer acquiring

    producer acquiring

    producer acquiring

    producer acquiring

    producer acquiring

    producer yield

    consumer yield

    producer acquiring

    producer acquiring

    producer acquiring

    producer acquiring

    producer acquiring

    producer yield

    consumer yield

    producer acquiring

    producer acquiring

    producer acquiring

    producer acquiring

    producer acquiring

    producer yield

    consumer yield

    producer acquiring

    producer acquiring

    producer acquiring

    producer acquiring

    producer acquiring

    producer yield

    consumer yield

    producer acquiring

    producer acquiring

    producer acquiring

    producer acquiring

    producer acquiring

    producer yield

    consumer yield

    producer acquiring

    producer acquiring

    producer acquiring

    producer acquiring

    producer acquiring

    producer yield

    consumer yield

    producer acquiring

    producer E: ***** BUS FAULT *****

    E:   Stacking error

    E:   NXP MPU error, port 3

    E:     Mode: Supervisor, Data Address: 0x200015b8

    E:     Type: Write, Master: 0, Regions: 0x8100

    E: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x00000000

    E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x00000000

    E:  xpsr:  0x01000000

    E: Faulting instruction address (r15/pc): 0x0000b91a

    E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0

    E: Current thread: 0x20000278 (unknown)

    E: Halting system
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux,)
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: zephyr-v3.2.0-3842-g7ffc20082023

hakehuang, 2023-01-25:
@dleach02 @mmahadevan108

mmahadevan108, 2023-01-31:
@hakehuang . I am unable to reproduce this with `3.3 rc1`. Log below

```
*** Booting Zephyr OS build v3.3.0-rc1-39-g362874bc0991 ***
Running TESTSUITE rtio_api
===================================================================
START - test_rtio_chain
rtio chain simple
starting one shot
sqe ok callback
starting one shot
sqe ok callback
starting one shot
sqe ok callback
starting one shot
sqe ok callback
consume 0
consume 1
consume 2
consume 3
rtio chain concurrent
I: submit
I: head SQE in chain 0x20000180
I: setting up task 0
I: submitted sqe 0x20000180
I: tail SQE in chain 0x200001bc
I: resuming suspended task 0
starting one shot
sqe ok callback
starting one shot
sqe ok callback
starting one shot
sqe ok callback
starting one shot
sqe ok callback
I: sweeping oldest task 0
consume 0
consume 1
consume 2
consume 3
 PASS - test_rtio_chain in 0.114 seconds
===================================================================
START - test_rtio_multiple_chains
rtio multiple simple
calling submit from test case
starting one shot
waiting for 4 completions
waiting on completion 0
sqe ok callback
starting one shot
result 0, would block is -11, inval is -22
waiting on completion 1
sqe ok callback
starting one shot
result 0, would block is -11, inval is -22
waiting on completion 2
sqe ok callback
starting one shot
result 0, would block is -11, inval is -22
waiting on completion 3
sqe ok callback
result 0, would block is -11, inval is -22
rtio_multiple concurrent
calling submit from test case
I: submit
I: head SQE in chain 0x20000248
I: setting up task 0
I: submitted sqe 0x20000248
I: tail SQE in chain 0x2000025c
I: head SQE in chain 0x20000270
[... 219 lines cut ...]
producer acquiring
producer yield
 PASS - test_spsc_threaded in 0.230 seconds
===================================================================
TESTSUITE rtio_spsc succeeded

------ TESTSUITE SUMMARY START ------

SUITE PASS - 100.00% [rtio_api]: pass = 4, fail = 0, skip = 0, total = 4 duration = 0.352 seconds
 - PASS - [rtio_api.test_rtio_chain] duration = 0.114 seconds
 - PASS - [rtio_api.test_rtio_multiple_chains] duration = 0.123 seconds
 - PASS - [rtio_api.test_rtio_simple] duration = 0.054 seconds
 - PASS - [rtio_api.test_rtio_syscalls] duration = 0.061 seconds

SUITE PASS - 100.00% [rtio_spsc]: pass = 4, fail = 0, skip = 0, total = 4 duration = 0.233 seconds
 - PASS - [rtio_spsc.test_int_wrap_around] duration = 0.001 seconds
 - PASS - [rtio_spsc.test_produce_consume_size1] duration = 0.001 seconds
 - PASS - [rtio_spsc.test_produce_consume_wrap_around] duration = 0.001 seconds
 - PASS - [rtio_spsc.test_spsc_threaded] duration = 0.230 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION SUCCESSFUL
```

dleach02, 2023-01-31:
weird... I reproduce it with that SHA. In fact, I went back to 3.1 and it reproduced. It is a stack size problem.

PR #54291 (2023-02-02): boards: arm: frdm_k64f: Set TEST_EXTRA_STACK_SIZE
The RTIO API test case needs a little extra stack size when run on the frdm_k64f platform.

Fixes #54086
commit ca843964 (2023-02-02): boards: arm: frdm_k64f: Set TEST_EXTRA_STACK_SIZE

The RTIO API test case needs a little extra stack size when run
on the frdm_k64f platform.

Fixes #54086
closed 2023-02-02 by stephanosio