zephyrproject-rtos/zephyr #54092: ZCBOR code generator generates names not compatible with C++

kborowski3, 2023-01-25:
**Describe the bug**
There is a problem with compiling ZBOR code generated from .cddl files when we use it in the application written in C++. 
To be precise the problem occurs if we want to generate Union type. The generated code can be compiled in C but it is not working for C++ due to generated variables names conflicts. 

**To Reproduce**
Simple example showing the problem was created and it is available on my repository -> [https://github.com/kborowski3/zcbor_union_test](https://github.com/kborowski3/zcbor_union_test)

This is a simple application, which just includes the generated zcbor code. ZCBOR code is generated using `zcbor_union_test.cddl` which contains a simple union definition.

`zcbor_union_test.cddl` file content
```
Foo = int / bstr
```
where Foo is either an int or bstr.

The code is generated using ZCBOR code generator:
```
python3 ${ZCBOR_PATH}/zcbor/zcbor.py code -c zcbor_union_test.cddl --encode -t Foo --oc src/zcbor_union_test.c --oh src/zcbor_union_test.h
```

The problem is in the `zcbor_union_test.h` file. The content of the file was pasted below.
```[C]
/*
 * Generated using zcbor version 0.6.99
 * https://github.com/NordicSemiconductor/zcbor
 * Generated with a --default-max-qty of 3
 */

#ifndef ZCBOR_UNION_TEST_TYPES_H__
#define ZCBOR_UNION_TEST_TYPES_H__

#include <stdint.h>
#include <stdbool.h>
#include <stddef.h>
#include <string.h>
#include "zcbor_encode.h"

/** Which value for --default-max-qty this file was created with.
 *
 *  The define is used in the other generated file to do a build-time
 *  compatibility check.
 *
 *  See `zcbor --help` for more information about --default-max-qty
 */
#define DEFAULT_MAX_QTY 3

struct Foo_ {
	union {
		int32_t _Foo_int;
		struct zcbor_string _Foo_bstr;
	};
	enum {
		_Foo_int,
		_Foo_bstr,
	} _Foo_choice;
};

#endif /* ZCBOR_UNION_TEST_TYPES_H__ */
```

As you can see there is a problem with the enum's and union's  fields names. Both union and enum has a member `_Foo_int` and `_Foo_bstr`, which cannot be compiled in C++ code.

**Steps to reproduce the behavior:**
1. git clone https://github.com/kborowski3/zcbor_union_test.git
2. cd zcbor_union_test
3. west build -p always -b nucleo_wb55rg ./
4. See error

**Expected behavior**
The application compiles successfully. 

**Logs and console output**
```
/zcbor_union_test/src/zcbor_union_test_types.h:31:17: error: '_Foo_int' conflicts with a previous declaration
   31 |                 _Foo_int,
      |                 ^~~~~~~~
/zcbor_union_test/src/zcbor_union_test_types.h:27:25: note: previous declaration 'int32_t Foo_::<unnamed union>::_Foo_int'
   27 |                 int32_t _Foo_int;
      |                         ^~~~~~~~
/zcbor_union_test/src/zcbor_union_test_types.h:32:17: error: '_Foo_bstr' conflicts with a previous declaration
   32 |                 _Foo_bstr,
      |                 ^~~~~~~~~
/zcbor_union_test/src/zcbor_union_test_types.h:28:37: note: previous declaration 'zcbor_string Foo_::<unnamed union>::_Foo_bstr'
   28 |                 struct zcbor_string _Foo_bstr;
      |       
```
**Environment**
- Zephyr SDK version 0.15.1
- ZCBOR version 0.6.99

nordicjm, 2023-01-25:
This should really have been posted to https://github.com/NordicSemiconductor/zcbor

kborowski3, 2023-01-25:
Actually I can see that there are some C++ improvements commits but they are not available on [zephyr zcbor](https://github.com/zephyrproject-rtos/zcbor) forked repository. Can you bump there the newest version?

jgl-meta, 2023-06-20:
@oyvindronningstad this has been marked as stale 3 times. Is this an issue? If it isn't can you close this?

oyvindronningstad, 2023-06-22:
It is an issue. It is tracked by https://github.com/NordicSemiconductor/zcbor/issues/314.

r2r0, 2023-08-22:
@oyvindronningstad Any chance to upstream zcbor fix to zephyr?

r2r0, 2023-10-23:
@oyvindronningstad Ping

r2r0, 2024-01-03:
@oyvindronningstad If upstreaming is not possibel or planned then please close this issue.

closed 2024-02-23 by oyvindronningstad