zephyrproject-rtos/zephyr #54098: flash: shell: Add flash testing for read/write/erase speed output

nordicjm, 2023-01-25:
**Describe the solution you'd like**
Whilst investigating a bottleneck in flash access speeds using MCUmgr, it became apparent that it would be good to run tests on-device without need of an external interface other than the shell that can be used to run small tests on a flash device to determine the rough read, write and erase speeds of the device.

For read, this would just read data (without checking for accuracy of the data) at a specific offset, of a specific size, for a set number of runs -> speed would show `total time / number of runs`

For write, this would just write data at a specific offset, of a specific size, for a set number of runs -> speed would show `total time / number of runs`

For erase, this would just erase a specified number of sectors at a specific offset for a set number of runs -> speed would show `total time / number of runs`

**Describe alternatives you've considered**
None

Laczen, 2023-01-25:
@nordicjm, what would be the meaning of these numbers ? Are they device (not board, but real physical device) dependent ? Is one offset sufficient ? Is it dependent on other parts being enabled (e.g. radio) ? ...

It seems like a meaningless test with a possible bad effect on device lifetime. 

So -1 from my side

nordicjm, 2023-01-25:
It's not something you would ever leave on for a production build, I've already implemented it and used it, for an external QSPI erasing and writing takes about 2.2 seconds, doing the same test from the filesystem layer takes about 2.7 seconds, so that's proven to be very valuable given that doing the same test from a remote device, using MCUmgr over bluetooth, takes 8 seconds. And that would be the sole purpose, to test the read/write/erase speed of a flash device to determine where throughput issues are in applications

Laczen, 2023-01-25:
@nordicjm, I don't understand the test you are describing for the external QSPI device.

nordicjm, 2023-01-25:
An example - flash driver test:

```
uart:~$ flash erase_write_test mx25r6435f@0 0x0 0x4000 4
Loop #1 done in 243 ticks.
Loop #2 done in 242 ticks.
Loop #3 done in 242 ticks.
Loop #4 done in 242 ticks.
Total tick count: 969
Tick per loop: 242
uart:~$ flash erase_write_test flash-controller@4001e000 0xf8000 0x4000 4
Loop #1 done in 1028 ticks.
Loop #2 done in 1028 ticks.
Loop #3 done in 1022 ticks.
Loop #4 done in 1028 ticks.
Total tick count: 4106
Tick per loop: 1026
```

File system test on QSPI:

```
uart:~$ fs erase_write_test /lfs1/testfile 0x20000 1
Loop #0 done in 2595 ticks.
Total tick count: 2595
Tick per loop: 2595
```

So if doing a file upload over MCUmgr takes 7.4 seconds to the same device, that is a 4.8 second overhead - from "somewhere"

nordicjm, 2023-01-25:
Bluetooth was not used at all for this test, the UART was. The test shows there is a unexplained very large overhead of 4 seconds (rounding down as assuming a bit for going up and down layers and UART) that at the moment cannot be accounted for. The flash tests shows that the issue was not that the maximum throughput of the flash device was the cause - if I did not create the flash test, how would it be known that there was an issue at all?

Laczen, 2023-01-25:
> Bluetooth was not used at all for this test, the UART was. The test shows there is a unexplained very large overhead of 4 seconds (rounding down as assuming a bit for going up and down layers and [...]

By testing what needed to be tested, the communication layer. This should be tested by introducing a stub flash device that allows you to exclude the flash subsystem from the equation.

PR #54130 (2023-02-20): flash shell and fs shell: Add read/erase/write test commands
    Adds commands which can be used for timing flash device tasks and
    outputting a rough speed.

    Adds an optional shell command for testing the speed of the flash
    device and file system, which outputs a rough speed.

Fixes #54098
closed 2023-02-20 by fabiobaltieri