zephyrproject-rtos/zephyr #54101: bluetooth: shell: Lots of checks of type (unsigned < 0) which is bogus

finikorg, 2023-01-25:
Unsigned cannot be less then zero so there is no need to check for it.

Example of checks:
https://github.com/zephyrproject-rtos/zephyr/blob/cf871aec645dc45503b20c2e7684a4babbc436dc/subsys/bluetooth/shell/bap_broadcast_assistant.c#L315-L316
 or
https://github.com/zephyrproject-rtos/zephyr/blob/cf871aec645dc45503b20c2e7684a4babbc436dc/subsys/bluetooth/shell/bap_broadcast_assistant.c#L321-L322

etc

Thalley, 2023-01-25:
@finikorg Do you have a way to generate the full list? 

It's quite an effort if someone manually have to go through each `strtoX` in the shell modules to verify every single parameter.

finikorg, 2023-01-25:
No, I cannot, but I think it does make sense to check them all, there are other bugs like checking (uint32 > UINT32_MAX), etc

Thalley, 2023-01-27:
Fixing all LE Audio (and ISO) shell `strtoX` in https://github.com/zephyrproject-rtos/zephyr/pull/54165

Thalley, 2023-01-29:
@finikorg I think I fixed all the bugs in the LE audio files. Please review :)

Thalley, 2023-02-20:
@finikorg Can you verify if there are still some missing in the non-LE Audio files?

finikorg, 2023-02-20:
> Can you verify if there are still some missing in the non-LE Audio files?

Thanks. I think all fixed, I let you know if there are some bugs left.