zephyrproject-rtos/zephyr #54104: Bluetooth: Host: Bonding information distribution in the non-bondable mode

kapi-no, 2023-01-25:
**Describe the bug**
The Zephyr Bluetooth Host (SMP module) seems to violate the Bluetooth specification in the following pairing corner case. With the Bonding Flag (BF) cleared in the Pairing Response, the stack should set the key distribution flags to zero for the key distribution initiator and the responder. In this case, the specification requires no bonding information distribution (IRK, Identity Address, LTK, etc.).

This requirement is based on the 9.4.2 section, Part C, Vol 3 of the Bluetooth Core specification v5.3:

> 
> 9.4.2 Non-bondable mode
>  
> 9.4.2.1 Description
> A device in the non-bondable mode does not allow a bond to be created with a peer device.
>  
> 9.4.2.2 Conditions
> If a device does not support pairing as defined in the Security Manager section then it is considered to be in non-bondable mode. If Security Manager pairing is supported, the Host shall set the Bonding_Flags to ‘No Bonding’ as defined in [Vol 3] Part H, Section 3.5.1 and bonding information shall not be exchanged or stored.

**To Reproduce**
Set the non-bondable mode for the Bluetooth Peripheral device using the following configuration ```CONFIG_BT_BONDABLE=n``` and pair with the Central device (e.g. Android device) that requests bonding in the Pairing Request. Observe in the sniffer trace or HCI logs that bonding information is exchanged between devices.

**Expected behavior**
Bonding information shall not be exchanged in the non-bondable mode.

kruithofa, 2023-02-02:
@seko-nordic do you have any thoughts?

seko-nordic, 2023-02-02:
For me it looks like we violating the spec requirement (shall). The fact that there is currently no conformance tests for this specific requirement does not mean they will not appear later. From IOP perspective I think it would be better to clear key distribution flags in that case, preventing peer from taking ambiguous decision to accept / distribute keys in Non-bondable mode.

jhedberg, 2023-02-02:
@seko-nordic does it make any difference if we're doing Secure Connections pairing or legacy LE pairing? For SC, some of the flags are about generation, not distribution, which wouldn't violate the word of the spec.

seko-nordic, 2023-02-03:
@jhedberg It should not make any difference. If you are thinking about LTK in LESC, it is always generated and the spec explicitly prohibit to distribute it, see
3.6.1 Key distribution and generation
BLUETOOTH CORE SPECIFICATION Version 5.3 | Vol 3, Part H page 1611
In LE Secure Connections pairing, when SMP is running on the LE
transport, then the EncKey field shall be ignored. EDIV and Rand shall be
set to zero and shall not be distributed.

jhedberg, 2023-02-03:
I've submitted PR #54429 as an attempt to fix this, but I don't have the possibility of running any actual tests with this change, so someone else needs to test it. The effect of that PR should be that we set both initiator and responder distribution bits to zero whenever either side has indicated that it does not intend to bond.

PR #54429 (2023-02-08): Bluetooth: SMP: Fix setting key distribution bits for no-bonding
9.4.2 section, Part C, Vol 3 of the Bluetooth Core specification v5.3 states the following in the case that the device is in non-bondable mode:

If Security Manager pairing is supported, the Host shall set the Bonding_Flags to ‘No Bonding’ as defined in [Vol 3] Part H, Section 3.5.1 and bonding information shall not be exchanged or stored.

Fixes #54104
commit 9c459b33 (2023-02-08): Bluetooth: SMP: Fix setting key distribution bits for no-bonding

9.4.2 section, Part C, Vol 3 of the Bluetooth Core specification v5.3
states the following in the case that the device is in non-bondable
mode:

If Security Manager pairing is supported, the Host shall set the
Bonding_Flags to ‘No Bonding’ as defined in [Vol 3] Part H, Section
3.5.1 and bonding information shall not be exchanged or stored.

Fixes #54104
closed 2023-02-08 by stephanosio