zephyrproject-rtos/zephyr #54117: West flash fails on upload to Nucleo F303K8

moritz-geier, 2023-01-25:
## Bug Description
Flashing of the Nucleo F303K8 on Windows works but under Linux(Ubuntu) fails. I also tested the Nucleo F466RE which works on both systems.

```(.venv) moritz-geier@Galactus:~/Documents/zephyrproject/zephyr$ west flash
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner openocd
-- runners.openocd: Flashing file: /home/moritz-geier/Documents/zephyrproject/zephyr/build/zephyr/zephyr.hex
Open On-Chip Debugger 0.11.0+dev-00725-gc5c47943d (2022-11-22-06:32)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : DEPRECATED target event trace-config; use TPIU events {pre,post}-{enable,disable}
Info : clock speed 1000 kHz
Info : STLINK V2J40M27 (API v2) VID:PID 0483:374B
Info : Target voltage: 3.278628
Warn : target stm32f3x.cpu examination failed
Error: jtag status contains invalid mode value - communication failure
Polling target stm32f3x.cpu failed, trying to reexamine
Examination failed, GDB will be halted. Polling again in 100ms
Info : Previous state query failed, trying to reconnect
Error: jtag status contains invalid mode value - communication failure
Polling target stm32f3x.cpu failed, trying to reexamine
Examination failed, GDB will be halted. Polling again in 300ms
Info : starting gdb server for stm32f3x.cpu on 3333
Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* stm32f3x.cpu       hla_target little stm32f3x.cpu       unknown

Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Error: read_memory: read at 0xe0042004 with width=32 and count=1 failed
Error executing event examine-end on target stm32f3x.cpu:
/home/moritz-geier/zephyr-sdk-0.15.2/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/mem_helper.tcl:5: Error: read_memory: failed to read memory
in procedure 'ocd_process_reset' 
in procedure 'ocd_process_reset_inner' called at file "embedded:startup.tcl", line 876
in procedure 'stm32f3x_default_examine_end' called at file "/home/moritz-geier/zephyr-sdk-0.15.2/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/target/stm32f3x.cfg", line 100
in procedure 'mmw' called at file "/home/moritz-geier/zephyr-sdk-0.15.2/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/target/stm32f3x.cfg", line 81
in procedure 'mrw' called at file "/home/moritz-geier/zephyr-sdk-0.15.2/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/mem_helper.tcl", line 30
at file "/home/moritz-geier/zephyr-sdk-0.15.2/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/mem_helper.tcl", line 5
Info : Previous state query failed, trying to reconnect
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08001058 msp: 0x20000fa8
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
Polling target stm32f3x.cpu failed, trying to reexamine
Info : [stm32f3x.cpu] Cortex-M4 r0p1 processor detected
Info : [stm32f3x.cpu] target has 6 breakpoints, 4 watchpoints
Info : device id = 0x10016438
Info : flash size = 64kbytes
Warn : Adding extra erase range, 0x080036dc .. 0x080037ff
Error: error writing to flash at address 0x08000000 at offset 0x00000000
auto erase enabled
FATAL ERROR: command exited with status 1: /home/moritz-geier/zephyr-sdk-0.15.2/sysroots/x86_64-pokysdk-linux/usr/bin/openocd -s /home/moritz-geier/Documents/zephyrproject/zephyr/boards/arm/nucleo_f303k8/support -s /home/moritz-geier/zephyr-sdk-0.15.2/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts -f /home/moritz-geier/Documents/zephyrproject/zephyr/boards/arm/nucleo_f303k8/support/openocd.cfg '-c init' '-c targets' -c 'reset init' -c 'flash write_image erase /home/moritz-geier/Documents/zephyrproject/zephyr/build/zephyr/zephyr.hex' -c 'reset init' -c 'verify_image /home/moritz-geier/Documents/zephyrproject/zephyr/build/zephyr/zephyr.hex' -c 'reset run' -c shutdown```

## To Reproduce
Steps to reproduce the behavior:

west build -p auto -b nucleo_f303k8 samples/basic/blinky
west flash
Expected behavior
To be able to flash the Nucleo F303K8 on Ubuntu with west flash

I## mpact
annoyance

## Environment (please complete the following information):
Windows:
- OS: Windows Build 19044
- Toolchain Zephyr SDK 15.2
- openocd 0.11.0 via choco

Ubuntu:
- OS: Ubuntu 22.04
- Toolchain Zephyr SDK 15.2

Firmware on ST-Links are up-to date

FRASTM, 2023-01-26:
Could it be a pb with SDK ? No flashing error seen with my zephyr-sdk-0.15.0

FRASTM, 2023-01-30:
I suggest to add the pyocd runner for the boards to flash and debug :
```
diff --git a/boards/arm/nucleo_f303k8/board.cmake b/boards/arm/nucleo_f303k8/board.cmake
index f061108044..294493a928 100644
--- a/boards/arm/nucleo_f303k8/board.cmake
+++ b/boards/arm/nucleo_f303k8/board.cmake
@@ -1,6 +1,10 @@
 # SPDX-License-Identifier: Apache-2.0
 
 board_runner_args(jlink "--device=STM32F303K8" "--speed=4000")
+board_runner_args(pyocd "--target=stm32f303k8tx")
+board_runner_args(pyocd "--flash-opt=-O reset_type=hw")
+board_runner_args(pyocd "--flash-opt=-O connect_mode=under-reset")
 
+include(${ZEPHYR_BASE}/boards/common/pyocd.board.cmake)
 include(${ZEPHYR_BASE}/boards/common/openocd.board.cmake)
 include(${ZEPHYR_BASE}/boards/common/jlink.board.cmake)
```

If the stm32f303k8tx is not in the list of targets for pyocd : 
`$ pyocd list --targets`
then add the pack for pyocd
`$ pyocd pack install stm32f303`

closed 2023-02-02 by fabiobaltieri