zephyrproject-rtos/zephyr #54121: Intel CAVS: tests/subsys/zbus/user_data fails

laurenmurphyx64, 2023-01-26:
**Describe the bug**
On CAVS 15, 18 and 25 tests/subsys/zbus/user_data/user_data.channel_user_data.user_data_regression fails.

**To Reproduce**
Build and run test on CAVS 15, 18 or 25. Twister options given in our CI:

```
-W --hardware-map /home/ztest/cavs/cavs.map
--device-testing
-x=CONFIG_BOOT_DELAY=500
-v --no-skipped-report
```

**Expected behavior**
Passes.

**Impact**
Annoyance? Not sure.

**Logs and console output**
```
--
***** delaying boot 500ms (per build configuration) *****
*** Booting Zephyr OS build zephyr-v3.2.0-3949-gfae717305bc9 (delayed boot 500ms) ***
Running TESTSUITE user_data
===================================================================
START - test_channel_user_data
PASS - test_channel_user_data in 0.001 seconds
===================================================================
START - test_user_data_regression
Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/zbus/user_data/src/main.c:106: user_data_test_user_data_regression: *count not equal to 2
FAIL - test_user_data_regression in 1.001 seconds
===================================================================
TESTSUITE user_data failed.
------ TESTSUITE SUMMARY START ------
SUITE FAIL -  50.00% [user_data]: pass = 1, fail = 1, skip = 0, total = 2 duration = 1.002 seconds
- PASS - [user_data.test_channel_user_data] duration = 0.001 seconds
- FAIL - [user_data.test_user_data_regression] duration = 1.001 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: a0708fa32149c424ca397918000f7969
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Latest main (fae717305bc9)

stephanosio, 2023-01-31:
@rodrigopex I just noticed that "Zbus" area is missing in the MAINTAINERS.yml. Can you please send a PR to add that?

rodrigopex, 2023-01-31:
> @rodrigopex I just noticed that "Zbus" area is missing in the MAINTAINERS.yml. Can you please send a PR to add that?

Sure, I will.

rodrigopex, 2023-01-31:
@laurenmurphyx64 I will try to reproduce it here and let you know what it could be. Ok?

teburd, 2023-01-31:
Some hints here...

Sharing a stack value by reference across threads can/will cause failures on cAVS. So if you know of something that does that here, that would likely be the cause. Stacks should be viewed as thread local information in Zephyr really but I don't think this is documented anywhere.

rodrigopex, 2023-01-31:
> Sharing a stack value by reference across threads can/will cause failures on cAVS.

This is precisely the case here. The user data is a channel's pointer to an external user data portion. @teburd  Do you think it is worth trying, or is there no way to fix this on the platform? If so, I would remove the support to the platform for this sample and add some information to the documentation.

teburd, 2023-05-11:
The platform will never fix sharing stack data as that is a cached memory region and this is a cache incoherent platform.

The solution is to use global statics for sharing as those will be placed into uncached memory.

rodrigopex, 2023-05-11:
I need help reproducing the steps you described @laurenmurphyx64. Could you help me? If so, talk to me on Discord (RODRIGOPEX).

The twister test with ```twister -T tests/subsys/zbus/ -p intel_adsp_cavs15``` worked, but it only builds.

[image]

teburd, 2023-05-11:
@rodrigopex sadly I don't think there's a way you could test the fix, but if you are kind enough to change any shared data to static data and link a patch or branch I'm very happy to test on a board

rodrigopex, 2023-05-11:
@teburd The zbus communication is fundamentally based on shared memory. The channel is a shared memory portion itself. I am reviewing the code to see what we can switch from shared to static. Sadly the channel cannot be static. 

If you look at the [code](https://github.com/zephyrproject-rtos/zephyr/blame/main/tests/subsys/zbus/user_data/src/main.c#L24-L33), the `user_data` variable ```static int my_user_data;``` is currently _static_.

rodrigopex, 2023-05-12:
@SzymonRichert @teburd  Is there a way of sharing memory without using stack (and heap) within your platform?

Currently, the channels are shared on the stack, and I can retrieve them using `extern`. However, if I put them on static memory portions, I cannot use `extern` to retrieve them on other files (got it?). 

I don't have a possible solution in mind. Do you have one? Using the heap would be bad, because of how things are structured in zbus' code. So, iterable sections could help. Any clue?

closed 2023-07-27 by github-actions[bot]