zephyrproject-rtos/zephyr #54123: Intel CAVS 25: tests/boards/intel_adsp/ssp fails

laurenmurphyx64, 2023-01-26:
**Describe the bug**
tests/boards/intel_adsp/ssp/boards.intel_adsp.ssp.adsp_ssp_transfer fails with a fatal exception.

**To Reproduce**
Build and run test on CAVS 25. Twister options given in our CI:

```
-W --hardware-map /home/ztest/cavs/cavs.map
--device-testing
-x=CONFIG_BOOT_DELAY=500
-v --no-skipped-report
```

**Expected behavior**
Passes.

**Logs and console output**
```
INFO:cavs-client:Receive: start_log
INFO:cavs-client:Start to monitor log output...
--
I: intel_adsp_gpdma_init: dma dma@7d000 initialized
I: intel_adsp_gpdma_init: dma dma@7c000 initialized
***** delaying boot 500ms (per build configuration) *****
*** Booting Zephyr OS build zephyr-v3.2.0-3906-gc348fb51c9d5 (delayed boot 500ms) ***
Running TESTSUITE adsp_ssp
===================================================================
I: dai_ssp_pm_runtime_en_ssp_power en_ssp_power index 0
I: dai_ssp_pm_runtime_en_ssp_power I2SLCTL
I: dai_ssp_update_bits base 77000, reg 8, mask 80, value 80
START - test_adsp_ssp_config_set
I: dai_ssp_log_ssp_data dai index: 0
I: dai_ssp_log_ssp_data plat_data base: 487424
I: dai_ssp_log_ssp_data plat_data irq: 0
I: dai_ssp_log_ssp_data plat_data fifo playback offset: 487440
I: dai_ssp_log_ssp_data plat_data fifo playback handshake: 2
I: dai_ssp_log_ssp_data plat_data fifo capture offset: 487440
I: dai_ssp_log_ssp_data plat_data fifo capture handshake: 3
I: dai_ssp_set_config_tplg
I: dai_ssp_set_config_tplg going for loopback!
I: dai_ssp_set_config_tplg sscr0 = 0xc1d0003f, sscr1 = 0xd0400004, ssto = 0x00000000, sspsp = 0x2200000
I: dai_ssp_set_config_tplg sscr2 = 0x00004002, sspsp2 = 0x00000000, sscr3 = 0x07070000, ssioc = 0x00000020
I: dai_ssp_set_config_tplg ssrsa = 0x00000003, sstsa = 0x00000003
PASS - test_adsp_ssp_config_set in 0.002 seconds
===================================================================
START - test_adsp_ssp_transfer
I: dai_ssp_get_properties dai_index 0
I: dai_ssp_get_properties fifo 487440
I: dai_ssp_get_properties handshake 2
I: dai_ssp_get_properties init delay 0
dma block 0 block_size 48, source addr 9e023aa0, dest addr 77010
set next block pointer to 0x9e02a690
dma block 1 block_size 48, source addr 9e023aa0, dest addr 77010
Configuring the dma tx transfer on channel 0
I: dai_ssp_get_properties dai_index 0
I: dai_ssp_get_properties fifo 487440
I: dai_ssp_get_properties handshake 3
I: dai_ssp_get_properties init delay 0
dma block 0 block_size 48, source addr 77010, dest addr 9e02a000
set next block pointer to 0x9e02a634
dma block 1 block_size 48, source addr 77010, dest addr 9e02a0c0
Configuring the dma rx transfer on channel 1
Starting the transfer on channels 0 and 1 and waiting completion
I: dai_ssp_pre_start
I: dai_ssp_mn_set_mclk mclk_rate 24576000, mclk_source_clock 0
I: dai_ssp_set_mclk_divider mclk_id 0 mdivr_val 1
I: dai_ssp_find_mn for freq 24576000 bclk 3072000
I: dai_ssp_mn_set_bclk bclk_rate 3072000, *out_scr_div 8, m 1, n 1
I: dai_ssp_bclk_prepare_enable sscr0 = 0xc1d0077f
I: dai_ssp_update_bits base 77000, reg 4, mask 300000, value 300000
I: dai_ssp_update_bits base 77000, reg 0, mask 80, value 80
I: dai_ssp_early_start SSE set for SSP0
I: dai_ssp_pre_start
I: dai_ssp_update_bits base 77000, reg 4, mask 300000, value 300000
I: dai_ssp_update_bits base 77000, reg 0, mask 80, value 80
I: dai_ssp_early_start SSE set for SSP0
I: dai_ssp_start
I: dai_ssp_update_bits base 77000, reg 34, mask 100, value 100
I: dai_ssp_start
I: dai_ssp_update_bits base 77000, reg 30, mask 100, value 100
rx giving xfer_sem
E:  ** FATAL EXCEPTION
E:  ** CPU 0 EXCCAUSE 9 (load/store alignment)
E:  **  PC 0xbe019d2a VADDR 0x9e025581
E:  **  PS 0x60220
E:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:2 CALLINC:2)
E:  **  A0 0xbe019d5c  SP 0xbe0263f0  A2 0x9e0255e8  A3 0x9e02a180
E:  **  A4 0x7e0133e9  A5 0xbe026500  A6 0xbe026500  A7 (nil)
E:  **  A8 (nil)  A9 0x9e02557d A10 0x9e0255e8 A11 0x60225
E:  ** A12 (nil) A13 0x1 A14 0xbe01e088 A15 (nil)
E:  ** LBEG (nil) LEND (nil) LCOUNT (nil)
E:  ** SAR 0x1f
Backtrace:0xbe019d27:0xbe0263f0 0xbe019d59:0xbe026410 0xbe019eac:0xbe026430 0xbe019421:0xbe026450 0xbe011195:0xbe026470 0xbe014e51:0xbe026490 0xbe01447d:0xbe0264c0 0xbe013765:0xbe0264e0 0x7e0133f5:0xbe026500  |<-CORRUPTED
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x9e02a220 (idle 00)
E: Halting system
```

**Environment**
 - Linux
 - Zephyr SDK
 - c348fb51c9d5

teburd, 2023-01-31:
@juimonen any ideas? This is your area of expertise, at first glance... unaligned access suggests a pointer cast from a byte array perhaps?

juimonen, 2023-01-31:
@teburd ok I'll try if I can figure out something... is this new or has this test been passing before?

juimonen, 2023-02-06:
fix here https://github.com/zephyrproject-rtos/zephyr/pull/54513

@teburd I'm not totally sure about the fix, it works for me, but not sure has there been some change in dma config etc. as it used to work for me before.

teburd, 2023-02-06:
> fix here #54513
> 
> @teburd I'm not totally sure about the fix, it works for me, but not sure has there been some change in dma config etc. as it used to work for me before.

Nothing in the DMA config should be effecting this as far as I'm aware no. Unclear to me why this fixes the problem exactly, but a fix is a fix.

closed 2023-02-07 by laurenmurphyx64