zephyrproject-rtos/zephyr #54138: Buetooth: shell: `bt adv-data` isn't working properly

theob-pro, 2023-01-26:
`bt adv-data` is not working, the following commands should set the advertiser name to `Nordic1-thb1` but does not.

```
bt init
bt adv-create conn-scan
bt adv-data 0d094e6f72646963312d74686231
bt adv-start
```

The `adv-data` command was working ~1 year ago, it look like some recent refactoring broke it.

Thalley, 2023-01-26:
> Could you have look at it @Thalley please.

I'm sorry, but this is outside of the domain (LE Audio) that I am working on. Hopefully someone else can take a look at it. 

@theob-pro please also follow the Bug template when raising a bug: https://github.com/zephyrproject-rtos/zephyr/issues/new?assignees=&labels=bug&template=bug_report.md&title=

jori-nordic, 2023-01-26:
This seems to have been broken by you @Thalley or by @MariuszSkamra :
we narrowed it down to [an uninitialized var](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/shell/bt.c#L1657) and the new `ad_init` fn not working properly.

Thalley, 2023-01-26:
> This seems to have been broken by you @Thalley or by @MariuszSkamra : we narrowed it down to [an uninitialized var](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/shell/bt.c#L1657) and the new `ad_init` fn not working properly.

I see. Thanks for letting me know. I'll take a look at it then :)

closed 2023-01-30 by fabiobaltieri