zephyrproject-rtos/zephyr #54139: Bluetooth: Audio: race hazard bt_audio_discover() callback vs unicast_client_ase_cp_discover()

Chris-StJohn-Bose, 2023-01-26:
`bt_audio_discover()` has a callback when an ASE is discovered on a remote device
`unicast_client_ase_cp_discover()` asynchronously subscribes to notifications of ASE state changes

There's no way of the application knowing when `unicast_client_ase_cp_discover()` has completed so that it is safe to call `bt_audio_stream_config()`

This causes occasional race hazards when stream configuration occurs before state changes are notified.

I'm working on a PR to resolve this, probably calling the discovery callback **after** notifications have been enabled.

Relevant lines in mainline:
https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/audio/unicast_client.c#L1882

Chris-StJohn-Bose, 2023-01-27:
Example of that happens over-the-air
[image]

Chris-StJohn-Bose, 2023-01-27:
Adding a `cp_subscribe[index].subscribe` callback and re-using `bt_gatt_discover_params` seems to work plus I got rid of `cp_disc[]`

Chris-StJohn-Bose, 2023-02-02:
> Moved the commit to a separate PR to get it in ASAP: #54283

Yes that looks a lot like the fix I'm using. I'll stop preparing a PR and watch for #54283 merge. Thanks!

Thalley, 2023-02-02:
> > Moved the commit to a separate PR to get it in ASAP: #54283
> 
> Yes that looks a lot like the fix I'm using. I'll stop preparing a PR and watch for #54283 merge. Thanks!

Feel free to add your review to https://github.com/zephyrproject-rtos/zephyr/pull/54283 :)

Chris-StJohn-Bose, 2023-02-03:
Note sure about the priority:low as this is currently preventing streaming to a 3rd party receiver that doesn't automatically notify on CCC write, but it's being dealt with so the priority is fairly irrelevant.

Thalley, 2023-02-03:
> Note sure about the priority:low as this is currently preventing streaming to a 3rd party receiver that doesn't automatically notify on CCC write, but it's being dealt with so the priority is fairly irrelevant.

It's something that has been in the code for more than 2 years and haven't seen any reports about it until now, so arguably it isn't that big of an issue. Aside from that, it would be quite trivial for an application to add a small waiting period between the discover callback and the next procedure. 

Finally, until https://github.com/zephyrproject-rtos/zephyr/pull/53484 is merged, none of the notifications does anything besides being logged

closed 2023-02-06 by stephanosio