zephyrproject-rtos/zephyr #54148: qemu_x86_tiny places picolibc text outside of pinned.text

keith-packard, 2023-01-26:
The qemu_x86_tiny test target creates linker scripts with a special `.pinned.text` region that should contain all code necessary for running outside of task context. However, this region does not end up including the text from picolibc.

**To Reproduce**
Steps to reproduce the behavior (requires changing lib/libc/Kconfig to enable picolibc for x86):
1. west build -b qemu_x86_tiny samples/kernel/condition_variables/simple -- -DCONFIG_PICOLIBC=y
2. cd build
3. ninja run
4. See failure (application hangs after printing `Booting from ROM..`)

**Expected behavior**
Sample should run correctly.

**Impact**
Prevents picolibc from being used with this one board.

**Logs and console output**
Here's the memory regions containing program text:
```
$ objdump -h zephyr/zephyr.elf | grep text
  0 boot.text     00001000  40100000  00100000  00001000  2**2
  5 pinned.text   00007000  40101000  00101000  00002000  2**0
 21 text          00001000  40117000  00517000  00011000  2**0
```
And here's where it stuck strncpy, which is called from z_setup_new_thread:
```
$ objdump -t zephyr/zephyr.elf | grep strncpy
40117142 g     F text	0000002b strncpy
```
Note that strncpy is in the `text` section, not the `pinned.text` section. This section is not
mapped for access when z_setup_new_thread is running at zephyr startup.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.16.0-beta1
 - Commit SHA or Version used: bf384bf41a394e38b22132844691333320f109d6

**Additional context**
Need to patch lib/libc/Kconfig to enable Picolibc use with this board:
```
$ git diff
diff --git a/lib/libc/Kconfig b/lib/libc/Kconfig
index 9e8b618975..20ccd9b4a6 100644
--- a/lib/libc/Kconfig
+++ b/lib/libc/Kconfig
@@ -18,7 +18,6 @@ config SUPPORT_MINIMAL_LIBC
 # Picolibc with C++ support in Zephyr SDK is handled by Zephyr SDK's own Kconfig.
 config PICOLIBC_SUPPORTED
        bool
-       depends on ARC || ARM || ARM64 || MIPS || RISCV
        depends on "$(ZEPHYR_TOOLCHAIN_VARIANT)" != "arcmwdt"
        depends on !(CPP && "$(ZEPHYR_TOOLCHAIN_VARIANT)" = "zephyr")
        default y
```

keith-packard, 2023-01-28:
PR #54186 includes a patch that addresses this issue, https://github.com/zephyrproject-rtos/zephyr/pull/54186/commits/9c3717157d051e3ac5a8837555f4de70c873219d -- I had thought that wasn't right because many tests continued to fail, but I've found other fixes for almost all of those in that same series.

stephanosio, 2023-01-31:
@keith-packard Is this fixed by #54186?

keith-packard, 2023-01-31:
yes, sorry I forgot to stick a Closed tag in the PR.

closed 2023-01-31 by keith-packard