zephyrproject-rtos/zephyr #54162: Mass-Storage-Sample - USB HS support for the stm32f723e_disco board

whati001, 2023-01-27:
Hi Zephyr team,

I am currently trying to get the Mass-Storage-Sample up and running on the `stm32f723e_disco` board with USB-HS support. 
Unfortunately, it does not work for me, my linux machine recognized the USB device, but not the disks (block devices). Hence, I am unable to mount and interact with them. 

Based on my investigation, the issue is linked somehow to the `.wMaxPacketSize` property from the USB endpoints. If the max package size is set to 512 (which is the default for HS), the code stops when it tries to read from the OUT endpoint. If we change it to some strange value, for example 256 (yes it is not valid) the code seems to read successfully from the OUT endpoint, but the linux machine complains about some unaligned data and reconfigures the device over and over.

**dmesg with 512 as package size:**
```
[12070.883242] usb 1-9: new high-speed USB device number 46 using xhci_hcd
[12071.023814] usb 1-9: New USB device found, idVendor=2fe3, idProduct=0008, bcdDevice= 3.02
[12071.023836] usb 1-9: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[12071.023846] usb 1-9: Product: Zephyr MSC sample
[12071.023854] usb 1-9: Manufacturer: ZEPHYR
[12071.023862] usb 1-9: SerialNumber: 30375119004B004A
[12071.026439] usb-storage 1-9:1.0: USB Mass Storage device detected
[12071.027041] scsi host4: usb-storage 1-9:1.0
```
**dmesg with (strange) 256 as package size:**
```
[11728.339745] usb 1-9: new high-speed USB device number 40 using xhci_hcd
[11728.483675] usb 1-9: config 1 interface 0 altsetting 0 bulk endpoint 0x81 has invalid maxpacket 256
[11728.483688] usb 1-9: config 1 interface 0 altsetting 0 bulk endpoint 0x1 has invalid maxpacket 256
[11728.484026] usb 1-9: New USB device found, idVendor=2fe3, idProduct=0008, bcdDevice= 3.02
[11728.484032] usb 1-9: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[11728.484036] usb 1-9: Product: Zephyr MSC sample
[11728.484041] usb 1-9: Manufacturer: ZEPHYR
[11728.484046] usb 1-9: SerialNumber: 30375119004B004A
[11728.485689] usb-storage 1-9:1.0: USB Mass Storage device detected
[11728.486042] scsi host4: usb-storage 1-9:1.0
[11729.496762] scsi 4:0:0:0: Direct-Access     ZEPHYR   ZEPHYR USB DISK  0.01 PQ: 0 ANSI: 0 CCS
[11729.497215] sd 4:0:0:0: [sdc] 192 512-byte logical blocks: (98.3 kB/96.0 KiB)
[11729.497326] sd 4:0:0:0: [sdc] Write Protect is off
[11729.497328] sd 4:0:0:0: [sdc] Mode Sense: 03 00 00 00
[11729.497447] sd 4:0:0:0: [sdc] No Caching mode page found
[11729.497449] sd 4:0:0:0: [sdc] Assuming drive cache: write through
[11729.620094] usb 1-9: reset high-speed USB device number 40 using xhci_hcd
[11729.896635] usb 1-9: reset high-speed USB device number 40 using xhci_hcd
[11730.180075] usb 1-9: reset high-speed USB device number 40 using xhci_hcd
[11730.470105] usb 1-9: reset high-speed USB device number 40 using xhci_hcd
[11730.753401] usb 1-9: reset high-speed USB device number 40 using xhci_hcd
[11731.040112] usb 1-9: reset high-speed USB device number 40 using xhci_hcd
[11731.184702] sd 4:0:0:0: [sdc] Unaligned partial completion (resid=3840, sector_sz=512)
[11731.184718] sd 4:0:0:0: [sdc] tag#0 CDB: Read(10) 28 00 00 00 00 00 00 00 08 00
[11731.184732] sd 4:0:0:0: [sdc] tag#0 FAILED Result: hostbyte=DID_ERROR driverbyte=DRIVER_OK cmd_age=1s
[11731.184738] sd 4:0:0:0: [sdc] tag#0 CDB: Read(10) 28 00 00 00 00 00 00 00 08 00
[11731.184743] I/O error, dev sdc, sector 0 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 2
[11731.184756] Buffer I/O error on dev sdc, logical block 0, async page read
[11731.313504] usb 1-9: reset high-speed USB device number 40 using xhci_hcd

```
Not sure if this is relevant though.

**To Reproduce**
proj.conf
```
CONFIG_STDOUT_CONSOLE=y
#USB related configs
CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_DEVICE_PRODUCT="Zephyr MSC sample"
CONFIG_USB_DEVICE_PID=0x0008
CONFIG_LOG=y
CONFIG_USB_DRIVER_LOG_LEVEL_ERR=y
CONFIG_USB_MASS_STORAGE=y
CONFIG_USB_DEVICE_LOG_LEVEL_ERR=y
CONFIG_USB_MASS_STORAGE_LOG_LEVEL_ERR=y
CONFIG_USB_DEVICE_INITIALIZE_AT_BOOT=n
CONFIG_USB_DEVICE_LOG_LEVEL_DBG=y
CONFIG_LOG_BUFFER_SIZE=65536
CONFIG_MAIN_STACK_SIZE=1536
```

stm32f723e_disco.overlay
```
// disable USB-FS
&usbotg_fs {
    status = "disabled";
};

// enable USB-HS
&usbotg_hs {
    // I assume, this config is for nothging, but mandatory pinctrl-0 and pinctrl-names
    pinctrl-0 = <&usb_otg_fs_dm_pa11 &usb_otg_fs_dp_pa12>;
    pinctrl-names = "default";
    // this is important -> enable USB-HS
    status = "ok";
};
```

Compile the unchanged [Mass-Storage-Sample](https://docs.zephyrproject.org/latest/samples/subsys/usb/mass/README.html) code. 
```bash
$ west build -p -b stm32f723e_disco samples/subsys/usb/mass/ -- -DCONFIG_APP_MSC_STORAGE_RAM=y
$ west flash
```

**Expected behavior**
To receive a mass storage device. The code works with an nrf52840-dk board, already tested (I have one at home).

**Impact**
Some lack of USB-HS support for the specified board.

**Logs and console output**
Please find attached the log output and a usb capture taken with wireshark. The device address should be 1.52.* based on lsusb:
```bash
$ lsusb  
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 052: ID 2fe3:0008 NordicSemiconductor Zephyr MSC sample
Bus 001 Device 002: ID 0bda:5650 Realtek Semiconductor Corp. Integrated Webcam_HD
Bus 001 Device 007: ID 0a5c:5834 Broadcom Corp. 5880
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
[usbhs_stm32f723e_disco.tar.gz](https://github.com/zephyrproject-rtos/zephyr/files/10518576/usbhs_stm32f723e_disco.tar.gz)

**Environment (please complete the following information):**
 - OS: Linux Arch 6.1.7-arch1-1
 - Toolchain: Zephyr SDK
 - Commit SHA: 56ec06f3441c10dbb6678790322f92144163c753 (use this commit, because in the main we need to define the clock in the device tree; here not)

**Additional information**
I have noticed that this example does not work, because I am actually trying to reimplement a different USB protocol for my master thesis. This protocol also requires USB-HS with a package size of 512 and is vastly the same as the sample. Hence, if we fix the sample also my problem shoul
[usbhs_stm32f723e_disco.tar.gz](https://github.com/zephyrproject-rtos/zephyr/files/10518573/usbhs_stm32f723e_disco.tar.gz)
d be solve.

erwango, 2023-01-30:
@loicpoulain Would you have time to have a look ?

whati001, 2023-01-31:
> @loicpoulain Would you have time to have a look ?

If there are any question, please feel free to ping me.

GeorgeCGV, 2023-02-01:
I had an issue with the [USB MSC sample](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/subsys/usb/mass) back in the day (_December 2022, v3.2.0_) using custom board based on STM32H730. 

The issue was related to constant periodic pipe resets.

Observed pattern:
[image]

Luckily, @desowin had time to investigate the USB capture file (_discord usb channel_). The outcome:

> It looks (given the very limited insight) that for whatever reason the USB device driver sends STALL as response instead of simply NAK when it has nothing to send
>
> The URB_BULK in complete status is probably something alongside endpoint stalled
>
> The interesting finding from the capture is that the device responds correctly if the IN data is less than max packet size. If it is more (host tries to read 65536 bytes), then device STALLs the IN endpoint. That must be some ST USB device driver "feature"
>
> I doubt the problem is anywhere else than in the ST USB device driver because 65536 bytes reads work just fine with mass storage sample on nRF52840 DK

Maybe the issues are related.

_PS: STM32Cube USB Device MSC RAM example worked fine. Didn't test with v3.3.0-rc1._

whati001, 2023-02-15:
issue seems to remain in Zephyr OS build v3.3.0-rc3.
To run the application, the PR from [ISSUE-54719](https://github.com/zephyrproject-rtos/zephyr/issues/54719) needs to be merged/applied.

Here my devicetree config:
```dtc
&usbotg_fs {
    status = "disabled";
};

&usbotg_hs {
    clocks = <&rcc STM32_CLOCK_BUS_AHB1 0x20000000>,
        <&rcc STM32_SRC_PLL_Q CK48M_SEL(0)>;
    pinctrl-0 = <&usb_otg_hs_dm_pb14 &usb_otg_hs_dp_pb15>;
    pinctrl-names = "default";
    status = "ok";
};
```

loicpoulain, 2023-02-15:
So it could be because of missing ZLP packet when packet is equal to max packet size... drivers seems to manage that differently. I think this is mostly because mass storage use the low level usb_write()/read() API. Ideally it should be converted to use the transfer API with, e.g. `usb_transfer_sync()` instead of `usb_write()`.

whati001, 2023-02-15:
No it's not because of an ZLP packet; did some research today and seems like it's related to the RX-FIFO size. 
If we checkout the example code from the STM32CubeF7 USB-Device app, we see that they use a RX-FIFO size of 0x200 (512) for USB_OTG_HS ([here](https://github.com/STMicroelectronics/STM32CubeF7/blob/7e354cf248d84d84424e852ab012b345e3155640/Projects/STM32F723E-Discovery/Applications/USB_Device/DFU_Standalone/Src/usbd_conf.c#L333)): 

Zephyr (3.3.0-rc3) computes the size; by dividing the total amount of ram by 4 and again by the maximum allowed endpoints + 1. Concrete for the stm32f723e_disco board (stm32f.dtsi), this would be ((4096 /  4) / (9 + 1)), which is 102.4 and way less than the 512. 

**DeviceTree:**
```dts
usbotg_hs: usb@40040000 {
			num-bidir-endpoints = <9>;
			ram-size = <4096>;
                       //....
		};
```
This value is than used to initialize the RX-FIFO inside the `usb_dc_stm32_init` function:
[image]

There is actually a TODO haha.....

For testing, I have changed the value to 0x200 if we have a USB_OTG_HS device:
```c
#if DT_HAS_COMPAT_STATUS_OKAY(st_stm32_otghs) // USBOTG_HS
	HAL_PCDEx_SetRxFiFo(&usb_dc_stm32_state.pcd, 0x200);
#else // USBOTG_FS
	HAL_PCDEx_SetRxFiFo(&usb_dc_stm32_state.pcd, FIFO_EP_WORDS);
#endif
```
After this change, the linux machine at least identifies that there some sdc partition available. However, I am not sure how to change this in a nice fashion, though. 
```bash
[33401.407073] usb 1-9: new high-speed USB device number 28 using xhci_hcd
[33401.547740] usb 1-9: New USB device found, idVendor=2fe3, idProduct=0008, bcdDevice= 3.03
[33401.547754] usb 1-9: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[33401.547760] usb 1-9: Product: Zephyr MSC sample
[33401.547765] usb 1-9: Manufacturer: ZEPHYR
[33401.547769] usb 1-9: SerialNumber: 30375119004B004A
[33401.549871] usb-storage 1-9:1.0: USB Mass Storage device detected
[33401.550254] scsi host4: usb-storage 1-9:1.0
[33402.561448] scsi 4:0:0:0: Direct-Access     ZEPHYR   ZEPHYR USB DISK  0.01 PQ: 0 ANSI: 0 CCS
[33402.562678] sd 4:0:0:0: [sdc] 64 512-byte logical blocks: (32.8 kB/32.0 KiB)
[33402.562923] sd 4:0:0:0: [sdc] Write Protect is off
[33402.562938] sd 4:0:0:0: [sdc] Mode Sense: 03 00 00 00
[33402.563125] sd 4:0:0:0: [sdc] No Caching mode page found
[33402.563135] sd 4:0:0:0: [sdc] Assuming drive cache: write through
```

In addition, as mentioned above in the `Additional information` section, I have noticed this issue by developing a different application and also this application seems to work after this "fix".

PS: do not forget to apply the fix from this [PR](https://github.com/zephyrproject-rtos/zephyr/pull/54716/files) as well.

loicpoulain, 2023-02-15:
> Or properly simply increase the memory for the usb device such as: `ram-size = <20480>;`??

Good catch ! But we can't change the ram size, it's not main memory but a dedicated 4kb ram for USB HS. Ideally we need to better adjust the allocated memory for the endpoints instead of just splitting it equally between all (possibly unused) endpoints. We can e.g limit fifo 0 (ep0) to 64 bytes. 

Note that the size is normally in word unit (32bit) so setting tx/rx fifo to 0x80 should allocate 512 bytes.

A quick workaround would be to limit the number of available endpoint to 4 instead of 9, which should give you enough fifo space.

tmon-nordic, 2023-02-16:
> So it could be because of missing ZLP packet when packet is equal to max packet size

Definitely not because Mass Storage class does not use ZLPs.

whati001, 2023-02-16:
> A quick workaround would be to limit the number of available endpoint to 4 instead of 9, which should give you enough fifo space.
Sure, but what if somebody needs more than 4 eps? This can be easily the case. Is actually `num-bidir-endpoints` fixed by the MCU or the user? IIRC the USB standard supports up to 16 ep?

In addition, isn't there a single RX-FIFO, which should be `MAX(foreach: ep.wMaxPacketSize)` in size and a TX-FIFO for each ep, which should be exactly ep.wMaxPacketSize in size?

loicpoulain, 2023-02-16:
> Sure, but what if somebody needs more than 4 eps? This can be easily the case. Is actually `num-bidir-endpoints` fixed by the MCU or the user? IIRC the USB standard supports up to 16 ep?

The 1 + 8 endpoints are defined by MCU spec, so it's a hardware property. Limiting it to 4 is more a local temp workaround and is usually fine for single function devices.

> In addition, isn't there a single RX-FIFO, which should be `MAX(foreach: ep.wMaxPacketSize)` in size and a TX-FIFO for each ep, which should be exactly `ep.wMaxPacketSize` in size?

Yes, but since EP configuration and enabling is dynamically done at runtime we don't know that in configuration steps. Wondering if we could do it more dynamically in endpoint enabling (will submit a change). Also, each FIFO can contain more than one packet, so it theorically something we should fine tune for the application and expected performances.

whati001, 2023-02-17:
got your point. 

But do normal endpoints such as BULK also count to `num-bidir-endpoints `? or just EP0, so eps which can handle in and out?

loicpoulain, 2023-03-06:
> For example my ep has the address 0x81; than I can change the fifo size by calling:
> 
> ```c
> HAL_PCDEx_SetTxFiFo(&usb_dc_stm32_state.pcd, 1, my_new_size);
> ```

Yes, there is an helper function to convert endpoing address to its corresponding index: `USB_EP_GET_IDX`
num-bidir-endpoints is simply the number of endpoints that are configurable either as IN or OUT (ep0 included).

closed 2023-05-21 by github-actions[bot]