zephyrproject-rtos/zephyr #54175: Bluetooth: gatt: Allow to use CONFIG_BT_GATT_DYNAMIC_DB without CONFIG_BT_GATT_SERVICE_CHANGED

MarekPieta, 2023-01-27:
**Describe the solution you'd like**
If GATT Services are registered/unregistered only before Bluetooth is enabled, the GATT database would not change (from remote device's perspective). We could allow to use the dynamic database GATT API when Service Changed support is disabled in that case (register/unregister functions could return an error when GATT Service is registered after Bluetooth was enabled).

**Additional context**
Disabling support for Service Changed could allow developers to save quite a lot of memory (even~3500B of FLASH + ~ 500B of RAM when I checked locally; needed to remove implementation of `sc_indicate` to build successfully). The feature could be useful for memory-limited devices.

piotrParf, 2023-01-31:
Hi, Have you investigated how to apply this feature(removal of Service Changed)? Have you prepared some sort of patch? I just came to this topic and found your question.

MarekPieta, 2023-01-31:
To quickly test the feature, you could remove dependency on `CONFIG_BT_GATT_SERVICE_CHANGED` from `CONFIG_BT_GATT_DYNAMIC_DB` and comment out the mentioned `sc_indicate` implementation in `bluetooth/host/gatt.c` (to prevent build failures).  Then you can disable the `CONFIG_BT_GATT_SERVICE_CHANGED` in your configuration. Make sure that you do not register any GATT Services after Bluetooth is enabled in that case too.

Proper solution may require applying additional changes to the code (to ensure that there are no hidden assumptions related to the removed dependency). Apart from that. I think that  it would be good to add checks ensuring that no GATT Service is registered/unregistered after Bluetooth is enabled (we could either return an error or at least assert that).

I do not have patch for this one, but applying fixes mentioned in my first comment locally should be straightforward.