zephyrproject-rtos/zephyr #54190: RP2040 cannot be compiled with C11 enabled

ThadHouse, 2023-01-28:
**Describe the bug**
When compiling a project targeting RP2040 and C11 enabled, errors occur because both assert.h (from the std lib) and config_autogen.h (from zephyr) define `static_assert` as a macro, with the one in config_autogen not being guarded.

**To Reproduce**
Steps to reproduce the behavior:
1. Set C11 flag on any project (`set_property(TARGET tgt PROPERTY C_STANDARD 11)`)
2. mkdir build; cd build
3. cmake -DBOARD=rpi_pico
4. make
5. See error

**Expected behavior**
No compiler error

**Impact**
C11 cannot be used for RP2040 projects

**Environment (please complete the following information):**
 - OS: Linux and Windows
 - Toolchain: Zephyr SDK 0.15.0
 - Zephyr 3.1 (Still occurs in 3.2)

cfriedt, 2023-01-28:
Building Zephyr for C11 should likely involve using a Kconfig option rather than setting a CMake property in an arbitrary CMakeLists.txt, but this should be investigated in any case.

A Kconfig value such as `CONFIG_LIBC_STD_C11` or something along those lines  would probably be smart.

stephanosio, 2023-01-28:
> Building Zephyr for C11 should likely involve using a Kconfig option rather than setting a CMake property in an arbitrary CMakeLists.txt, but this should be investigated in any case.
> 
> A Kconfig value such as `CONFIG_LIBC_STD_C11` or something along those lines would probably be smart.

@cfriedt I would suggest moving that discussion to https://github.com/zephyrproject-rtos/zephyr/issues/45776 since this issue does not really have much to do with C language support in Zephyr.

closed 2023-01-28 by cfriedt