zephyrproject-rtos/zephyr #54199: dns resolve: fails after 60 seconds uptime

joelguittet, 2023-01-28:
**Describe the bug**

I'm running an HTTP client application on STM32L4A6 Nucleo board + ENC28J60 module.
After exactly 60 seconds uptime, the DNS resolve fails to query addresses. I have observed this in my application but also in the console just doing `net dns query google.fr` plenty of time up to the first failure, which occurs EXACTLY after 60 seconds uptime.

In my application `getaddrinfo` function returns error code -101.

My prj.conf particularly contains:

```
CONFIG_DNS_RESOLVER=y
CONFIG_DNS_RESOLVER_LOG_LEVEL_DBG=y
```

**To Reproduce**
Steps to reproduce the behavior:
1. On nucleo_l4a6 + enc28j60 build a projet with Ethernet support
2. in the console type `net dns query google.fr`

**Expected behavior**
DNS resolution working.

**Impact**
Blocking the development.

**Logs and console output**

In the console and after activating debug logs I see it is because of timeout:

```
[00:01:08.776,000] <dbg> net_dns_resolve: dns_resolve_name: (): DNS id will be 34471
[00:01:08.776,000] <dbg> net_dns_resolve: dns_write: (): [0] submitting work to server idx 0 for id 34471 hash 24745
[00:01:10.776,000] <dbg> net_dns_resolve: query_timeout: (sysworkq): Query timeout DNS req 34471 type 1 hash 24745
[00:01:10.776,000] <dbg> net_dns_resolve: dns_resolve_cancel_with_hash: (sysworkq): Cancelling DNS req 34471 (name my-server.com type 1 hash 24745)
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - I'm actually at zephyr commit ID e9a59518fcbd54f2ee27ea3f190a8fe7bd688f35 (v3.3.0-rc1 tag).

**Additional context**
None

Thanks for any support,
Joel

RomainPelletant, 2023-01-30:
Hi @joelguittet,

Can you describe you network configuration please? (DHCP/IP static adress/DNS entries..)

joelguittet, 2023-01-30:
Hello @RomainPelletant 

Sure. I have DHCPv4 activated; no static IP addres. DNS server is the local router (properly shown in the shell) at 192.168.1.1.
Information added in the post above.

I just ran the dns_resolve sample on an other board (with no enc28j60 so te config is much different). It's properly working.
In the dns_resolve sample I notice there is a specific configuration as follow:

```
# Enable the DNS resolver
CONFIG_DNS_RESOLVER=y
# Enable additional buffers
CONFIG_DNS_RESOLVER_ADDITIONAL_BUF_CTR=5
# Enable additional queries
CONFIG_DNS_RESOLVER_ADDITIONAL_QUERIES=2
# Enable mDNS support
CONFIG_MDNS_RESOLVER=y
# Enable LLMNR support
CONFIG_LLMNR_RESOLVER=n

CONFIG_DNS_RESOLVER_MAX_SERVERS=2
CONFIG_DNS_SERVER_IP_ADDRESSES=y
CONFIG_DNS_NUM_CONCUR_QUERIES=5
```

But the documentation at https://docs.zephyrproject.org/latest/connectivity/networking/api/dns_resolve.html do not really explains the details of the CONFIG. Just to check the KConfig which is not enough I think.

I should maybe play with these settings, right ?

Regards,

RomainPelletant, 2023-01-30:
If I am not wrong, DHCP shall add the right DNS entry (in your case 192.168.1.1).
To know where is the problem, you could add fix DNS entry like the following

```
CONFIG_DNS_SERVER_IP_ADDRESSES=y
CONFIG_DNS_SERVER1="192.168.1.1"
CONFIG_DNS_SERVER2="8.8.8.8"
```

Using the IP directly is working?

joelguittet, 2023-01-30:
Seems this help, but after a quick moment, then I'm no more able to query DNS resolution. Using the shell, timeout or -11 error is reported. I will try to use some configuration settings from the dns_resolve sample.

joelguittet, 2023-02-01:
Using configuration of the DNS from dns_resolve sample I got the same result: DNS resolution working at the beginning, but failing after few minutes.

Maybe it's due to enc28j60 driver ? Don't know how to determine this :-(

rlubos, 2023-02-03:
I'd check if the DNS server address does not get corrupted somehow (you can list configured servers with `net dns`). 

Otherwise, if it's a driver issue, I'm not really of much use as I don't have experience with that particular one. Are you able to send any other packets when DNS starts to malfunction (for example with `net udp` command)?

joelguittet, 2023-02-05:
Hello

`net dns` still show the DNS address after the issue occurs.
I have checked if something else can be done using the `net ping 8.8.8.8` in the console. Before the issue occurs it's working. After the issue occurs I get "Ping timeout" error.

Note: When I opened the ticket the issue occurred 1 minute after uptime. But it seems to be generally 3 or 4 minutes after uptime.

It seems the issue is related to the ENC28J60 driver but it's strange I'm the first to report it. Isn't it ? Or maybe it is not used a lot ? I should receive an ESP01 module soon, I will check with this module if I get the error with my hardware and my application, this will confirm (or not) if this is due to enc28j60 driver.

Joel

joelguittet, 2023-02-07:
Hello

I have build and ran my application on stm32f746g_disco board, which has RMII LAN8742A Ethernet PHY without any issue. The difference is only the Ethernet PHY. The configuration used was the same (DHCPv4, dynamic DNS address...)

We can conclude this is an issue with the ENC28J60 driver. I'm updating the first post of this thread to indicate this.

Joel

carlescufi, 2023-02-08:
> We can conclude this is an issue with the ENC28J60 driver. I'm updating the first post of this thread to indicate this.

Unfortunately this driver is currently unmaintained, so we need a volunteer to step up and provide a fix. Unassigning @rlubos since he is the maintainer of the networking stack and not individual drivers.

joelguittet, 2023-02-08:
@carlescufi no problem I understand this. Will try W5500 and ESP01 as a replacement for my project. I'm not enough experienced with zephyr to start this kind of deep debugging on the networking parts :-)

RomainPelletant, 2023-02-08:
@joelguittet FYI i am using eth_enc424j600 without any DNS issues

joelguittet, 2023-02-08:
Thanks @RomainPelletant I just received this afternoon a W5500 module. Wired, reconfigured in the dts. Working. No modification of the app again. This is a double confirmation enc28j60 has issues.

jfischer-no, 2023-04-13:
I briefly tested it on nRF52840DK today. I used zperf sample (`net ping 2001:db8::affe -c 1000 -i 10 -s 1000`) and see MPU FAULT  `<err> eth_enc28j60: Failed to read memory` somewhere in buf_simple. And this reminds me of a weird bug that seemed to be a controller RX FIFO overflow that I fought in this driver few years ago, I finally gave up and implemented enc424j600 driver. If anyone wants to spend time on this, please reassign.

joelguittet, 2023-07-27:
Hello

Working with W5500 interface since several months now I realize I now get some issues (my application is growing...)
And the reason is pretty simple and it was probably the root cause of the ENC28J60 issue here : this kind of chip offload the sockets but it's not integrated in Zephyr using offloading possibilities (see for example eswifi of winc1500 drivers in Zephyr).

As a consequence only a single socket can be used at a time.

Driver to be reworked to permit offloading the sockets instead of just using the first socket in a raw-access mode.

Joel

jfischer-no, 2023-07-27:
> Working with W5500 interface since several months now I realize I now get some issues (my application is growing...)
> And the reason is pretty simple and it was probably the root cause of the [...]

This is ingenious causal inference.

jhedberg, 2023-09-26:
@jfischer-no are you planning to work on this? (it's getting the Stale label auto-added)

jfischer-no, 2023-09-26:
> @jfischer-no are you planning to work on this? (it's getting the Stale label auto-added)

https://github.com/zephyrproject-rtos/zephyr/issues/54199#issuecomment-1506944544
I do not think it is worth it.

jhedberg, 2023-09-26:
> > @jfischer-no are you planning to work on this? (it's getting the Stale label auto-added)
> 
> [#54199 (comment)](https://github.com/zephyrproject-rtos/zephyr/issues/54199#issuecomment-1506944544) I do not think it is worth it.

Ok, let’s close this then

closed 2023-09-26 by jhedberg