zephyrproject-rtos/zephyr #54202: decawave_dwm1001_dev: i2c broken due to pinctrl_nrf fix

sdalu, 2023-01-29:
Building the sample demo `samples/sensor/lis2dh` with the  decawave_dwm1001_dev result in 

~~~
uart:~$ *** Booting Zephyr OS build zephyr-v3.2.0  ***
Device lis2dh12@19 is not ready
~~~

It seems to be related to this commit,  according to git bisect, https://github.com/zephyrproject-rtos/zephyr/commit/32a87215d7aa97191097357f602237f7da4a3536  : drivers: pinctrl_nrf: Fix nrf_pin_configure() implementation

nordicjm, 2023-01-30:
If that is the case then the i2c pinctrl lines need `bias-pull-up;` added.

sdalu, 2023-02-02:
I changed `boards/arm/decawave_dwm1001_dev/decawave_dwm1001_dev-pinctrl.dtsi` to include `bias-pull-up` as defined in the fragment below, but it doesn't fix the problem

~~~
	i2c0_default: i2c0_default {
		group1 {
			psels = <NRF_PSEL(TWIM_SDA, 0, 29)>,
				<NRF_PSEL(TWIM_SCL, 0, 28)>;
			bias-pull-up;
		};
	};

	i2c0_sleep: i2c0_sleep {
		group1 {
			psels = <NRF_PSEL(TWIM_SDA, 0, 29)>,
				<NRF_PSEL(TWIM_SCL, 0, 28)>;
			bias-pull-up;
			low-power-enable;
		};
	};
~~~

nordicjm, 2023-02-02:
Remove the pull up from the sleep mode. That should be all you need, it would enable pulling the lines up which i2c needs. You'd need to check with a scope to debug further if that does not work.

sdalu, 2023-02-02:
Not working better :( 
I don't have a scope to perform additional check :(

jfischer-no, 2023-02-08:
The changes mentioned above fixes the issue. But there is something fishy with the flash runner, it does not reset core after flashing, you have to press reset switch on the board after the J-Link status LED on the board starts to pulse again.
@sdalu Would you like to continue and open a PR to fix this issue?

```diff
diff --git a/boards/arm/decawave_dwm1001_dev/decawave_dwm1001_dev-pinctrl.dtsi b/boards/arm/decawave_dwm1001_dev/decawave_dwm1001_dev-pinctrl.dtsi
index 8e47b5d13c6..8cc231c74b1 100644
--- a/boards/arm/decawave_dwm1001_dev/decawave_dwm1001_dev-pinctrl.dtsi
+++ b/boards/arm/decawave_dwm1001_dev/decawave_dwm1001_dev-pinctrl.dtsi
@@ -23,6 +23,7 @@
                group1 {
                        psels = <NRF_PSEL(TWIM_SDA, 0, 29)>,
                                <NRF_PSEL(TWIM_SCL, 0, 28)>;
+                       bias-pull-up;
                };
        };
 
```
```
*** Booting Zephyr OS build v3.3.0-rc2-51-g4e2bd62b6a95 ***
Polling at 0.5 Hz
#1 @ 416 ms: x -0.229824 , y 0.153216 , z 9.461088
#2 @ 2421 ms: x -0.268128 , y 0.153216 , z 9.499392
#3 @ 4426 ms: x -0.268128 , y 0.076608 , z 9.576000
#4 @ 6431 ms: x -0.229824 , y 0.076608 , z 9.614304
#5 @ 8437 ms: x -0.229824 , y 0.076608 , z 9.576000
#6 @ 10442 ms: x -0.344736 , y 0.114912 , z 9.959040
```

nordicjm, 2023-02-09:
If the board doesn't reset then it's probably because nRESET is not correctly wired. Can work around the issue by adding `board_runner_args(nrfjprog "--softreset")` to the `board.cmake` file

jfischer-no, 2023-02-09:
> If the board doesn't reset then it's probably because nRESET is not correctly wired. Can work around the issue by adding `board_runner_args(nrfjprog "--softreset")` to the `board.cmake` file

It uses jlink runner, but thanks for the hint, I just opened PR now to get this issue out of the way.

sdalu, 2023-02-09:
> If the board doesn't reset then it's probably because nRESET is not correctly wired. Can work around the issue by adding `board_runner_args(nrfjprog "--softreset")` to the `board.cmake` file

Is there something equivalent for openocd ?

PR #54652 (2023-02-10): boards: decawave_dwm1001_dev: enable pull-up on pins used for I2C, reset core after load
Enable pull-up on pins used for I2C, reset core after load.

Fixes: #54202
closed 2023-02-10 by stephanosio