zephyrproject-rtos/zephyr #54209: USB C PD dead battery support

deyro, 2023-01-30:
**Describe the bug**
I am testing the `sink` sample of USB C stack on a STM `b_g474e_dpow1` board. The board has dead battery support. My end application will not have batteries so the dead battery feature is mandatory for me. If I have the power jumper of the board selected to `STLK` (similar to having batteries) then everything works fine. But when I am trying to power the board directly from the USB C port (power jumper selected to USBC), as in dead battery, the board does not boots up, it keeps restarting and prints the following logs.

I have tested the same `b_g474e_dpow1` board with STM Cube example `USB-PD_Consumer_1port` and did not face this issue, so the hardware is fine.

Is there some configuration that needs to be enabled for dead battery support?

**To Reproduce**
Steps to reproduce the behavior:
1. build `sink` sample under `zephyr/samples/subsys/usb_c` for board `b_g474e_dpow1`
2. open a serial terminal
3. run the sample on the `b_g474e_dpow1` board having the `STLK` power selection jumper shorted. Test with a USB PD source, the sample should work fine and logs should print on the serial terminal
4. now disconnect the USBs and short the `USBC` power selection jumper. Connect the micro-USB to view serial messages. Now try powering the board via a USBC PD source and see behavior
5. See logs

**Expected behavior**
The board should boot up and negotiate the appropriate power via USBC PD protocol

**Impact**
showstopper

**Logs and console output when trying to power directly through USB C (dead battery)**
```
*** Booting Zeph[00:00:00.000,000] <dbg> ucpd_stm32: ucpd_init: Pinctrl signals configuration
*** Booting Zephyr [00:00:00.000,000] <dbg> ucpd_stm32: ucpd_init: Pinctrl signals configuration
*** Booting [00:00:00.000,000] <dbg> ucpd_stm32: ucpd_init: Pinctrl signals configuration
*** Booting [00:00:00.000,000] <dbg> ucpd_stm32: ucpd_init: Pinctrl signals configuration
*** Booting Zep[00:00:00.000,000] <dbg> ucpd_stm32: ucpd_init: Pinctrl signals configuration
*** Booting Zephyr[00:00:00.000,000] <dbg> ucpd_stm32: ucpd_init: Pinctrl signals configuration
*** Booting [00:00:00.000,000] <dbg> ucpd_stm32: ucpd_init: Pinctrl signals configuration
*** Booting Zephyr [00:00:00.000,000] <dbg> ucpd_stm32: ucpd_init: Pinctrl signals configuration
*** Booting Z[00:00:00.000,000] <dbg> ucpd_stm32: ucpd_init: Pinctrl signals configuration
*** Booting Z[00:00:00.000,000] <dbg> ucpd_stm32: ucpd_init: Pinctrl signals configuration
*** Booting [00:00:00.000,000] <dbg> ucpd_stm32: ucpd_init: Pinctrl signals configuration
*** Booting Zephyr[00:00:00.000,000] <dbg> ucpd_stm32: ucpd_init: Pinctrl signals configuration
*** Booting Zephyr [00:00:00.000,000] <dbg> ucpd_stm32: ucpd_init: Pinctrl signals configuration
*** Booting Z[00:00:00.000,000] <dbg> ucpd_stm32: ucpd_init: Pinctrl signals configuration
*** Booting Z[00:00:00.000,000] <dbg> ucpd_stm32: ucpd_init: Pinctrl signals configuration
*** Booting Zep[00:00:00.000,000] <dbg> ucpd_stm32: ucpd_init: Pinctrl signals configuration
*** Booting Zeph[00:00:00.000,000] <dbg> ucpd_stm32: ucpd_init: Pinctrl signals configuration
*** Booting [00:00:00.000,000] <dbg> ucpd_stm32: ucpd_init: Pinctrl signals configuration
*** Booting Zephyr [00:00:00.000,000] <dbg> ucpd_stm32: ucpd_init: Pinctrl signals configuration
*** Booting Zeph[00:00:00.000,000] <dbg> ucpd_stm32: ucpd_init: Pinctrl signals configuration
*** Booting Zep[00:00:00.000,000] <dbg> ucpd_stm32: ucpd_init: Pinctrl signals configuration
*** Booting Zeph[00:00:00.000,000] <dbg> ucpd_stm32: ucpd_init: Pinctrl signals configuration
*** Booting Zephyr [00:00:00.000,000] <dbg> ucpd_stm32: ucpd_init: Pinctrl signals configuration
*** Booting Z[00:00:00.000,000] <dbg> ucpd_stm32: ucpd_init: Pinctrl signals configuration
*** Booting Zephyr[00:00:00.000,000] <dbg> ucpd_stm32: ucpd_init: Pinctrl signals configuration
*** Booting Zephy[00:00:00.000,000] <dbg> ucpd_stm32: ucpd_init: Pinctrl signals configuration
*** Booting Zephy[00:00:00.000,000] <dbg> ucpd_stm32: ucpd_init: Pinctrl signals configuration
```

**Environment:**
- OS: Linux
- Toolchain: zephyr-sdk-0.15.2
- Zephyr SDK zephyr-v3.2.0-3518-g7b5780e007
- Hardware: ST b_g474e_dpow1

erwango, 2023-01-30:
@sambhurst Feel free to convert to "Enhancement" if this is yet to be implemented.

erwango, 2023-02-06:
fixed by https://github.com/zephyrproject-rtos/zephyr/pull/54244

closed 2023-02-06 by erwango