zephyrproject-rtos/zephyr #54211: tests: kernel: timer: timer_behavior: kernel.timer.timer fails

nordic-piks, 2023-01-30:
**Describe the bug**
The tests/kernel/timer/timer_behavior/kernel.timer.timer fails

Observed for:
- nrf9160dk_nrf9160_ns

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf9160dk_nrf9160_ns connected
2. go to your zephyr dir
3. call ./scripts/twister -T tests/kernel/timer/timer_behavior -p nrf9160dk_nrf9160_ns --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build v3.3.0-rc1-5-gef9d5bbddfa5 ***
Running TESTSUITE timer_jitter_drift
===================================================================
START - test_jitter_drift
periodic timer behavior test
periodic timer samples gathered, calculating statistics
timer clock rate 32768, kernel tick rate 32768, expected period 1000 us, samples 10000, period 32.768000 cycles, expected period drift 7.080078 us
statistics samples 10000, rollovers 0, mean 1007.086182 us, variance 0.372492 us, stddev 0.610321 us, min 1007.080078 us, max 1068.115234 us
statistics samples 10000, rollovers 0, mean 33.000200 cycles, variance 0.000400 cycles, stddev 0.019999 cycles, min 33 cycles, max 35 cycles
timer start cycle 17613, end cycle 347615, total time 10070861.816406, expected time 10000000.000000, expected time drift 70800.781250, difference 61.035156
PASS - test_jitter_drift in 10.343 seconds
===================================================================
TESTSUITE timer_jitter_drift succeeded
Running TESTSUITE timer_tick_train
===================================================================
START - test_one_tick_timer_train
Initializing 4 Timers, Tick Rate 32768Hz, Expecting 81920 callbacks in 11000 ms
Starting Timers with Skews
Faking busy work, remaining timers is 4, timer callbacks 1
Faking busy work, remaining timers is 4, timer callbacks 7126
Faking busy work, remaining timers is 4, timer callbacks 14089
Faking busy work, remaining timers is 4, timer callbacks 21057
Faking busy work, remaining timers is 4, timer callbacks 28021
Faking busy work, remaining timers is 4, timer callbacks 34986
Faking busy work, remaining timers is 4, timer callbacks 41950
Faking busy work, remaining timers is 4, timer callbacks 48920
Faking busy work, remaining timers is 4, timer callbacks 55886
Faking busy work, remaining timers is 4, timer callbacks 62847
Faking busy work, remaining timers is 4, timer callbacks 69807
Faking busy work, remaining timers is 4, timer callbacks 76749
One Tick Timer Train Done, took 10066 ms, busy loop ran 12 times
Timer 0 max delta 8 cycles or 244 us
Timer 1 max delta 8 cycles or 244 us
Timer 2 max delta 8 cycles or 244 us
Timer 3 max delta 8 cycles or 244 us
Assertion failed at ../../../../../../../../../zephyr/zephyr/tests/kernel/timer/timer_behavior/src/tick_timer_train.c:143: timer_tick_train_test_one_tick_timer_train: (busy_loops > acceptable_busy_loops is false)
Expected thread to run while 1 tick timers are firing
FAIL - test_one_tick_timer_train in 10.183 seconds
===================================================================
TESTSUITE timer_tick_train failed.
------ TESTSUITE SUMMARY START ------
SUITE PASS - 100.00% [timer_jitter_drift]: pass = 1, fail = 0, skip = 0, total = 1 duration = 10.343 seconds
- PASS - [timer_jitter_drift.test_jitter_drift] duration = 10.343 seconds
SUITE FAIL -   0.00% [timer_tick_train]: pass = 0, fail = 1, skip = 0, total = 1 duration = 10.183 seconds
- FAIL - [timer_tick_train.test_one_tick_timer_train] duration = 10.183 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: 8f41a8dee34890ad19db7a442d248b9f
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.14.1
 - Commit SHA or Version used:  v3.3.0-rc1-5-gef9d5bbddfa5

nordic-piks, 2023-01-30:
Seems to be the same as in #52415 but affected platforms are limited.

stephanosio, 2023-01-31:
cc @teburd

nordic-piks, 2023-02-10:
Fails also at nrf5340dk_nrf5340_cpunet

teburd, 2023-02-10:
The timer train failing here indicates the default tick rate is likely too high to actually be processed reasonably. As in, getting the full 32.786 kHz worth of timer interrupts on this platform doesn't leave enough time to run threads reasonably.

teburd, 2023-02-15:
This may very well be fixed by #54756

FRASTM, 2023-02-22:
looks similar to https://github.com/zephyrproject-rtos/zephyr/issues/54926

teburd, 2023-02-28:
The test passes for me on main as of commit 9f46db90c832e56363c5d7bb42892651b39b271d on the nrf9160dk

teburd, 2023-02-28:
Closing, please re-open if I'm wrong

nordic-piks, 2023-03-01:
This test still fails on nrf5340dk_nrf5340_cpunet:
```
*** Booting Zephyr OS build zephyr-v3.3.0-530-g1751c8f0f59d ***
Running TESTSUITE timer_jitter_drift
===================================================================
START - test_jitter_drift
periodic timer behavior test (approx 7 seconds)
periodic timer samples gathered, calculating statistics
timer clock rate 32768, kernel tick rate 32768
expected period drift: 7.0800781 us
period duration statistics for 7000 samples (0 rollovers):
expected: 1000 us,       32.768000 cycles
min:      1007.080078 us, 33 cycles
max:      1037.597656 us, 34 cycles
mean:     1007.084438 us, 33.000143 cycles
variance: 0.133027 us, 0.000143 cycles
stddev:   0.364729 us, 0.011951 cycles
timer start cycle 12913, end cycle 243914,
total time 7049591.064453 us, expected time 7000000.000000 us,
expected time drift 49560.546875 us, difference 30.517578 us
PASS - test_jitter_drift in 7.270 seconds
===================================================================
TESTSUITE timer_jitter_drift succeeded
Running TESTSUITE timer_tick_train
===================================================================
START - test_one_tick_timer_train
Initializing 4 Timers, Tick Rate 32768Hz, Expecting 81920 callbacks in 11000 ms
Starting Timers with Skews
Faking busy work, remaining timers is 4, timer callbacks 0
Faking busy work, remaining timers is 4, timer callbacks 6992
Faking busy work, remaining timers is 4, timer callbacks 14000
Faking busy work, remaining timers is 4, timer callbacks 21007
Faking busy work, remaining timers is 4, timer callbacks 26360
Faking busy work, remaining timers is 4, timer callbacks 33357
Faking busy work, remaining timers is 4, timer callbacks 40357
Faking busy work, remaining timers is 4, timer callbacks 47351
Faking busy work, remaining timers is 4, timer callbacks 54356
Faking busy work, remaining timers is 4, timer callbacks 61371
Faking busy work, remaining timers is 4, timer callbacks 68370
Faking busy work, remaining timers is 4, timer callbacks 75364
One Tick Timer Train Done, took 10016 ms, busy loop ran 12 times
Perfect delta 4 cycles or 122 us
Timer 0 max delta 9 cycles or 275 us, 46611 late callbacks (3.5%)
Timer 1 max delta 9 cycles or 275 us, 46614 late callbacks (3.5%)
Timer 2 max delta 9 cycles or 275 us, 46605 late callbacks (3.5%)
Timer 3 max delta 9 cycles or 275 us, 46612 late callbacks (3.5%)
!! Some ticks were missed.
!! Consider making CONFIG_SYS_CLOCK_TICKS_PER_SEC smaller.
!! The busy loop didn't run as much as expected.
!! Consider making CONFIG_SYS_CLOCK_TICKS_PER_SEC smaller.
Assertion failed at ../../../../../../../../../zephyr/zephyr/tests/kernel/timer/timer_behavior/src/tick_timer_train.c:163: timer_tick_train_test_one_tick_timer_train: (busy_loops > acceptable_busy_loops is false)
Expected thread to run while 1 tick timers are firing
FAIL - test_one_tick_timer_train in 10.205 seconds
===================================================================
TESTSUITE timer_tick_train failed.
------ TESTSUITE SUMMARY START ------
SUITE PASS - 100.00% [timer_jitter_drift]: pass = 1, fail = 0, skip = 0, total = 1 duration = 7.270 seconds
- PASS - [timer_jitter_drift.test_jitter_drift] duration = 7.270 seconds
SUITE FAIL -   0.00% [timer_tick_train]: pass = 0, fail = 1, skip = 0, total = 1 duration = 10.205 seconds
- FAIL - [timer_tick_train.test_one_tick_timer_train] duration = 10.205 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: f5489f42c78c6eca0af4ed0c41e75593
PROJECT EXECUTION FAILED
```
With version zephyr-v3.3.0-530-g1751c8f0f59d

nordic-piks, 2023-03-01:
Can someone reopen this bug? (I am unable to do it)

nordic-piks, 2023-03-07:
Found failing on nrf9160dk_nrf9160 still but not frequently.

PR #56318 (2023-04-05): tests: kernel: timer_behavior: Decrease tick rate for nRF
Nordic targets use 24-bit RTC peripheral for system clock. Nordic system clock timeout implementation relies on RTC CC (capture compare) when the timeout is in future. Nordic system clock driver allows setting alarm only to 3 or more counts from current counter value due to silicon limitation (to ensure that CC event triggers before counter overflow).

RTC CC limitation does not have much impact on normal applications where there is no need to schedule such short timeouts, but is problematic in a timer test that expects being able to re [... cut]
commit c55266a9 (2023-04-05): tests: kernel: timer_behavior: Decrease tick rate for nRF

Nordic targets use 24-bit RTC peripheral for system clock. Nordic system
clock timeout implementation relies on RTC CC (capture compare) when
the timeout is in future. Nordic system clock driver allows setting
alarm only to 3 or more counts from current counter value due to silicon
limitation (to ensure that CC event triggers before counter overflow).

RTC CC limitation does not have much impact on normal applications where
there is no need to schedule such short timeouts, but is problematic in
a timer test that expects being able to r [... cut]
closed 2023-04-05 by carlescufi