zephyrproject-rtos/zephyr #54213: net: wifi: connection to WPA2 Enterprise/RADIUS network

RomainPelletant, 2023-01-30:
**Is your enhancement proposal related to a problem? Please describe.**
Based on wifi_mgmt, we are able to connect to WPA2 and WPA3 personnal AP. Our need is to provide a basic support for WPA2 Enterprise : 

- Authentification method:
    - EAP-TLS
    - EAP-PEAP
    - EAP-TTLS

- Identity (32 bytes)
- Password (32 bytes) 

**Describe the solution you'd like**

1. Add new entry for enum wifi_security_type : WIFI_SECURITY_TYPE_IEEE8021X and another entry for Authentification type (TLS/PEAP/TTLS)
2. Or add new entries for enum wifi_security_type : 

- WIFI_SECURITY_TYPE_IEEE8021X_TLS
- WIFI_SECURITY_TYPE_IEEE8021X_PEAP
- WIFI_SECURITY_TYPE_IEEE8021X_TTLS

3. Re-use SAE password and add identify as data members into wifi_connect_req_params structure :
`struct wifi_connect_req_params {
	const uint8_t *ssid;
	uint8_t ssid_length; /* Max 32 */

	uint8_t *psk;
	uint8_t psk_length; /* Min 8 - Max 64 */

	uint8_t *sae_password; /* Optional with fallback to psk */
	uint8_t sae_password_length; /* No length restrictions */

        uint8_t *identity; /* Only for IEEE8021X networks */
        uint8_t identify_length; /* No length restrictions */

	uint8_t band;
	uint8_t channel;
	enum wifi_security_type security;
	enum wifi_mfp_options mfp;
	int timeout; /* SYS_FOREVER_MS for no timeout */
};`

**Describe alternatives you've considered**
No alternative. I tried to based the proposal upon wpa_supplicant API : [https://wiki.archlinux.org/title/wpa_supplicant#802.1x/radius](url)

**Additional context**
ESP32 seems to support WPA2 Enterprise [https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/Wi-Fi_AT_Commands.html#cmd-jeap](url)

matt-wood-ct, 2024-05-24:
I'm confused why this hasn't got any traction? Surely enterprise Wi-Fi is a must have feature no?

jukkar, 2024-05-24:
The hostap / wpa_supplicant support for zephyr is slowly maturing. Currently the enterprise support is not fully there but there is work being done for it. I do not know the timeline for finalizing it.

krish2718, 2024-07-27:
I have linked the PR that is adding EAP-TLS support.

TjazVracko, 2024-10-23:
@krish2718 Now that #78921 has been merged, should this issue be closed as completed?

closed 2024-10-23 by RomainPelletant