zephyrproject-rtos/zephyr #54224: Intel CAVS : tests/posix/ fails

majunkier, 2023-01-30:
**Describe the bug**
On CAVS 15, 18 and 25 tests/posix/ fails.

**To Reproduce**
Build and run test on CAVS 15, 18 or 25. Twister options given in our CI:
`-W --hardware-map ../../cavs/cavs.map --device-testing -x=CONFIG_BOOT_DELAY=500 -T tests/posix/ -v --no-skipped-report`

**Expected behavior**
Passes.

**Impact**
Annoyance.

**Logs and console output**
`INFO    - 1) tests/posix/eventfd_basic/portability.posix.eventfd_basic.picolibc on intel_adsp_cavs18 error (No Console Output(Timeout))
INFO    - 2) tests/posix/eventfd_basic/portability.posix.eventfd_basic.picolibc.posix_api on intel_adsp_cavs18 error (No Console Output(Timeout))
INFO    - 3) tests/posix/fs/portability.posix.fs.picolibc on intel_adsp_cavs18 error (No Console Output(Timeout))
INFO    - 4) tests/posix/fs/portability.posix.fs.tls.picolibc on intel_adsp_cavs18 error (No Console Output(Timeout))
INFO    - 5) tests/posix/common/portability.posix.common.picolibc on intel_adsp_cavs18 error (No Console Output(Timeout))
INFO    - 6) tests/posix/getopt/portability.posix.getopt.picolibc on intel_adsp_cavs18 error (No Console Output(Timeout))`

**Environment (please complete the following information):**
 - OS: Linux
 - Zephyr SDK
 - [https://github.com/zephyrproject-rtos/zephyr/commit/ef9d5bbddfa53ac5636437df8c4e720aaff974d5](url)

stephanosio, 2023-01-31:
cc @keith-packard

stephanosio, 2023-01-31:
@majunkier Can you please post a full console output?

nashif, 2023-01-31:
@majunkier  also, please verify running the test directly without twister. I can see the test passes if I do that.

keith-packard, 2023-01-31:
How can I help debug this? Is there something special about the console on this device that would require extra code to hook it up through the picolibc stdio bits?

cfriedt, 2023-01-31:
> How can I help debug this? Is there something special about the console on this device that would require extra code to hook it up through the picolibc stdio bits?

Good question - I'd also like to help if possible, but have no way to repro.

keith-packard, 2023-02-01:
so this bisects to just the picolibc version update? Without being able to reproduce this ourselves, perhaps you could bisect the picolibc module to find which commit in that tree causes the regression? There aren't many changes in that tree which could affect output like this; the bulk of the 1.8 version involved adding long double math library support.

nashif, 2023-02-01:
Looks like the real culprit is 7fa6a2b2b870417c4102d84eac62a6223ac0fb37. This and probably other picolibc tests never ran on those platforms, with this commit, the tests are not filtered anymore.

stephanosio, 2023-02-01:
@nashif Can you check if this is a simple output issue or the kernel failing to boot with picolibc altogether? (e.g. set a breakpoint on main and see if it gets there?)

nashif, 2023-02-01:
It is more than just the output, I ran this test on a simulator for another platform and getting this:

```
WARN: 31385300.3/324: ( [ core0.0 ]  ) *WARNING* IllegalInstructionException at double exception vector 0xa00303c0.
WARN: 31385300.3/324: ( [ core0.0 ]  ) *WARNING* IllegalInstructionException at double exception vector 0xa00303c0.
WARN: 31385300.3/324: ( [ core0.0 ]  ) *WARNING* IllegalInstructionException at double exception vector 0xa00303c0.
WARN: 31385400.3/326: ( [ core0.0 ]  ) *WARNING* IllegalInstructionException at double exception vector 0xa00303c0.
WARN: 31385400.3/326: ( [ core0.0 ]  ) *WARNING* IllegalInstructionException at double exception vector 0xa00303c0.
```

ran with
`west  build -b intel_adsp_ace15_mtpm -T tests/kernel/common/kernel.common.picolibc -t run` it is a different test, but I think it is the same. (We need to close all those duplicate bugs and just have one.)

stephanosio, 2023-02-01:
> ```
> WARN: 31385400.3/326: ( [ core0.0 ]  ) *WARNING* IllegalInstructionException at double exception vector 0xa00303c0.
> ```

What (function) is at 0xa00303c0?

nashif, 2023-02-01:
0xa00303c0 is ...

```
Disassembly of section .DoubleExceptionVector.text:

a00303c0 <_DoubleExceptionVector>:
        movi a3, 1
        movi a2, SYS_exit
        simcall
#elif XCHAL_HAVE_DEBUG
/* Signals an unhandled double exception */
1:      break   1, 4
a00303c0:       004140                  break   1, 4
#else
1:
#endif
        j       1b
a00303c3:       fffe46                  j       a00303c0 <_DoubleExceptionVector>

Disassembly of section .text:
```

nashif, 2023-02-01:
@teburd  @dcpleung  can you please help with debugging this?

nashif, 2023-02-01:
setting as a release blocker, we either find a proper fix or exclude those platforms like before and make them with PICOLIBC_UNSUPPORTED until we have a solution.

teburd, 2023-02-01:
I'll look today at this

teburd, 2023-02-01:
It seems like whats meant to be an inline zeroing loop in soc/xtensa/intel_adsp/common/include/soc.h is being replaced with a call to memset

This is a critical section of code that must be inlined and should not be jumping out with a call like this. The address to the call results in illegal op codes.

nashif, 2023-02-01:
Fix works for me on those platforms.

nashif, 2023-02-01:
@keith-packard  @stephanosio  any thoughts why this only happens with picolibc?

stephanosio, 2023-02-01:
> @keith-packard @stephanosio any thoughts why this only happens with picolibc?

https://github.com/zephyrproject-rtos/zephyr/blob/d686af9fa2c34cdbba728a3ca8f40ea992743d36/CMakeLists.txt#L282-L286

teburd, 2023-02-01:
> > @keith-packard @stephanosio any thoughts why this only happens with picolibc?
> 
> https://github.com/zephyrproject-rtos/zephyr/blob/d686af9fa2c34cdbba728a3ca8f40ea992743d36/CMakeLists.txt#L282-L286

That'll do it

keith-packard, 2023-02-01:
yup, that'll do it -- I'd *really* like to encourage the removal of -ffreestanding as it lets the compiler do some important optimizations when faced with standard C library calls, as well as do compile-time buffer overflow detection. Linux doesn't use -ffreestanding for most of the kernel, adding this flag for a few functions which (like the code above) must not be cleverly optimized by the compiler in these ways.

For this target, I'd suggest adding `-fno-tree-loop-distribute-patterns` as that will make gcc skip this particular class of optimizations.

stephanosio, 2023-02-01:
> For this target, I'd suggest adding `-fno-tree-loop-distribute-patterns` as that will make gcc skip this particular class of optimizations.

@keith-packard It turns out this issue can be worked around without specifying a GCC-specific flag/attribute (see https://github.com/zephyrproject-rtos/zephyr/pull/54314).

PR #54314 (2023-02-01): intel_adsp: bbzero/bmemcpy with picolibc fix
When building with picolibc and gcc, the loops to do zeroing/copying get replaced by gcc with calls to memset/memcpy. This fails this early in the boot process and results in an illegal instruction exception.

Marking the variables being manipulated in the calls as volatile prevents the compiler from optimizing the loops (replacing them with memset/memcpy).

Fixes #54224 for me
closed 2023-02-01 by stephanosio