zephyrproject-rtos/zephyr #54225: Intel CAVS: tests/lib/c_lib/ fails

majunkier, 2023-01-30:
**Describe the bug**
On CAVS 15, 18 and 25 tests/lib/c_lib/ fails.

**To Reproduce**
Build and run test on CAVS 15, 18 or 25. Twister options given in our CI:
`-W --hardware-map ../../cavs/cavs.map --device-testing -x=CONFIG_BOOT_DELAY=500 -T tests/lib/c_lib/  -v --no-skipped-report`

**Expected behavior**
Passes.

**Impact**
Annoyance.

**Logs and console output**
`tests/lib/c_lib/libraries.picolibc on intel_adsp_cavsX error (No Console Output(Timeout))`

**Environment (please complete the following information):**
 - OS: Linux
 - Zephyr SDK
 - [https://github.com/zephyrproject-rtos/zephyr/commit/ef9d5bbddfa53ac5636437df8c4e720aaff974d5](url)

majunkier, 2023-01-31:
`67db251`

stephanosio, 2023-02-01:
Closing as a duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/54224.

closed 2023-02-01 by stephanosio