zephyrproject-rtos/zephyr #54226: Code coverage collection is broken

SeppoTakalo, 2023-01-30:
**Code coverage collection is broken**
When running Twister using coverage reporting, it does not anymore produce LCOV output.

**To Reproduce**
Steps to reproduce the behavior:
1. `cd zephyr`
2. `twister -p native_posix -T tests/net/lib/lwm2m -c -C`
4. See error

**Expected behavior**
I would expect the Twister command above to produce a HTML file.
Instead I see error messages `lcov: ERROR: no valid records found in tracefile`

**Impact**
Cannot follow code coverage from Zephyr.

**Logs and console output**
```
❯ twister -p native_posix -T tests/net/lib/lwm2m -C
Renaming output directory to /home/seta/src/ncs/zephyr/twister-out.1
INFO    - Using Ninja..
INFO    - Zephyr version: v3.3.0-rc1-11-g8512bd18c666
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testsuite list...
INFO    - Writing JSON report /home/seta/src/ncs/zephyr/twister-out/testplan.json
INFO    - JOBS: 8
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - Total complete:    8/   8  100%  skipped:    4, failed:    0
INFO    - 8 test scenarios (8 test instances) selected, 4 configurations skipped (0 by static filter, 4 at runtime).
INFO    - 4 of 8 test configurations passed (100.00%), 0 failed, 4 skipped with 0 warnings in 13.57 seconds
INFO    - In total 88 test cases were executed, 99 skipped on 1 out of total 540 platforms (0.19%)
INFO    - 4 test configurations executed on platforms, 0 test configurations were only built.
INFO    - Generating coverage files...
/home/seta/src/ncs/zephyr/twister-out/native_posix/tests/net/lib/lwm2m/engine/net.lwm2m.engine/zephyr/subsys/random/CMakeFiles/subsys__random.dir/rand32_entropy_device.c.gcno:version 'A94*', prefer 'B21*'
geninfo: ERROR: GCOV failed for /home/seta/src/ncs/zephyr/twister-out/native_posix/tests/net/lib/lwm2m/engine/net.lwm2m.engine/zephyr/subsys/random/CMakeFiles/subsys__random.dir/rand32_entropy_device.c.gcda!
lcov: ERROR: no valid records found in tracefile /home/seta/src/ncs/zephyr/twister-out/coverage.info
lcov: ERROR: no valid records found in tracefile /home/seta/src/ncs/zephyr/twister-out/coverage.info
lcov: ERROR: no valid records found in tracefile /home/seta/src/ncs/zephyr/twister-out/coverage.info
lcov: ERROR: no valid records found in tracefile /home/seta/src/ncs/zephyr/twister-out/coverage.info
genhtml: ERROR: no valid records found in tracefile /home/seta/src/ncs/zephyr/twister-out/coverage.info
INFO    - Saving reports...
INFO    - Writing JSON report /home/seta/src/ncs/zephyr/twister-out/twister.json
INFO    - Writing xunit report /home/seta/src/ncs/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /home/seta/src/ncs/zephyr/twister-out/twister_report.xml...
INFO    - Run completed
```

When running on Zephyr tag `v3.0.0`, I get expected output.
```
❯ twister -p native_posix -T tests/net/lib/lwm2m -c -C
Deleting output directory /home/seta/src/ncs/zephyr/twister-out
INFO    - Zephyr version: zephyr-v3.0.0
INFO    - JOBS: 8
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testcase list...
INFO    - 4 test scenarios (4 configurations) selected, 0 configurations discarded due to filters.
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - Total complete:    4/   4  100%  skipped:    0, failed:    0
INFO    - 4 of 4 test configurations passed (100.00%), 0 failed, 0 skipped with 0 warnings in 29.44 seconds
INFO    - In total 99 test cases were executed, 0 skipped on 1 out of total 427 platforms (0.23%)
INFO    - 4 test configurations executed on platforms, 0 test configurations were only built.
INFO    - Generating coverage files...
INFO    - HTML report generated: /home/seta/src/ncs/zephyr/twister-out/coverage/index.html
INFO    - Saving reports...
INFO    - Writing xunit report /home/seta/src/ncs/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /home/seta/src/ncs/zephyr/twister-out/twister_report.xml...
INFO    - Run completed
```
**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.15.1
 - Commit SHA or Version used `8512bd18c6`

**Additional context**
According to Git Bisect, this commit broke the build:
```
47b675c6ee12084fb1c83afa946f5c2c0cb0adf5 is the first bad commit
commit 47b675c6ee12084fb1c83afa946f5c2c0cb0adf5
Author: Yuval Peress <peress@google.com>
Date:   Tue Nov 29 11:18:06 2022 -0700

    twister: Set a default gcov-tool for llvm
    
    When using llvm we need to set the gcov-tool to "llvm-cov gcov" but
    the lcov tool is incapable of passing arguments to the gcov-tool. i.e.
    the following cannot work:
    
    $ lcov --gcov-tool "llvm-cov gcov" ...
    
    Instead, create a symlink to llvm-cov prefixed as `gcov` which by the
    documentation of llvm-cov will alias to `llvm-cov gcov` subcommand.
    
    Signed-off-by: Yuval Peress <peress@google.com>

```

SeppoTakalo, 2023-01-30:
Breaking PR: https://github.com/zephyrproject-rtos/zephyr/pull/52644

SeppoTakalo, 2023-01-30:
Reverting the breaking commit seem to fix. Is this a LCOV version issue, or something else? Why the default is not working for me?

```
❯ twister -p unit_testing -c -C -i -T tests/unit/intmath/
Deleting output directory /home/seta/src/ncs/zephyr/twister-out
INFO    - Using Ninja..
INFO    - Zephyr version: v3.3.0-rc1-12-g83ffead48cb9
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testsuite list...
INFO    - Writing JSON report /home/seta/src/ncs/zephyr/twister-out/testplan.json
INFO    - JOBS: 8
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - Total complete:    1/   1  100%  skipped:    0, failed:    0
INFO    - 1 test scenarios (1 test instances) selected, 0 configurations skipped (0 by static filter, 0 at runtime).
INFO    - 1 of 1 test configurations passed (100.00%), 0 failed, 0 skipped with 0 warnings in 2.32 seconds
INFO    - In total 1 test cases were executed, 0 skipped on 1 out of total 540 platforms (0.19%)
INFO    - 1 test configurations executed on platforms, 0 test configurations were only built.
INFO    - Generating coverage files...
INFO    - HTML report generated: /home/seta/src/ncs/zephyr/twister-out/coverage/index.html
INFO    - Saving reports...
INFO    - Writing JSON report /home/seta/src/ncs/zephyr/twister-out/twister.json
INFO    - Writing xunit report /home/seta/src/ncs/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /home/seta/src/ncs/zephyr/twister-out/twister_report.xml...
INFO    - Run completed
❯ git reset --hard upstream/main
HEAD is now at 8512bd18c6 Bluetooth: Shell: Fix unintialized ad_len bug
❯ twister -p unit_testing -c -C -i -T tests/unit/intmath/
Deleting output directory /home/seta/src/ncs/zephyr/twister-out
INFO    - Using Ninja..
INFO    - Zephyr version: v3.3.0-rc1-11-g8512bd18c666
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testsuite list...
INFO    - Writing JSON report /home/seta/src/ncs/zephyr/twister-out/testplan.json
INFO    - JOBS: 8
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - Total complete:    1/   1  100%  skipped:    0, failed:    0
INFO    - 1 test scenarios (1 test instances) selected, 0 configurations skipped (0 by static filter, 0 at runtime).
INFO    - 1 of 1 test configurations passed (100.00%), 0 failed, 0 skipped with 0 warnings in 2.23 seconds
INFO    - In total 1 test cases were executed, 0 skipped on 1 out of total 540 platforms (0.19%)
INFO    - 1 test configurations executed on platforms, 0 test configurations were only built.
INFO    - Generating coverage files...
/home/seta/src/ncs/zephyr/twister-out/unit_testing/tests/unit/intmath/utilities.intmath/CMakeFiles/testbinary.dir/home/seta/src/ncs/zephyr/subsys/testsuite/ztest/src/ztest_new.c.gcno:version 'A94*', prefer 'B21*'
geninfo: ERROR: GCOV failed for /home/seta/src/ncs/zephyr/twister-out/unit_testing/tests/unit/intmath/utilities.intmath/CMakeFiles/testbinary.dir/home/seta/src/ncs/zephyr/subsys/testsuite/ztest/src/ztest_new.c.gcda!
lcov: ERROR: no valid records found in tracefile /home/seta/src/ncs/zephyr/twister-out/coverage.info
lcov: ERROR: no valid records found in tracefile /home/seta/src/ncs/zephyr/twister-out/coverage.info
lcov: ERROR: no valid records found in tracefile /home/seta/src/ncs/zephyr/twister-out/coverage.info
lcov: ERROR: no valid records found in tracefile /home/seta/src/ncs/zephyr/twister-out/coverage.info
genhtml: ERROR: no valid records found in tracefile /home/seta/src/ncs/zephyr/twister-out/coverage.info
INFO    - Saving reports...
INFO    - Writing JSON report /home/seta/src/ncs/zephyr/twister-out/twister.json
INFO    - Writing xunit report /home/seta/src/ncs/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /home/seta/src/ncs/zephyr/twister-out/twister_report.xml...
INFO    - Run completed
❯ git revert 47b675c6ee12084fb1c83afa946f5c2c0cb0adf5
Auto-merging scripts/pylib/twister/twisterlib/environment.py
[revert edcf00e301] Revert "twister: Set a default gcov-tool for llvm"
 3 files changed, 15 insertions(+), 41 deletions(-)
❯ twister -p unit_testing -c -C -i -T tests/unit/intmath/
Deleting output directory /home/seta/src/ncs/zephyr/twister-out
INFO    - Using Ninja..
INFO    - Zephyr version: v3.3.0-rc1-12-gedcf00e30190
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testsuite list...
INFO    - Writing JSON report /home/seta/src/ncs/zephyr/twister-out/testplan.json
INFO    - JOBS: 8
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - Total complete:    1/   1  100%  skipped:    0, failed:    0
INFO    - 1 test scenarios (1 test instances) selected, 0 configurations skipped (0 by static filter, 0 at runtime).
INFO    - 1 of 1 test configurations passed (100.00%), 0 failed, 0 skipped with 0 warnings in 2.20 seconds
INFO    - In total 1 test cases were executed, 0 skipped on 1 out of total 540 platforms (0.19%)
INFO    - 1 test configurations executed on platforms, 0 test configurations were only built.
INFO    - Generating coverage files...
INFO    - HTML report generated: /home/seta/src/ncs/zephyr/twister-out/coverage/index.html
INFO    - Saving reports...
INFO    - Writing JSON report /home/seta/src/ncs/zephyr/twister-out/twister.json
INFO    - Writing xunit report /home/seta/src/ncs/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /home/seta/src/ncs/zephyr/twister-out/twister_report.xml...
INFO    - Run completed

```

SeppoTakalo, 2023-01-30:
Specifying the proper tool seem to work
```
❯ twister -p unit_testing -c -C -i -T tests/unit/intmath/ --gcov-tool gcov
Deleting output directory /home/seta/src/ncs/zephyr/twister-out
INFO    - Using Ninja..
INFO    - Zephyr version: v3.3.0-rc1-11-g8512bd18c666
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testsuite list...
INFO    - Writing JSON report /home/seta/src/ncs/zephyr/twister-out/testplan.json
INFO    - JOBS: 8
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - Total complete:    1/   1  100%  skipped:    0, failed:    0
INFO    - 1 test scenarios (1 test instances) selected, 0 configurations skipped (0 by static filter, 0 at runtime).
INFO    - 1 of 1 test configurations passed (100.00%), 0 failed, 0 skipped with 0 warnings in 2.11 seconds
INFO    - In total 1 test cases were executed, 0 skipped on 1 out of total 540 platforms (0.19%)
INFO    - 1 test configurations executed on platforms, 0 test configurations were only built.
INFO    - Generating coverage files...
INFO    - HTML report generated: /home/seta/src/ncs/zephyr/twister-out/coverage/index.html
INFO    - Saving reports...
INFO    - Writing JSON report /home/seta/src/ncs/zephyr/twister-out/twister.json
INFO    - Writing xunit report /home/seta/src/ncs/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /home/seta/src/ncs/zephyr/twister-out/twister_report.xml...
INFO    - Run completed
```

nashif, 2023-02-01:
I am looking into this.

PR #54551 (2023-02-20): twister: restore logic to determine which gcov we should use
We do not want to use gcov from the SDK for native posix and unit tests,
instead we want the system gcov. Restore logic that determines that.

Fixes #54226
commit 1440b19b (2023-02-20): twister: restore logic to determine which gcov we should use

We do not want to use gcov from the SDK for native posix and unit tests,
instead we want the system gcov. Restore logic that determines that.

Fixes #54226
commit 835777c4 (2023-03-01): twister: restore logic to determine which gcov we should use

We do not want to use gcov from the SDK for native posix and unit tests,
instead we want the system gcov. Restore logic that determines that.

Fixes #54226

(cherry picked from commit 1440b19bded1b6508ecf5401bd5cc1f9512c5495)
closed 2023-02-20 by nashif