zephyrproject-rtos/zephyr #54230: STM32H7: Kernel crash with BCM4=0

jkrautmacher, 2023-01-30:
**Describe the bug**
The current Zephyr kernel from the `main` branch panics on the m4 core of
STM32H7 microcontrollers if the option byte for m4 boot on reset is set to zero
(`BCM4=0`).

The bug was initially discovered with a proprietary firmware but can be
reproduced with the basic blinky sample.

A `git bisect` session between zephyr `v2.6.0` and `v3.2.0` showed that the
commit with the hash `efc0e754631214c3f5886f8bb82d70a1f89333c7` introduced this
behaviour at least in blinky. This commit is based on Zephyr `v3.0.0-rc1`.

Switching to the previous commit with the hash
`c0cc6c7ba2d064b7ac888b6d306915e4b890c0de` or setting `BCM4=1` solves the
issue.

**To Reproduce**
Setup a Zephyr workspace as explained in the [getting started guide][1]. Then
run the following with a STM32H747i Disco board connected:

```
cd ~/zephyrproject/zephyr
git checkout efc0e754631214c3f5886f8bb82d70a1f89333c7
cd ..
west update
west build -p always -b stm32h747i_disco_m4 zephyr/samples/basic/blinky
west flash
west build -p always -b stm32h747i_disco_m7 zephyr/samples/basic/blinky
west flash
```

After the flashing you can set `BCM4=0` with the [STM32 Cube Programmer][2]:

```
STM32_Programmer_CLI -c port=SWD mode=UR -ob BCM4=0
```

You should see that the green LED (m7 core) blinks while the blue LED (m4 core)
does not. The m4 core failed due a kernel panic.

**Expected behavior**
The kernel should boot without a panic.

**Impact**
This makes it impossible for users of the `BCM4` option byte to update to
recent versions of Zephyr.

It has to be noted that without an unpatched Zephyr kernel it is anyway
impossible to deactivate m4 booting with `BCM4=0` because the kernel on the m7
force-boots the m4 if it is not running. Since this behaviour can be avoided
with a patch and should maybe be part of upstream Zephyr too the impact of
`BCM4=0` should be taken seriously.

**Logs and console output**
This is a GDB backtrace of the m4 kernel panic which was possible to capture
with a Segger J-Link (OpenOCD does not support m4 core debugging right now):

```
(gdb) backtrace
#0  arch_system_halt (reason=reason@entry=4) at /home/jan/zephyrproject/zephyr/kernel/fatal.c:30
#1  0x081026a4 in k_sys_fatal_error_handler (reason=reason@entry=4, esf=esf@entry=0x30000a10 <z_interrupt_stacks+1808>) at /home/jan/zephyrproject/zephyr/kernel/fatal.c:44
#2  0x081026c8 in z_fatal_error (reason=4, esf=0x30000a10 <z_interrupt_stacks+1808>) at /home/jan/zephyrproject/zephyr/kernel/fatal.c:127
#3  0x081007b2 in _oops () at /home/jan/zephyrproject/zephyr/arch/arm/core/aarch32/swap_helper.S:498
#4  <signal handler called>
#5  0x08100df0 in z_stm32_hsem_lock (retry=0, hsem=3) at /home/jan/zephyrproject/zephyr/soc/arm/st_stm32/common/stm32_hsem.h:125
#6  0x08100ede in stm32_clock_control_on (dev=<optimized out>, sub_system=0x8102f68 <gpio_stm32_cfg_k+12>) at /home/jan/zephyrproject/zephyr/drivers/clock_control/clock_stm32_ll_h7.c:375
#7  0x08102416 in gpio_stm32_clock_request (dev=<optimized out>, on=true) at /home/jan/zephyrproject/zephyr/drivers/gpio/gpio_stm32.c:255
#8  gpio_stm32_init (dev=<optimized out>) at /home/jan/zephyrproject/zephyr/drivers/gpio/gpio_stm32.c:624
#9  0x081017a4 in z_sys_init_run_level (level=level@entry=0) at /home/jan/zephyrproject/zephyr/kernel/device.c:71
#10 0x081018d4 in z_cstart () at /home/jan/zephyrproject/zephyr/kernel/init.c:420
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

**Environment (please complete the following information):**
- OS: Linux
- Toolchain: Zephyr SDK
- Commit SHA: `efc0e754631214c3f5886f8bb82d70a1f89333c7`

**Additional context**
- Option byte setting: `BCM4=0`

[1]: https://docs.zephyrproject.org/latest/develop/getting_started/index.html
[2]: https://www.st.com/en/development-tools/stm32cubeprog.html

jkrautmacher, 2023-01-30:
Minor update: The error also occurs on the current `main` branch (currently `ef9d5bbddf` which is based on `v3.3.0-rc1`).

gautierg-st, 2023-01-31:
I am not sure to understand the problem, though I admit I am not too familiar with the dual core aspect of the STM32H7.

I have tested the blinky sample on both the current main and the old commit you mentioned.

On the current main, with BOOT_CM4=0, the green LED (driven by M7 core) was blinking, and the blue LED (driven by M4 core) was not. Which seems logical to me since the BOOT_CM4 disables the boot of the CM4 core.

On the old commit (c0cc6c7ba2d064b7ac888b6d306915e4b890c0de), with BOOT_CM4=0, both LEDs are blinking, which seems strange as the M4 core should not boot and thus should not drive the blue LED.

Again, I'm not really familiar with this dual core setting, but I am not shocked that a kernel panic occurs on the M4 core when its boot is disabled.

I suppose what I'd like to understand is: what are you expecting the M4 core to do when it is disabled?

gautierg-st, 2023-01-31:
Ok, I've just seen that the M4 core can be waked up from the M7 side and that it is indeed done at boot if BOOT_CM4=0. So never mind my question

gautierg-st, 2023-01-31:
Increasing the value `#define HSEM_LOCK_DEFAULT_RETRY` (for example to 0xFFFFFFFEU, but that's a bit extreme) in stm32_hsem.h prevents the kernel panic.

gautierg-st, 2023-02-02:
@jkrautmacher I have pushed PR #54350 that should resolve the crash. Could you please give your feedback? Thanks

PR #54350 (2023-02-02): Fix STM32H7 Kernel crash with BCM4=0
When running the blinky example on STM32H747, with the BOOT_CM4 bit set to 0, the M4 core goes into panic.
Increasing the value of the hardware semaphore retry prevents this.

Fixes #54230