zephyrproject-rtos/zephyr #54246: Sample:subsys/ipc/rpmsg_server:The two cores cannot communicate in nRF5340

zzih1114, 2023-01-31:
**Two cores cannot communicate in nRF5340**
A clear and concise description of what the bug is.
Two cores cannot communicate in nRF5340，Used the Sample:subsys/ipc/rpmsg_server,the zephyr version is 3.2 and nrf version is 2.2.0.
but when I Used the same sample,change the zephyr version to 2.4 and nrf version to 1.5.0. the communication is normal.
How to let zephyr v3.2 's two cores communication sucess?

henrikbrixandersen, 2023-01-31:
Can you reproduce the issue on current zephyr `main` branch?

zzih1114, 2023-02-01:
> Can you reproduce the issue on current zephyr `main` branch?

Yes,I just tested it,the issue has reproduced on `main` branch. The nrf version is v2.2.0

carlescufi, 2023-02-08:
@zzih1114 since you are using nRF Connect SDK, you should ask in the Nordic Devzone, where someone from Nordic will gladly help you out. Thanks!
https://devzone.nordicsemi.com/

closed 2023-02-08 by carlescufi