zephyrproject-rtos/zephyr #54253: v3.3.0-rc1: stm32: IPv6 neighbour solicitation packets are not received without CONFIG_ETH_STM32_MULTICAST_FILTER

arbrauns, 2023-01-31:
**Describe the bug**
With v3.3.0-rc1, zephyr no longer receives neighbour solicitation packets (destination MAC `33:33:ff:xx:yy:zz`) from stm32 ethernet MACs. With `CONFIG_ETH_STM32_MULTICAST_FILTER=y`, which I would expect to make multicast reception *more* restrictive, the packets are received again and neighbour discovery works as expected.

**To Reproduce**
Unfortunately not very clean, sorry about that, but I think it should be reproducible with any STM32 board with ethernet:
1. Build with:
   ```
   CONFIG_NETWORKING=y
   CONFIG_NET_L2_ETHERNET=y
   CONFIG_ETH_STM32_MULTICAST_FILTER=n
   ```
2. From an attached (linux) device, ping the zephyr device's (link-local) IPv6 address
4. In packet dumps, observe that the neighbour solicitations go unanswered
5. Build with `CONFIG_ETH_STM32_MULTICAST_FILTER=y`
6. Try again, neighbour discovery works

**Expected behavior**
IPv6 neighbour discovery works as expected even without the multicast filter enabled.

**Impact**
This effectively breaks IPv6 connectivity by default for stm32 based devices.

**Logs and console output**
-

**Environment**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: v3.3.0-rc1 (e9a59518fc)

**Additional context**
This is a pre-release, so maybe this is expected behaviour and just not documented in the changelog yet.

cc #53850/@cperera-aud

arbrauns, 2023-01-31:
Yes, reverting a5f9fc2382f3d1a1d6c0118ee369a6cf97c501d5 and c76aaf3818317fd85250d3990352588882d0cb60 fixes the problem.

cperera-aud, 2023-01-31:
@arbrauns, sorry for the regression, looks like you've already put in a fix so I'll let you continue with it.

PR #54266 (2023-02-08): drivers: ethernet: stm32: fix multicast filter configuration
With `CONFIG_ETH_STM32_MULTICAST_FILTER=n`, `MACFFR` was being set to 0 instead of `ETH_MULTICASTFRAMESFILTER_NONE`, blocking all multicast traffic.

Instead, reset all the relevant bits first, then set either `ETH_MULTICASTFRAMESFILTER_HASHTABLE` or `ETH_MULTICASTFRAMESFILTER_NONE` depending on Kconfig.

Fixes #54253.

cc @erwango
closed 2023-02-08 by carlescufi