zephyrproject-rtos/zephyr #54301: Console doesn't work with power management enabled

mkettn, 2023-02-01:
**Describe the bug**
I cannot use the console (e.g. `console_getchar`) with power management enabled (`CONFIG_PM=`y). I tested this only on a esp32 board, but I am sure others are affected as well.

**To Reproduce**
Steps to reproduce the behavior:
1. goto `zephyr/samples/subsys/console/echo/`
2. echo "CONFIG_PM=y" >> prj.conf
3. mkdir build; cd build
4. cmake -GNinja -DBOARD=esp32
5. ninja flash
6. west espressif monitor -n LF

...and try to enter any characters, the console wont work.

**Expected behavior**
The sample to work as intended, even with power management turned on.

**Impact**
The device can be set to "config mode" and expects user input, during this mode I'd like to turn off power management. Not a showstopper, but annoying.

**Logs and console output**
not applicable

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - Toolchain: Zephyr SDK 0.15.1
 - Commit: 883e9d367f6cd489666a562c83df0f56458c94b2

nordicjm, 2023-02-01:
Works fine here on bl653_dvk, marking as espressif issue

sylvioalves, 2023-03-22:
@mkettn, sorry for taking so long to reply. The simple explanation is that PM features on ESP32 are still under development. Currently the PM only works for GPIO and timer wake up. Check `samples/boards/esp32/deep_sleep` and  samples/boards/esp32/light_sleep`.  
It means that enabling PM for uart wake up (when interrupt-driven is enabled) won't work yet.

mkettn, 2023-03-22:
Hello @sylvioalves 
thank you for the status update. When I have time I'll take a closer look at the samples you mentioned.

nashif, 2023-05-30:
@sylvioalves  Can you please look at this bug please and see if it still exists?

sylvioalves, 2023-05-31:
Yes, it does exists. We haven't added PM support over UART driver yet.

closed 2023-08-15 by github-actions[bot]