zephyrproject-rtos/zephyr #54330: West build takes hang issue during

SadanandPenore-eaton, 2023-02-01:
**Describe the bug**
When executing the west build command as shown in the attached image, the build process stops at random command.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?  -  Windows
 - What have you tried to diagnose or workaround this issue? - None
 - Is this a regression? If yes, have you been able to "git bisect" it to a Zephyr 3.1.0 SDK 0.14.2
   specific commit? 
 - ...

**To Reproduce**
Steps to reproduce the behavior:
1. Use west build -b nucleo_f429zi -p
2. See error
[image]
[image]

**Expected behavior**
After executing the build command the execution should not pause for long time and should not hang.

**Impact**
It is annoying to wait for long time after code is done and we are building the code.

**Logs and console output**
Refer locations where the build execution took a long pause or hanged. But the location is random.

[image]
[image]

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows) - Windows
 - Toolchain (e.g Zephyr SDK, ...) - Zephyr 3.1.0 SDK 0.14.2
 - Commit SHA or Version used Zephyr 3.1.0 SDK 0.14.2

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

nashif, 2023-02-01:
@mbolivar-nordic  FYI

nashif, 2023-02-01:
@SadanandPenore-eaton Can you please revise the title?

prayassamriya, 2023-02-05:
I think same issue is observed in 3.2 as well.. One thing to note that when we used to have arm embedded tool chain, it was not seen but as soon as we switched to zephyr sdk, we started seeing this issue randomly( not always).. some time it just hangs for 4-5 minutes without generating any new logs and we need to terminate the current process and re-start.
Is there a possibility that some antivirus/firewall could be playing the role? Have you or zephyr community came across any such observation where Corporate policy was restricting the zephyr tools and executables and some exceptions were required to mitigate. As we don't know, in which area such exception could be applied, therefore we are trying to gather as much information as possible.

SadanandPenore-eaton, 2023-02-07:
@prayassamriya and me are in same team. Also refer the additional comments from prayas.
The issue is also reproduced on 3.2.
I had already mentioned it for SDK 0.15.2 which uses 3.2.

mbolivar-nordic, 2023-02-08:
It would be good to know if any recent release is working for you. If not, then it does indeed strongly point to an environment issue on your end since nobody else has reported this issue. Antivirus is a good guess, but it's impossible to say from the outside.

carlescufi, 2023-03-06:
Reassigning to @nordicjm since this seems like a build system issue.

carlescufi, 2023-03-06:
@SadanandPenore-eaton @prayassamriya can you confirm this is still an issue for you?

nordicjm, 2023-03-06:
Run the configure commands using cmake instead of west and include the `--trace` argument to see where it is going wrong. When it goes wrong, what process executable is "stuck", what is the CPU utilisation of that process?

> Is there a possibility that some antivirus/firewall could be playing the role?

It could be an antivirus, what system is being used on these systems?

prayassamriya, 2023-03-06:
we don't know how to run it without west. for example what will be equivalent of "west build  -b nucleo_f429zi -p"

nordicjm, 2023-03-06:
```
cd <app_folder>
mkdir build
cd build
cmake -GNinja -DBOARD=nucleo_f429zi --trace ..
```

nordicjm, 2023-03-24:
@prayassamriya do you have any feedback on this?

SadanandPenore-eaton, 2023-03-27:
I tried this ninja command. It did not stuck anywhere.
Finished the whole process in one minute.

For the same folder structure west is taking 5 and half minutes to complete. ( If it did not stuck anywhere )

nordicjm, 2023-03-28:
Sounds like a west issue, assigning to @mbolivar-nordic

mbolivar-nordic, 2023-03-28:
In the screenshots showing a hung process, `west build` is just waiting for the `cmake` subprocess to finish. It isn't doing anything else. So none of this makes any sense to me, unfortunately.

zcmake.py lines 62-64:

```
    log.dbg('Running CMake:', quote_sh_list(cmd), level=log.VERBOSE_NORMAL)
    p = subprocess.Popen(cmd, **kwargs)
    out, _ = p.communicate()
```

you can see this if you run in verbose mode:

```
$ west -vvv build --cmake-only samples/hello_world
[...]
Running CMake: /usr/bin/cmake -DWEST_PYTHON=/home/mbolivar/.virtualenvs/west-dev/bin/python -B/home/mbolivar/zp/zephyr/build -GNinja -DBOARD=nrf52dk_nrf52832 -S/home/mbolivar/zp/zephyr/samples/hello_world
Loading Zephyr default modules (Zephyr base).
-- Application: /home/mbolivar/zp/zephyr/samples/hello_world
[...]
```

It seems very unlikely that the standard library's subprocess module is not working as documented, so I still have trouble believing this is a problem in west build.

It seems far more likely to me to be a problem in either the environment calling west build, or some specific command run by cmake in the build system.

@SadanandPenore-eaton if you want me to try to diagnose further, I'm going to need more detailed information in the form of `west -vvv build ... -- --trace-expand` output for a situation where the process hangs. For example, if you can get it to happen with the hello_world sample on the qemu_cortex_m3 board, then the complete output from this command would be useful:

```
$ west -vvv build -b qemu_cortex_m3 samples/hello_world/ -- --trace-expand
```

Please **paste** the **complete output** somewhere like a github gist:

- use copy/paste; do not post a picture of a screenshot
- paste the command's entire output, not just the parts that seem most relevant to you

Thanks!

carlescufi, 2023-05-19:
Closing due to inactivity, let us know if this is still an issue.

closed 2023-05-19 by carlescufi

prayassamriya, 2024-03-13:
An update on this:
The issue came out to be .cache folder which gets created inside zephyr folder for toolchain testing capability. We had the patch for zephyr. To apply the patch, we needed to revert the zephyr repo using git clean command and as a consequence it was removing the .cache folder as well. so every time we apply the patch, it was taking significant time and just hang for 4-5 minutes to generate zephyr/.cache data