zephyrproject-rtos/zephyr #54335: tests/kernel/fatal/no-multithreading/kernel.no-mt.cpu_exception failing on qemu_cortex_m3

aasthagr, 2023-02-01:
**Describe the bug**
The testcase kernel.no-mt.cpu_exception fails on qemu_cortex_m3

**To Reproduce**
Steps to reproduce the behavior:
twister -W -vvv -p qemu_cortex_m3 -T tests/kernel/fatal/no-multithreading/

**Expected behavior**
Passing testcase

**Impact**
Failing testcase

**Logs and console output**
```Running TESTSUITE fatal_no_mt
===================================================================
START - test_fatal
cpu exception
E: ***** USAGE FAULT *****
E:   Illegal use of the EPSR
E: r0/a1:  0x0000000e  r1/a2:  0x0000000a  r2/a3:  0x90000000
E: r3/a4:  0x20000c3c r12/ip:  0x00000043 r14/lr:  0x000003d7
E:  xpsr:  0x40000000
E: Faulting instruction address (r15/pc): 0x20000c3c
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: (nil) (unknown)
Caught system error -- reason 0
Unexpected reason (exp: 35)
===================================================================
RunID: fe85d68e207c79fb1fd8a25f70d0e1c2
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK v0.15.2
 - Commit SHA: `60a20471b561a0a3a74b377991fe1976c2ea83d7`

nashif, 2023-02-02:
the real bug is why this was not caught or built by twister in CI...

aasthagr, 2023-02-02:
> the real bug is why this was not caught or built by twister in CI...

Ohh yeah, that is true. Maybe will open a separate bug.

PR #54337 (2023-02-02): tests: kernel: cpu_exception: Make expected reason code generic.
Change expected reason code for cpu exception to be generic and in compliance with a3774fd51a716ffafeddfb017f9c25a046013132

Fixes #54335
commit 2196efe1 (2023-02-02): tests: kernel: cpu_exception: Make expected reason code generic.

Change expected reason code for cpu exception to be generic and
in compliance with a3774fd51a716ffafeddfb017f9c25a046013132

Fixes #54335
closed 2023-02-02 by nashif