zephyrproject-rtos/zephyr #54347: zephyr/posix/fcntl.h header works differently on native_posix platform

SeppoTakalo, 2023-02-02:
**Describe the bug**
When using `fcntl.h` header from Zephyr, one has to pay attention of the one corner case, which is native_posix platform, perhaps others.

As discovered in https://github.com/zephyrproject-rtos/zephyr/issues/54332 this causes side effects on native_posix that are not visible on other platform. Workaround is to use different header on `ARCH_POSIX` like this:
```
#ifdef CONFIG_ARCH_POSIX
#include <fcntl.h>
#else
#include <zephyr/posix/fcntl.h>
#endif
```

This workaround is already used in `zephyr/subsys/net/lib/sockets/sockets.c` which forces anyone using `fcntl()` on sockets to use the same workaround. Without including the same header flags like `O_NONBLOCK` might end up using different value, and therefore calls like `fcntl(sock, F_SETFD, O_NONBLOCK)` do not cause the wanted effect, but neither they produce any error.

**To Reproduce**
Steps to reproduce the behavior:
1. Include only `<zephyr/posix/fcntl.h>`
2. try to make a socket non-blocking
3. build with `native_posix`
4. Observe that the socket is blocking

**Expected behavior**
I would expect this workaround to be inside `include/zephyr/posix/fcntl.h` so the code inside Zephyr does not need to implement various workaround for various platforms.

**Impact**
Annoying, and might slow the development.

SeppoTakalo, 2023-02-07:
Ping @rlubos 

Now UDP and DTLS sockets work differently.

Regardless of what header you included, this call either works, or don't, and it depends on the socket type.
```c
zsock_fcntl(sock_fd, F_SETFL, O_NONBLOCK);
```

On `native_posix`, if the socket is UDP socket, it works. If the socket is DTLS socket, it does not.
This is because internal socket code, like I showed in previous post, eiher use the `<zephyr/posix/fcntl.h>` or `<fcntl.h>`.

Only proper fix would be that all code inside Zephyr should include `<zephyr/posix/fcntl.h>`, and if needed on some platforms, that code should include the `<fcntl.h>`.

rlubos, 2023-02-07:
There's already #54458 to fix the issue, TLS sockets should be handled in there as well.

PR #54458 (2023-02-08): socketpair improvements
4 commits:
* improvements for socketpair sample app (it was broken for a while)
* rename `socketpair_example.c` to `main.c`
* minor improvements for `socketpair.c`
* additional `<fcntl.h>` corrections in the networking subsystem

Fixes #54347
Fixes #52360
closed 2023-02-08 by stephanosio