zephyrproject-rtos/zephyr #54373: Mcuboot swap type is `test` when update fails

clamattia, 2023-02-02:
**Describe the bug**
When installing an image with mcuboot swap using move, the installation can fail if the image is too big.
In that case the old image is loaded again.
If we ask the image for its swap type using the `mcuboot_swap_type`-function we expect it to say `none`, but we get `test` instead.

```
 #include <stdbool.h>
#include <dfu/dfu_target.h>
#include <dfu/dfu_target_mcuboot.h>
#include <dfu/mcuboot.h>
void main(void)
{
    mcuboot_swap_type();
}
```

**To Reproduce**
Steps to reproduce the behavior:
1. Make mcuboot update swap using move fail by providing an image that is too big
2. use the `mcuboot_swap_type();` function somewhere in the firmware.

**Expected behavior**
After the upgrade failed, mcuboot reverts to the previous image. It is expected to be of swap type `none` but instead it is of type `test`

**Impact**
If someone is using the flag to check for a "healthy" boot, they get unexpected behavior. Reset cycles for example.

**Logs and console output**
```
*** Booting Zephyr OS build v3.1.99-ncs1-12-gaa84ba70d601  ***
<inf> mcuboot_util: Swap type: none
```

Let me know if additional context is needed.

clamattia, 2023-02-10:
FYI: It is not possible to clear the bad `test` state using `boot_write_img_confirmed()`.
Is there another possible workaround?

clamattia, 2023-04-12:
Is there any update on this? I see also there was no activity on the case here: https://github.com/mcu-tools/mcuboot/issues/1592

What I did not test is, whether the flag is only wrong or whether the update will be reattempted on reboot.

clamattia, 2023-04-12:
Please remove the stale label.

de-nordic, 2023-05-09:
This is known issue and only possible fix is to not upload too big image in the first place.
MCUboot uses "trailer" part of image slot to set flags, it does not use some dedicated area for that.
This means that if image is uploaded that writes up to the end of slot, it is not possible to clear flags or even flags can be strange because image data is interpreted as flags.

carlescufi, 2023-05-09:
This will be indirectly fixed by the new checks in #56299

clamattia, 2023-05-09:
Thanks :+1:
Did I understand correctly, that with the proposed PR, the build would fail at the signing stage, if it would lead to a image, that is to big for the slot?

de-nordic, 2023-05-09:
> Thanks 👍 Did I understand correctly, that with the proposed PR, the build would fail at the signing stage, if it would lead to a image, that is to big for the slot?

Yes. That is a plan. There is still some work required, because some build parameters to MCUboot need to be somehow passed to application build.

clamattia, 2023-05-09:
> > Thanks +1 Did I understand correctly, that with the proposed PR, the build would fail at the signing stage, if it would lead to a image, that is to big for the slot?
> 
> Yes. That is a plan. There is still some work required, because some build parameters to MCUboot need to be somehow passed to application build.

:+1: (Sorry for spam, github seems to have issues today)

clamattia, 2023-08-07:
@de-nordic @carlescufi do you know what the state of this is?

de-nordic, 2023-08-07:
> @de-nordic @carlescufi do you know what the state of this is?

Currently awaiting in queue after more important things kicked it down the stairs.

clamattia, 2023-08-07:
> > @de-nordic @carlescufi do you know what the state of this is?
> 
> Currently awaiting in queue after more important things kicked it down the stairs.

Ok, thanks for the information.

nordicjm, 2023-08-15:
> FYI: It is not possible to clear the bad `test` state using `boot_write_img_confirmed()`. Is there another possible workaround?

Support for this is being added in https://github.com/zephyrproject-rtos/zephyr/pull/61342

clamattia, 2023-10-16:
I would like this issue to remain open until fixed, if possible?

nordicjm, 2023-10-16:
As #61342 has been merged, does this not resolve the issue?

clamattia, 2023-10-16:
> As #61342 has been merged, does this not resolve the issue?

Did I see correctly, that this addresses issues related to `mcumgr`? The issue I report here, is reproduceable without `mcumgr`.

1. Swap type is `none`
2. Load in image into secondary slot and mark it for update. Swap type is `test`.
3. Reboot. Swapping fails (for example image too big for slot).
4. **Expected** :Swap type is `none`, **Got**: Swap type is `test`

clamattia, 2023-12-18:
If possible, I would prefer this to remain open until fixed.
I can try retesting it with later versions if I manage.

nordicjm, 2024-02-19:
This isn't really an issue anymore since https://github.com/zephyrproject-rtos/zephyr/pull/66615

nordicjm, 2024-02-19:
> There are other ways of update failures except image size (signature, encryption, ...)

They are addressed by it

> There are other ways to prepare an update without making use of mcumngr (for example nordics dfu_target-library.)

Then these need to check the size, dfu_target is outside the scope of zephyr

clamattia, 2024-02-19:
> > There are other ways of update failures except image size (signature, encryption, ...)
> 
> They are addressed by it

Am I understanding you correctly, that the swap can no longer fail under no circumstances?
If so, this would indirectly resolve the issue of the swap-type being wrong in the failure case.

nordicjm, 2024-02-19:
> Am I understanding you correctly, that the swap can no longer fail under no circumstances? If so, this would indirectly resolve the issue of the swap-type being wrong in the failure case.

It is the responsibility of the user (or transfer system) to check that an update is suitable for a system i.e. correct signature, correct size, etc.

clamattia, 2024-02-19:
> It is the responsibility of the user (or transfer system) to check that an update is suitable for a system i.e. correct signature, correct size, etc.

Fair point.
Nonetheless, `mcuboot` boots the old image if an error happens, I would expect it to have the swap type `none`.
I can accpet, if you do not agree. So feel free to close it.

closed 2024-05-06 by github-actions[bot]