zephyrproject-rtos/zephyr #54376: net: lwm2m: DTLS x509 certificate based support

RomainPelletant, 2023-02-02:
**Is your enhancement proposal related to a problem? Please describe.**
Zephyr currently supports NoSec and DTLS/PSK connection.

**Describe the solution you'd like**
Add support for security mode 2 (in LwM2M specifications) to support DTLS x509 certificate based.

I plan to implement it : do you see some limitations (max. certificate size for example, etc...?)

boaks, 2023-02-09:
Though zephyr uses mbedTLS, it's more enabling the configuration and provide the right credentials from the lwm2m security object (hope that the right lwm2m object name). There should be no other limits as for TLS. AFAIK in LWM2M support for ECDSA is mandatory (at least implicit by RFC 7252), so the certificate size should not be an too hard issue.

RomainPelletant, 2023-03-08:
@boaks Thanks for your message. You are right : it works with leshan with the block-wise transfer limitation (bootstrap to device when overwriting security data shall not exceed 1024 bytes).
I think that limitation could be a good opportunity to implement EST over CoAP (from my side) to have something fully operational.
I saw "Roadmap to EST" discussion on leshan GitHub, so will try it.

walzsi, 2023-04-14:
@RomainPelletant could you share the configs needed to get it work with the bootstrap?

RomainPelletant, 2023-04-14:
@walzsi the following PR works with direct x509 connection : https://github.com/zephyrproject-rtos/zephyr/pull/54420
Due to the blockwise limitation, the bootstrap security object send (for x509 certificate, client and server certificates) could fail if the message containing these 3 opaque data are greater than 1024 bytes.

SeppoTakalo, 2023-06-20:
FYI: pending for review: https://github.com/zephyrproject-rtos/zephyr/pull/59019

RomainPelletant, 2023-07-02:
@SeppoTakalo solved this issue. Thank you

closed 2023-07-02 by RomainPelletant