zephyrproject-rtos/zephyr #54401: Uninitialized has_param struct sometimes causes BSIM "has.sh" test to fail

Tronil, 2023-02-03:
In has_test.c, only `has_param.type` is actually set to a value while the rest of the `has_param` structure is left uninitialized. When running the test outside of the CI docker it will (usually) fail with:

```
d_00: @00:00:00.006488  [00:00:00.006,469] <dbg> bt_has: bt_has_register: Preset sync support only available for binaural hearing aid type
d_00: @00:00:00.006488 ERROR: (WEST_TOPDIR/zephyr/tests/bluetooth/bsim_bt/bsim_test_audio/src/has_test.c:57): HAS register failed (err -22)
```
Since `param->preset_sync_support` will usually end up being non-zero.

I guess the structure was meant to be zero initialized(?) in which case changing line 32 to the following should fix it:
```
	struct bt_has_register_param has_param = { 0 };
```

aescolar, 2023-02-08:
@MariuszSkamra  I believe you added this test?

aescolar, 2023-02-08:
CC @larsgk @Emil-Gydesen-Bose  @asbjornsabo

aescolar, 2023-02-08:
@Thalley could you triage?

larsgk, 2023-02-08:
> I guess the structure was meant to be zero initialized...

I agree with @Tronil - this wasn't caught when we did the feature. It should be initialized to 0

aescolar, 2023-02-08:
@larsgk will you be so kind as to create a PR with the fix?

larsgk, 2023-02-08:
> @larsgk will you be so kind as to create a PR with the fix?

I am looking at it now. I can't recreate the problem (running the tests multiple times) but can submit a PR with the fix

aescolar, 2023-02-08:
> > @larsgk will you be so kind as to create a PR with the fix?
> 
> I am looking at it now. I can't recreate the problem (running the tests multiple times) but can submit a PR with the fix

Thanks. Uninitialized variable errors will typically be machine dependent. If you want, you can run the simulated devices in valgrind so it flags the issue. Probably ASAN will also flag it.