zephyrproject-rtos/zephyr #54412: [hci_uart] nrf52840 & BlueZ 5.55 - start / stop scanning breaks

Eystorhaf, 2023-02-03:
Hey there,
we are currently investigating an issue where we are not able to start or stop le-scanning anymore with bluetoothctl as it always fails with ‘org.bluez.Error.InProgress’. We are using an nrf52840 with hci-uart on Zephyr LTS 2.7.X

Our usecase is that we are constantly scanning for le-advertisements and extract the manufacturer info from those. Once in a while we connect to devices to make config changes.
All seems to work well when we are below ~100 devices. Once we cross that line and the kernel stops the scan, bluetoothd should start it again. But this never happens. bluetoothd -d shows that the scan start is not sent to the hci device because bluez thinks the scan is still running (see https://github.com/bluez/bluez/blob/5.55/src/adapter.c#L2159)

`bluetoothd[9177]: src/adapter.c:update_discovery_filter() filters were equal, deciding to not restart the scan.`

When we try to start the scan again with bluetoothctl scan on it will respond with the error inProgress. But the scan is definitely not running and can still be started via btmgmt find -l.
It seems that the internal state, if the adapter is scanning (adapter->discovering), goes out of sync with the real adapter state if too many devices are around.

Did anyone else encounter that or has anyone an Idea how to fix that?We would love to help investigate that issue.

Please also mention any information which could help others to understand
the problem you're facing:

What target platform are you using?
nrf52840 attached via uart to a raspberry pi compute module / raspberry pi 3b+ with Bluez 5.55

What have you tried to diagnose or workaround this issue?
We have a comparison of the behaviour of the hci-device based on the nrf52840 vs. the on-board broadcom hci module of raspberry pi 3b+. On the broadcom module this behavior does not happen, on the same RPi3b+, same OS, same BT devices around. Please see the attached video.

To Reproduce
We only have this issue with the nrf52840 and Zephyr hci firmware (hci_uart) installed if ~100 BLE devices or more are in range.
With less devices everything works fine.

Expected behavior
the scan should run until it is actively stopped

Impact
What impact does this issue have on your progress (e.g., annoyance, showstopper)
For us it is a showstopper since we need the hci device to continue to work with BlueZ also with more than ~100 devices present.

Environment (please complete the following information):

OS: Linux / Raspbian (Bullseye),
nrf52840 Zephyr LTS 2.7.X (only supports le), hci-uart project
default BlueZ 5.55

btmon dump: https://drive.google.com/file/d/1Shrqb9hD19NUSjlOLixTLf6jYmihw944/view
screen recording nrf52840(zephir) vs. rpi3b+ onboard(broadcom): https://drive.google.com/file/d/1emW1Z2nom02TmSn--4_O7GYYZFlBm5L3/view?usp=share_link

bluetooth debug output:
https://github.com/bluez/bluez/files/10569109/bluetoothd-debug.log

cvinayak, 2023-02-05:
Could it be related to the usage as described here: https://github.com/zephyrproject-rtos/zephyr/issues/33688#issuecomment-807432890 ?

bitcloud, 2023-02-05:
I don't think so as there are no connections involved. It is pure scanning. At that moment.
It seems that the way the devices answers after changing it's internal state to scan stopped. I don't know if it expects the ok event as the next message right away because there are some adv messages before the scan stop result event is sent

cvinayak, 2023-02-06:
> I don't know if it expects the ok event as the next message right away because there are some adv messages before the scan stop result event is sent

You mean the below could be confusing to BlueZ ?
[image]

As hci_uart (UART) is duplex, there is always a chance of a command and an event cross-over at the transport. Preferably, BlueZ implementation should account for such race. Bluetooth HCI Test Specification for the Controller states the below in as the pass verdict:

> The IUT does not send any more LE Advertising Report Events after it sends the
HCI_Command_Complete for the HCI_LE_Set_Scan_Enable command that disables scanning.

bitcloud, 2023-02-06:
This is currently our working theory as it seems that this is something that happens more often on the zephyr device. We compared with the onboard raspberry pi module which doesn't seem to provoke that issue.
We also see there that some events happen after scan stop was sent but bluez seems to stay in line there. Thats why we opened it here and at bluez as well as we couldn't really pinpoint it to one side so far.

I'm not quite sure if github is the right place for bluez issues. https://github.com/bluez/bluez/issues/464
Or perhaps you know a better place for bluez issues? ;-)

carlescufi, 2023-02-07:
> I'm not quite sure if github is the right place for bluez issues. [bluez/bluez#464](https://github.com/bluez/bluez/issues/464)
> Or perhaps you know a better place for bluez issues? ;-)

I recommend Slack and the BlueZ mailing lists:
http://www.bluez.org/contact/

Closing this now.

closed 2023-02-07 by carlescufi

Eystorhaf, 2023-02-07:
Dear @carlescufi , we are not getting any feedback on the blueZ channels. We already tried the mailing list and Slack. I would appreciate if we could keep the issue open for other people to have a look at it.

broon-s, 2023-02-21:
@cvinayak, i also work at the company where we have this issue.
I looked again in the capture and saw that the HCI_MON control-event to confirm the scan stop has a rejection status flag `0x0b` (https://github.com/bluez/bluez/blob/721d5a8ae86d1ef1eea35109890b7622324cfc98/lib/mgmt.h#L297)

[image]

Wireshark filter for stat/stop discovery command with response:
`(hci_mon.opcode == 16) || (hci_mon.event == 0x0001) && ((frame[6:1] == 24) || (frame[6:1] == 23)) `

As far as the code looks like bluez's interpretation of that is that the stop failed and is still running.
Therefore the internal state is not set back to `adapter->discovering = false`.
https://github.com/bluez/bluez/blob/5.55/src/adapter.c#L1926
But the hci-module seems to stop scanning after sending the MGMT_OP_STOP_DISCOVERY event with status 0x0b.

If my analysis is correct, the follow up question would be who is right...
Should the ble-module continue scanning if responding with 0x0b or bluez stop the internal state anyway....

cvinayak, 2023-02-21:
As for the Bluetooth Controller is concern, a request to disable scanning was successful and no  more LE Advertising Reports are generated:
[image]